alexpevzner/sane-airscan #103: The backend crashes when working with ADF on Lexmark MB2236adw and Xerox 3225 devices

motorBMW, 2020-12-02:
Александр, приветствую!

В рамках эксплуатации выявили дефект. Сканирование через автоподатчик на устройствах Lexmark 2236 и Xerox 3225 выполняется, но завершается с ошибкой. Со стекла все сканирует безупречно. Я собрал логи, посмотрите на них пожалуйста. Может у вас есть идеи как это исправить?

[ftscand-Lexmark-International-Lexmark-MB2236adw.log](https://github.com/alexpevzner/sane-airscan/files/5628224/ftscand-Lexmark-International-Lexmark-MB2236adw.log)
[ftscand-Xerox-WorkCentre-3225.log](https://github.com/alexpevzner/sane-airscan/files/5628227/ftscand-Xerox-WorkCentre-3225.log)

Спасибо и хорошего дня!

alexpevzner, 2020-12-02:
Здравствуйте,

попробуйте пересобрать-перезапустить, я там кое-чего поправил. Эти странные устройства присылают сообщение об ошибке с HTTP кодом 200, чего я не ожидал.

Присылайте, пожалуйста, первичные сообщения о проблемах по-английски, чтобы другим гражданам тоже было понятно. Ну а детали можно уже и по-русски обсуждать :-)

motorBMW, 2020-12-03:
Александр, добрый день.

Я протестировал, теперь все работает штатно. Огромное спасибо!

motorBMW, 2020-12-06:
Александр, добрый день.

Подскажите, а когда выйдет новый релиз с исправлениями?

Спасибо!

alexpevzner, 2020-12-06:
Ну вот я колеблюсь, то ли сегодня-завтра выпустить, то ли сначала сделать обещанный блеклист

motorBMW, 2020-12-10:
Александр, добрый день!

Жду от вас с нетерпением нового релиза, что бы была возможность тиражировать техническое решение.

Спасибо вам огромное за вашу работу и хорошего дня!

alexpevzner, 2020-12-10:
Да, извините за задержку. Выпустил 0.99.22, через некоторое время оно пересоберется на OpenSUSE build services.

alexpevzner, 2020-12-10:
Fixed in 0.99.22

closed 2020-12-10 by alexpevzner