alexpevzner/sane-airscan #105: A Xerox B215

debiantriage, 2020-12-07:
https://forums.linuxmint.com/viewtopic.php?f=51&t=336775

> ty=Xerox B215 Multifunction Printer

This device appears to have come on the market a couple of years ago. I got carried away towards the end of the conversation. Users really are attached to their vendor drivers!

> chris@chris-Swift-SF113-31:~$ scanimage -L
device `airscan:e0:Xerox B215 (XRX9C934EAF9646)' is a eSCL Xerox B215 (XRX9C934EAF9646) ip=192.168.0.10

> chris@chris-Swift-SF113-31:~$ airscan-discover
[devices]
  Xerox B215 (XRX9C934EAF9646) = http://192.168.0.10:80/eSCL/, eSCL
  Xerox B215 (XRX9C934EAF9646) = https://192.168.0.10:443/eSCL/, eSCL
  Xerox B215 (XRX9C934EAF9646) = http://192.168.0.10:8018/wsd/scan, WSD
  Xerox B215 (XRX9C934EAF9646) = http://[FE80::9E93:4EFF:FEAF:9646%252]:8018/wsd/scan, WSD
  Xerox B215 (XRX9C934EAF9646) = http://192.168.3.1:8018/wsd/scan, WSD

alexpevzner, 2020-12-10:
> This device appears to have come on the market a couple of years ago. I got carried away towards the end of the conversation. Users really are attached to their vendor drivers!

Yes. Actually, it seems to be not easy to find alternative solution, if you don't know exactly what are you looking for.

Printer manuals rarely contains words like "eSCL" or "IPP over USB", so in most cases users don't know the right keyword to search for,

debiantriage, 2021-10-04:
The device is not recorded as working in the README. Sorry to give you more work.

Cheers,

Brian.

alexpevzner, 2021-10-04:
It is actually listed, at the end of the list

closed 2021-10-04 by alexpevzner

debiantriage, 2021-10-04:
So it is! My incompetence.

Incidentally, how close are we to having 200 devices recorded?

alexpevzner, 2021-10-04:
Currently we have 124 devices. 

After the first 100, having 200 is not as interesting, as having 1000 :-)

debiantriage, 2021-10-04:
> After the first 100, having 200 is not as interesting, as having 1000 :-)

You are a hard taskmaster! We cannot meet that target by Christmas 2021. I am already working day and night  to make users aware of the goodness of airscan :). Thanks you for everything you do.