alexpevzner/sane-airscan #107: Wrong output on HP Laser MFP 135w

rubenquidam, 2020-12-17:
$ airscan-discover 
    [devices]
      HP Inc. HP Laser MFP 135w = http://192.168.23.21:8018/wsd/scan, WSD
      HP Inc. HP Laser MFP 135w = http://192.168.223.1:8018/wsd/scan, WSD

That model is eSCL only, port 8018 is not open on the device, /wsd/scan does not exist on the device's management site, and 192.168.223.1 is not routable. I can't tell if this is a airscan-discover or avahi bug.

Scanning works fine if I add these lines to /etc/sane.d/airscan.conf

    [devices]
      HP Inc. HP Laser MFP 135w = http://192.168.23.21/eSCL, eSCL

Full log:

```
$ airscan-discover -d
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.23.11
WSDD:  + fe80::7285:c2ff:fe7d:5272%2
WSDD:  + 172.16.1.46%2
WSDD:  + 192.168.0.33%2
WSDD:  + fe80::bc19:1717:b7e3:164c%11
WSDD: 192.168.23.11: started discovery, UDP port=35609
WSDD: fe80::7285:c2ff:fe7d:5272: started discovery, UDP port=50136
WSDD: 172.16.1.46: started discovery, UDP port=52899
WSDD: setsockopt(AF_INET,IP_ADD_MEMBERSHIP): Address already in use
WSDD: 192.168.0.33: started discovery, UDP port=44573
WSDD: fe80::bc19:1717:b7e3:164c: started discovery, UDP port=46697
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
WSDD: HTTP POST http://192.168.23.21:8018/wsd
WSDD: HTTP POST http://192.168.223.1:8018/wsd
WSDD: HTTP resolving 192.168.23.21 8018
WSDD: HTTP trying 192.168.23.21:8018
WSDD: HTTP resolving 192.168.223.1 8018
WSDD: HTTP trying 192.168.223.1:8018
WSDD: HTTP POST http://192.168.23.21:8018/wsd: got response headers (200)
WSDD: HTTP POST http://192.168.23.21:8018/wsd: 200 OK
WSDD: HTTP POST http://192.168.23.21:8018/wsd: OK
WSDD: HTTP POST http://192.168.223.1:8018/wsd: Cancelled
zeroconf: found urn:uuid:16a65700-007c-1000-bb49-040e3c7422f5
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp9s0)
zeroconf:   name:      -
zeroconf:   model:     HP Inc. HP Laser MFP 135w
zeroconf:   addresses:
zeroconf:     192.168.223.1
zeroconf:     192.168.23.21
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:     http://192.168.23.21:8018/wsd/scan
zeroconf:     http://192.168.223.1:8018/wsd/scan
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: WSDD not finished...
WSDD: fe80::bc19:1717:b7e3:164c: done discovery
WSDD: 172.16.1.46: done discovery
WSDD: 192.168.0.33: done discovery
WSDD: 192.168.23.11: done discovery
WSDD: fe80::7285:c2ff:fe7d:5272: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: HP Inc. HP Laser MFP 135w (0): can: WSD, use: WSD
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   WSD   "w0:HP Inc. HP Laser MFP 135w"
[devices]
zeroconf: device_list wait: requested
zeroconf: device_list wait: OK
  HP Inc. HP Laser MFP 135w = http://192.168.23.21:8018/wsd/scan, WSD
  HP Inc. HP Laser MFP 135w = http://192.168.223.1:8018/wsd/scan, WSD
```

alexpevzner, 2020-12-17:
Hi Ruben,

the problem is in the device firmware, may be in device configuration.

The device really advertises itself as provider of WS-Scan service at the specified endpoints. And port 192.168.23.21:8018 is open, because in the log we can see that it responds to HTTP queries, sent to this port:
```
WSDD: HTTP POST http://192.168.23.21:8018/wsd: got response headers (200)
```

Strange that device can't be discovered as eSCL device. It may be because Bounjour discovery is disabled in the device configuration.

What do the following commands print?
```
avahi-browse -rt _uscan._tcp
avahi-browse -rt _uscans._tcp
```

alexpevzner, 2021-01-29:
Is this issue still relevant?

closed 2021-03-07 by alexpevzner