alexpevzner/sane-airscan #108: Hp scanner M148dw  is detected after several minutes if not after reboot

lucamax, 2020-12-17:
Hp scanner M148dw  and Mint 20 64 bits
I have to wait several minutes after turning on the HP scanner for the scanner to be detected .

After a fresh install of Mint 20 the scanner was not detected at all .
Following suggestions of this topic https://forums.linuxmint.com/viewtopic.php?f=51&t=329760&sid=fbcc443cbc717890a971282644d7cb38&start=20

I did the following
- remove  **ippusbxd** package
- Install **ipp-usb**  package
- install **sane-airsca**   package

After that,  the scanner was discovered but the device found do not allows to scan with Simple scan.
I have to wait several minutes for finally having other devices found and pick the only one that works.

Command scanimage -L at first list only one device :
```
$ scanimage -L
device `hpaio:/usb/HP_LaserJet_Pro_M148-M149?serial=VNCNN12803' is a Hewlett-Packard HP_LaserJet_Pro_M148-M149 all-in-one
```
While after some minutes :
```
$ scanimage -L
device `hpaio:/usb/HP_LaserJet_Pro_M148-M149?serial=VNCNN12803' is a Hewlett-Packard HP_LaserJet_Pro_M148-M149 all-in-one
device `hpaio:/net/hp_laserjet_pro_m148-m149?ip=192.168.0.114&queue=false' is a Hewlett-Packard hp_laserjet_pro_m148-m149 all-in-one
device `airscan:e0:HP LaserJet Pro M148dw (D70BC8)' is a eSCL HP LaserJet Pro M148dw (D70BC8) ip=192.168.0.114
```
**Only the one named "eSCL HP LaserjetPro M 148-m149" worked**

Command $ airscan-discover do not return any device for about three minutes , then...
[
```
$ airscan-discover
[devices]
HP LaserJet Pro M148dw (D70BC8) = http://192.168.0.114:8080/eSCL/, eSCL
HP LaserJet Pro M148dw (D70BC8) = https://192.168.0.114:443/eSCL/, eSCL
HP LaserJet Pro M148dw (D70BC8) = http://192.168.0.114:53048/, WSD
HP LaserJet Pro M148dw (D70BC8) = http://[fe80::c5d5:7ad:c131:63f4%253]:53048/, WSD
```

Following a suggestion I did this :
    Edit **/etc/sane.d/dll.conf** and comment out all lines in the file.
    Edit **/etc/sane.d/dll.d/hplip** and comment out **hpaio**.
    
After a reboot , it works fine, the scanner is seen immediately as "**eSCL HP LaserjetPro M 148-m149**"  and woks correctly.

But if I turn off the scanner and then I turn it on again, the scanner is not seen anymore by the system, scanimage -L do not return any scanner.
I have to wait several minutes, sometimes 3, other times more than 5 to get the scanner detected again.

alexpevzner, 2021-01-29:
Hi,

Sorry for long delay with reply.

Is this issue still relevant?

lucamax, 2021-01-30:
HI,
yes it still relevant.
I updated to 20.1 leaving the modifications done , but I have to wait several minutes for the scanner to be seen by the system.

cbp-22, 2022-03-15:
I am also using a HP LaserJet Pro M148dw, so I thought I'd add my two cents. My scanner
was also not recognized out of the box. Here is what worked for me.

HP LaserJet Pro M148dw
sane-airscan 0.99.27-1 on Artix Linux

- Installed the sane-airscan package from distro repository. hplip is not installed on this machine.
- Following the OP's suggestion, I explicitly added my scanner in /etc/sane.d/airscan.conf:

```
[devices]
#"Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL
#"Some Unwanted Scanner" = disable
"HP LaserJet Pro M148dw" = http:// 192.168.1.103:8080
"HP LaserJet Pro M148dw" = http:// 192.168.1.103:443/eSCL
```

- In the same file, I also uncommented the entire [options] section. Without uncommenting these lines, my scanner would not be found.

```
[options]
discovery = enable
model = network
protocol = auto
ws-discovery = fast
socket_dir = /var/run
```

After reboot my scanner is almost instantly recognized (it takes a few seconds):

```
~$ airscan-discover
[devices]
  HP LaserJet Pro M148dw = http://192.168.1.103:53048/, WSD
  HP LaserJet Pro M148dw = http://[fe80::7612:b3ff:fe71:6b2%252]:53048/, WSD
```
```
~$ scanimage -L
device `airscan:w0:HP LaserJet Pro M148dw' is a WSD HP LaserJet Pro M148dw ip=192.168.1.
103
```

`~$ scanimage --format=png --output-file test.png --progress`

works fine. So does simple-scan.