alexpevzner/sane-airscan #111: Fails to compile on FreeBSD

probonopd, 2020-12-29:
```
FreeBSD:/tmp/sane-airscan% gmake 
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
gmake: /usr/bin/pkg-config: No such file or directory
ctags -R .
ctags: illegal option -- R
usage: ctags [-BFTaduwvx] [-f tagsfile] file ...
gmake: [Makefile:80: tags] Fehler 1 (ignoriert)
mkdir -p objs/
gcc -c -o objs/airscan.o airscan.c  -O2 -g -W -Wall -Werror -pthread            -fPIC
mkdir -p objs/
gcc -c -o objs/airscan-array.o airscan-array.c  -O2 -g -W -Wall -Werror -pthread            -fPIC
mkdir -p objs/
gcc -c -o objs/airscan-bmp.o airscan-bmp.c  -O2 -g -W -Wall -Werror -pthread            -fPIC
airscan-bmp.c: In function 'image_decoder_bmp_begin':
airscan-bmp.c:85:26: error: implicit declaration of function 'le16toh' [-Werror=implicit-function-declaration]
   85 |     file_header.bfType = le16toh(file_header.bfType);
      |                          ^~~~~~~
airscan-bmp.c:86:26: error: implicit declaration of function 'le32toh' [-Werror=implicit-function-declaration]
   86 |     file_header.bfSize = le32toh(file_header.bfSize);
      |                          ^~~~~~~
cc1: all warnings being treated as errors
gmake: *** [Makefile:73: objs/airscan-bmp.o] Fehler 1
```

probonopd, 2020-12-29:
```
FreeBSD:/tmp/sane-airscan% sudo ln -s /usr/local/bin/pkg-config /usr/bin/pkg-config

FreeBSD:/tmp/sane-airscan% gmake
ctags -R .
ctags: illegal option -- R
usage: ctags [-BFTaduwvx] [-f tagsfile] file ...
gmake: [Makefile:80: tags] Fehler 1 (ignoriert)
mkdir -p objs/
gcc -c -o objs/airscan-bmp.o airscan-bmp.c  -O2 -g -W -Wall -Werror -pthread   -D_REENTRANT -D_THREAD_SAFE -pthread -I/usr/local/include   -I/usr/local/include/libxml2   -I/usr/local/include -I/usr/local/include/p11-kit-1   -I/usr/local/include   -I/usr/local/include/libpng16  -fPIC
airscan-bmp.c: In function 'image_decoder_bmp_begin':
airscan-bmp.c:85:26: error: implicit declaration of function 'le16toh' [-Werror=implicit-function-declaration]
   85 |     file_header.bfType = le16toh(file_header.bfType);
      |                          ^~~~~~~
airscan-bmp.c:86:26: error: implicit declaration of function 'le32toh' [-Werror=implicit-function-declaration]
   86 |     file_header.bfSize = le32toh(file_header.bfSize);
      |                          ^~~~~~~
cc1: all warnings being treated as errors
gmake: *** [Makefile:73: objs/airscan-bmp.o] Fehler 1
```

alexpevzner, 2020-12-29:
How to fix the build `#include <endian.h>` or `#include <sys/endian.h>`, or both works?

probonopd, 2020-12-29:
I don't know but if it helps I can give you access to a FreeBSD machine.

alexpevzner, 2020-12-29:
But how did you test your HP printer? Or it was on Linux?

No, I'm not on IRC

probonopd, 2020-12-29:
I just opened the HTTP website using the web browser and saw that it says "WS-Scan: active".

alexpevzner, 2020-12-29:
Well. I usually add to the list only devices known to actually work, not just advertising the service. This is why for most devices WSD compatibility left blank: if device supports both eSCL and WSD, sane-airscan chooses eSCL, so even if device advertises WSD support, in most cases I don't have information, does it actually work (unless user bothers to switch to the manual protocol selection mode and test both protocols)

probonopd, 2020-12-29:
Yes, let's add it only after I was able to positively test it.

probonopd, 2020-12-29:
I am setting up a CI build on FreeBSD using Cirrus CI right now.

alexpevzner, 2020-12-29:
I've blindly attempted to fix this problem. Please, test.

There are may be some another FreeBSD issues, but I expect they will be quite minor, because driver already ported to OpenBSD

probonopd, 2020-12-29:
Thanks @alexpevzner, now I it compiles much more but then I am getting

```
airscan-netif.c: In function 'netif_init':
airscan-netif.c:582:9: error: implicit declaration of function 'ROUTE_FILTER' [-Werror=implicit-function-declaration]
  582 |         ROUTE_FILTER(RTM_NEWADDR) | ROUTE_FILTER(RTM_DELADDR);
      |         ^~~~~~~~~~~~
airscan-netif.c:583:52: error: 'ROUTE_MSGFILTER' undeclared (first use in this function)
  583 |     if (setsockopt(netif_rtnetlink_sock, AF_ROUTE, ROUTE_MSGFILTER,
      |                                                    ^~~~~~~~~~~~~~~
airscan-netif.c:583:52: note: each undeclared identifier is reported only once for each function it appears in
cc1: all warnings being treated as errors
gmake: *** [Makefile:73: objs/airscan-netif.o] Fehler 1
```

alexpevzner, 2020-12-29:
Try again, please

probonopd, 2020-12-29:
```
gcc -c -o objs/airscan-netif.o airscan-netif.c  -O2 -g -W -Wall -Werror -pthread   -D_REENTRANT -D_THREAD_SAFE -pthread -I/usr/local/include   -I/usr/local/include/libxml2   -I/usr/local/include -I/usr/local/include/p11-kit-1   -I/usr/local/include   -I/usr/local/include/libpng16  -fPIC
airscan-netif.c: In function 'netif_notifier_read_callback':
airscan-netif.c:500:12: error: comparison of integer expressions of different signedness: 'int' and 'long unsigned int' [-Werror=sign-compare]
  500 |     if (rc >= sizeof(struct rt_msghdr)) {
      |            ^~
cc1: all warnings being treated as errors
gmake: *** [Makefile:73: objs/airscan-netif.o] Fehler 1
```

alexpevzner, 2020-12-29:
Once again, please

probonopd, 2020-12-29:
```
gcc -c -o objs/airscan-os.o airscan-os.c  -O2 -g -W -Wall -Werror -pthread   -D_REENTRANT -D_THREAD_SAFE -pthread -I/usr/local/include   -I/usr/local/include/libxml2   -I/usr/local/include -I/usr/local/include/p11-kit-1   -I/usr/local/include   -I/usr/local/include/libpng16  -fPIC
airscan-os.c: In function 'os_progname_init':
airscan-os.c:99:5: error: #error FIX ME
   99 | #   error FIX ME
      |     ^~~~~
gmake: *** [Makefile:73: objs/airscan-os.o] Fehler 1
```

I can feel that we are close now ;-)

alexpevzner, 2020-12-29:
Try again

probonopd, 2020-12-29:
Yes! It works :+1: 

Thank you so very, very much.

```
FreeBSD:/tmp/sane-airscan% ./airscan-discover 
[devices]
  HP LaserJet 200 colorMFP M276n (5F036D) = http://[...]:53048/, WSD
  HP LaserJet 200 colorMFP M276n (5F036D) = http://...:53048/, WSD
  HP LaserJet 200 colorMFP M276n (5F036D) = http://[...%251]:53048/, WSD
```

I will try actual scanning next.

probonopd, 2020-12-29:
```
FreeBSD:/tmp/sane-airscan% ./test
sane_init: calling
sane_init: done, status=Success
sane_open: calling
sane_open: done, status=Success
sane_get_parameters: calling
sane_get_parameters: done, status=Success
image size: 2480x3507
sane_start: calling
sane_start: done, status=Success
```

I can hear the scanner working but then it just stalls there.

probonopd, 2020-12-29:
It would be nice if on FreeBSD it would install to `/usr/local`.

```
OptiPlex-780:/tmp/sane-airscan% sudo make install
ctags -R .
ctags: illegal option -- R
usage: ctags [-BFTaduwvx] [-f tagsfile] file ...
*** Error code 1 (ignored)
mkdir -p /usr/bin
mkdir -p /etc/sane.d
mkdir -p /etc/sane.d/dll.d
install -s -D airscan-discover /usr/bin
usage: install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 file2
       install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 ... fileN directory
       install -dU [-vU] [-g group] [-m mode] [-N dbdir] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               directory ...
*** Error code 64

Stop.
make: stopped in /tmp/sane-airscan
```

alexpevzner, 2020-12-29:
> I can hear the scanner working but then it just stalls there.

I need protocol trace. In the `/etc/sane.d/airscan.conf` uncomment the following lines:
```
[debug]
trace = ~/airscan/trace
enable = true
```

Logs will be in the `~/airscan/trace` directory, which will be created automatically. `.tar` files are not needed for now, only `.log` files.

> It would be nice if on FreeBSD it would install to /usr/local

May be, meson build can do it automatically?

probonopd, 2020-12-29:
Is the `-R` in `ctags -R` crucial?

alexpevzner, 2020-12-29:
For Linux, yes. But ctags itself is not critical, any errors from it are ignored (please notice `-` prefix)

probonopd, 2020-12-29:
> May be, meson build can do it automatically?

Is it documented somewhere?

I am trying 

```
meson --reconfigure --prefix /usr/local build
# Remove -r in Makefile; or else gmake fails in the next step
gmake -j$(nproc)
DESTDIR=./appdir gmake install
find appdir

appdir
appdir/etc
appdir/etc/sane.d
appdir/etc/sane.d/dll.d
appdir/usr
appdir/usr/bin
```

probonopd, 2020-12-29:
> I need protocol trace. 

Will provide, once I get it to install to `/usr/local`.

alexpevzner, 2020-12-29:
You may temporary install it by copying libsane-airscan.so.1 to the proper directory (the same directory where other SANE .so files go).

Please note, that `airscan.conf` is searched in the `/etc/sane.d` directory

probonopd, 2020-12-29:
That directory is `/usr/local/etc/sane.d/` on FreeBSD.

probonopd, 2020-12-29:
> I can hear the scanner working but then it just stalls there.

Maybe I was just too impatient, after a minute or so it ends with `API: sane_exit(): OK`. Is this the correct behavior? Should there be a scanned file somewhere?

alexpevzner, 2020-12-29:
You mean the `test` program that builds together with driver?

If you see something like this about 30 lines before the end of the output, everything looks ok:
> 26846724 bytes of data received

This program doesn't leave the scanned image somewhere on the disk.

alexpevzner, 2020-12-29:
Show me at least its output, so I will be able to tell more

probonopd, 2020-12-29:
```
HP LaserJet 200 colorMFP M276n (5F036D): DEVICE_STM_SCANNING->DEVICE_STM_DONE
HP LaserJet 200 colorMFP M276n (5F036D): DEVICE_STM_DONE->DEVICE_STM_IDLE
HP LaserJet 200 colorMFP M276n (5F036D): API: sane_read(): End of file reached
26092080 bytes of data received
HP LaserJet 200 colorMFP M276n (5F036D): API: sane_close(): called
HP LaserJet 200 colorMFP M276n (5F036D): DEVICE_STM_IDLE->DEVICE_STM_CLOSED
HP LaserJet 200 colorMFP M276n (5F036D): removed from device table
HP LaserJet 200 colorMFP M276n (5F036D): closed protocol "WSD"
HP LaserJet 200 colorMFP M276n (5F036D): closed decoder: png
HP LaserJet 200 colorMFP M276n (5F036D): device destroyed
HP LaserJet 200 colorMFP M276n (5F036D): API: sane_close(): done
API: sane_exit(): called
zeroconf: device gone urn:uuid:434e4438-4637-5036-4b33-f0921c5f036d
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 1 (em0)
zeroconf: device gone urn:uuid:434e4438-4637-5036-4b33-f0921c5f036d
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 1 (em0)
API: sane_exit(): OK
```

probonopd, 2020-12-29:
```
HP LaserJet 200 colorMFP M276n (...): DEVICE_STM_SCANNING->DEVICE_STM_DONE
HP LaserJet 200 colorMFP M276n (...): DEVICE_STM_DONE->DEVICE_STM_IDLE
HP LaserJet 200 colorMFP M276n (...): API: sane_read(): End of file reached
26092080 bytes of data received
HP LaserJet 200 colorMFP M276n (...): API: sane_close(): called
HP LaserJet 200 colorMFP M276n (...): DEVICE_STM_IDLE->DEVICE_STM_CLOSED
HP LaserJet 200 colorMFP M276n (...): removed from device table
HP LaserJet 200 colorMFP M276n (...): closed protocol "WSD"
HP LaserJet 200 colorMFP M276n (...): closed decoder: png
HP LaserJet 200 colorMFP M276n (...): device destroyed
HP LaserJet 200 colorMFP M276n (...): API: sane_close(): done
API: sane_exit(): called
zeroconf: device gone urn:uuid:...
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 1 (em0)
zeroconf: device gone urn:uuid:...
zeroconf:   method:    ZEROCONF_MDNS_HINT
zeroconf:   interface: 1 (em0)
API: sane_exit(): OK
```

alexpevzner, 2020-12-29:
Looks that it scans. You may manually put `libsane-airscan.so.1` to the proper place and try with some "normal" client (xsane or whatever you like). If it won't find configuration file, it will use reasonable defaults

probonopd, 2020-12-29:
Xsane errors out on me with a `hpaio:/` error, unrelated to `sane-airscan` (actually this was the reason I was searching for something like `sane-airscan` in the first place...) - is there a way to tell it that it should _just_ use `sane-airscan`?

alexpevzner, 2020-12-29:
You need to add `airscan` to `/usr/local/etc/sane.d/dll.conf` (I forgot to tell you that), or copy `dll.conf` to `/usr/local/etc/sane.d/dll.d/airscan`, how `make install` does (when it works)

probonopd, 2020-12-29:
Yes! **It works!** :+1: 

Are you interested in a pull request for continuous builds on FreeBSD?

alexpevzner, 2020-12-29:
> Are you interested in a pull request for continuous builds on FreeBSD?

Yes, but it should not break existent Makefile, too many things depend on its behaviour (for example, official Fedora/Debian builds and its derivatives).

As far as I can guess, the major problem is `make install`. What if FreeBSD version of `make install` will be handled by separate script (say, `install.freebsd`)? Is it possible to slightly modify the Makefile, so it will automatically detect FreeBSD and call this script when appropriate?

alexpevzner, 2020-12-29:
Oops. Looks I didn't understand what you are asking for.

I would appreciate if you will help resolving remaining FreeBSD issues, so everything will work out of box

alexpevzner, 2020-12-29:
What `uname` command prints for your OS?

probonopd, 2020-12-29:
```
uname
FreeBSD

uname -a
FreeBSD myhostname 12.1-RELEASE-p8 FreeBSD 12.1-RELEASE-p8 GENERIC  amd64
```

probonopd, 2020-12-29:
> Oops. Looks I didn't understand what you are asking for.

A continuous build server can build your project whenever you `git push` (or pull request) something to it. This way, you can always see if a commit (or pull request) breaks the build. I have set up this for FreeBSD and can send a pull request to your project if you'd like to integrate this into this project. It is almost no extra work for you.

> I would appreciate if you will help resolving remaining FreeBSD issues, so everything will work out of box

Sure, I am happy to help with testing and reporting what doesn't work yet. If we could get things installed into the locations that FreeBSD uses that would already be a great step.

alexpevzner, 2020-12-29:
> I have set up this for FreeBSD and can send a pull request to your project if you'd like to integrate this into this project

Yes, I'm interested

> Sure, I am happy to help with testing and reporting what doesn't work yet.

Are you able to write `install.freebsd` script, that installs things into the proper places, and `uninstall.freebsd` that perform the reverse task?

Basically, installation is very simple. On linux it installs:
* airscan.conf -> /etc/sane.d/airscan.conf
* dll.cond -> /etc/sane.d/dll.d/airscan
* libsane-airscan.so.1 -> /usr/lib64/sane/libsane-airscan.so.1 (the exact path may depend on arch and distro)
* airscan-discover -> /usr/bin/airscan-discover
* airscan-discover.1 -> /usr/share/man/man1/airscan-discover.1.gz
* sane-airscan.5 -> /usr/share/man/man5/sane-airscan.5.gz

Man pages are gzipped and binaries are stripped

probonopd, 2020-12-29:
A script, as in shell script? Isn't the `make install` part supposed to handle this?
Basically, what is `/etc` on Linux becomes `/usr/local/etc` on FreeBSD, and what is `/usr/...` on Linux becomes `/usr/local/...` on FreeBSD.

alexpevzner, 2020-12-30:
Two more questions:
* what is the installation path for `libsane-airscan.so.1`?
* what `pkg-config --variable=libdir sane-backends` prints?

probonopd, 2020-12-30:
```
/usr/local/lib/sane/libsane-...

pkg-config --variable=libdir sane-backends
/usr/local/lib
```

alexpevzner, 2020-12-31:
Hi!

I've made installation of FreeBSD to /usr/local. Please, test.

probonopd, 2021-01-02:
```
% sudo make install
make: "/tmp/sane-airscan/Makefile" line 38: Need an operator
make: "/tmp/sane-airscan/Makefile" line 40: Need an operator
make: "/tmp/sane-airscan/Makefile" line 42: Need an operator
make: "/tmp/sane-airscan/Makefile" line 44: Need an operator
make: "/tmp/sane-airscan/Makefile" line 46: Need an operator
make: "/tmp/sane-airscan/Makefile" line 48: Need an operator
make: Fatal errors encountered -- cannot continue
make: stopped in /tmp/sane-airscan
```

probonopd, 2021-01-02:
```
(...) client   -L/usr/local/lib -lxml2   -L/usr/local/lib -lgnutls  -lm -lpthread -fPIE  -L/usr/local/lib -ljpeg   -L/usr/local/lib -lpng16 -lz 
mkdir -p //usr/local/bin
mkdir -p //usr/local/etc/sane.d
mkdir -p //usr/local/etc/sane.d/dll.d
install -s -D airscan-discover //usr/local/bin
usage: install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 file2
       install [-bCcpSsUv] [-f flags] [-g group] [-m mode] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               [-B suffix] [-l linkflags] [-N dbdir]
               file1 ... fileN directory
       install -dU [-vU] [-g group] [-m mode] [-N dbdir] [-o owner]
               [-M log] [-D dest] [-h hash] [-T tags]
               directory ...
gmake: *** [Makefile:123: install] Error 64
```

Almost...

probonopd, 2021-01-02:
Like this it works for me:

```
install: all
        mkdir -p $(DESTDIR)/$(bindir)
        mkdir -p $(DESTDIR)/$(CONFDIR)
        mkdir -p $(DESTDIR)/$(CONFDIR)/dll.d
        $(INSTALL) $(STRIP) $(DISCOVER) $(DESTDIR)/$(bindir)
        [ -e $(DESTDIR)/$(CONFDIR)/airscan.conf ] || cp airscan.conf $(DESTDIR)/$(CONFDIR)
        [ -e $(DESTDIR)/$(CONFDIR)/dll.d/airscan ] || cp dll.conf $(DESTDIR)/$(CONFDIR)/dll.d/airscan
        mkdir -p $(DESTDIR)/$(libdir)/sane
        $(INSTALL) $(STRIP) $(BACKEND) $(DESTDIR)/$(libdir)/sane
        mkdir -p $(DESTDIR)/$(mandir)/man1
        mkdir -p $(DESTDIR)/$(mandir)/man5
        $(INSTALL) -m 644 $(MAN_DISCOVER) $(DESTDIR)/$(mandir)/man1
        $(INSTALL) -m 644 $(MAN_BACKEND) $(DESTDIR)/$(mandir)/man5
        [ "$(COMPRESS)" = "" ] || $(COMPRESS) -f $(DESTDIR)/$(mandir)/man1/$(MAN_DISCOVER)
        [ "$(COMPRESS)" = "" ] || $(COMPRESS) -f $(DESTDIR)/$(mandir)/man5/$(MAN_BACKEND)
```

probonopd, 2021-01-02:
Note that I still need `sudo ln -s /usr/local/bin/pkg-config /usr/bin/pkg-config`

or else I will get

```
airscan-xml.c:13:10: fatal error: libxml/parser.h: No such file or directory
   13 | #include <libxml/parser.h>
      |          ^~~~~~~~~~~~~~~~~
```

alexpevzner, 2021-01-02:
Please, try again

probonopd, 2021-01-02:
It compiles and installs without any errors, but xsane does not find a scanner now.

alexpevzner, 2021-01-02:
It is enabled by default. And I didn't recently touch anything that could be related to this problem.

If you've enabled protocol trace, in the `~/airscan/trace/XXX-zeroconf.log` files there should be few lines in the beginning with the actual configuration. Could you please look there or post it here with unedited `airscan.conf` file?

alexpevzner, 2021-01-02:
Note, on FreeBSD configuration file now loads from `/usr/local/etc/sane.d/airscan.conf`, not from `/etc/sane.d/airscan.conf`

probonopd, 2021-01-02:
https://github.com/alexpevzner/sane-airscan/blob/1d2869f83fadc52f2eb1bc36d67019d12a29a70f/airscan.conf#L45

After enabling it there, it is working for me.

alexpevzner, 2021-01-02:
It needs to be investigated somehow. discovery option is enabled by default, and on Linux it doesn't need to be uncommented.

Looks like something goes wrong on initialization time. I need trace logs (with unmodified `airscan.conf`) to investigate it. If you don't want to publish it here, you can send me it directly to pzz@apevzner.com

probonopd, 2021-01-02:
> trace logs (with unmodified airscan.conf) 

How do i do this? I think I need to edit it to turn on the logging?

As soon as I do this, it _works..._
I need to test more, maybe I was just impatient and the MFP takes a while after switching it on until the scanner is published on the network.

alexpevzner, 2021-01-02:
You may compare `scanimage -L` against `airscan-discover`. The `airscan-discover` always performs discovery, regardless of configuration file content.

alexpevzner, 2021-01-02:
Hi @espindola,

we now have a FreeBSD port, based on your OpenBSD port. I want to ask you to test, that there is no regression on OpenBSD.

Please note now `make install` on non-Linux systems installs things by default to `/usr/local`. I don't know, is it good for OpenBSD or not; if not, it can easily be changed.

probonopd, 2021-01-03:
I can confirm that everything is working now on FreeBSD as it should. 

Thank you very much @alexpevzner :+1:

probonopd, 2021-01-03:
Do you think you could upload an updated patch to https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=245540?

probonopd, 2021-01-03:
> Kubilay Kocak freebsd_committer freebsd_triage 2021-01-03 01:22:26 UTC
^Triage: To progress this issue:
>
>- Provide an updated patch if a "new version" (per comment 5) is appropriate, that is QA'd (with portlint, poudriere at least), in particular given "may be more difficult to port to FreeBSD" (has it been ported?), OR ...
>
> - Confirmation, or not, from reporter/author whether or not the existing patch is (still) appropriate in its existing form without further changes for inclusion in ports, per question in comment 3 and response in comment 4

[7 more replies]

closed 2021-01-09 by alexpevzner