alexpevzner/sane-airscan #112: Add HP LaserJet 200 colorMFP M276n

probonopd, 2020-12-29:
Possibly we can add

```
| HP LaserJet MFP M276n              | No                        | Yes                       |
```

Test is pending #3.

By the way, I noticed that the names of some HP printers are variations of how the printers call themselves. This one calls itself the `HP LaserJet 200 colorMFP M276n` on Zeroconf and `HP LaserJet 200 color MFP M276n` on the physical device.

alexpevzner, 2020-12-29:
Yes, I usually prefer the name as device calls itself, in a hope that users will be able to recognize their variant, instead of listing all possible variants of the same device (such a list is hard to obtain, but most likely, if some of the devices in series works, others will work too).

probonopd, 2020-12-29:
Test successful on FreeBSD both from glass and ADF.
Only thing is that there is no way to scan in black/white, only color and grayscale.
But this might be a limitation of the device and/or xsane.

alexpevzner, 2020-12-29:
Support of 1-bit B&W scan adds dependency on `libtiff` and PDF decoder, which I believe is too high price for such a small increment of functionality.

If there as a high demand of B&W scan, I'd better implement it in pure software (by requesting grayscale and resampling the resulting image). At least, it will work uniformly on any hardware.

I've added your device to the list, so I close this issue. Lets continue FreeBSD discussion under #111

closed 2020-12-29 by alexpevzner