alexpevzner/sane-airscan #113: A Canon LiDE 300

debiantriage, 2021-01-09:
Здравствуйте Алекс,

С наступившим Новым Годом.

https://lists.debian.org/debian-user/2021/01/msg00277.html

> 'airscan:e0:Canon LiDE 300 (USB)' is a eSCL Canon LiDE 300 (USB) ip=127.0.0.1

I think that this device works with sane-airscan but please reopen if my judgement is unsound.

I am bothered by pixma working with ipp-usb active. Is this a possibility?

Have we reached 100 devices yet?

closed 2021-01-09 by debiantriage

alexpevzner, 2021-01-09:
Дорогой Брайан,

поздравляю с наступившим Новым Годом и Рождеством! :-)

Yes, with LIDE 300 we have exactly 100 devices.

Coexistence of IPP over USB with legacy drivers makes me concerned, but for now I don't see any good solution.

IPP-over-USD daemon claims all IPP-over-USB interfaces, and if these interfaces are shared with some legacy USB function (printing or scanning), the corresponding interface becomes unavailable for legacy driver. If these interfaces are shared with essential legacy functions, depend on a particular device (wired on its hardware, roughly speaking) and cannot be changed outside.

Ubuntu 20.04 comes with IPP-over-USB enabled by default, and what makes things worse, it comes with `ippusbxd` instead of `ipp-usb` and with `sane-escl` instead of `sane-airscan`. So some users experience regression: device, that used to work with legacy print/scan driver, becomes non-functional after system upgrade.

The common simple advise for these users is to uninstall the `ippusbxd` package, which restores the old (before-upgrade) behaviour, but honestly speaking, I don't like this advice, because I believe that moving forward is strategically better, that moving back (though many end users will disagree).

Situation will become better with Ubuntu 20.10, which replaces `ippusbxd` with `ipp-usb` and (as far as I know) comes with `sane-airscan`.  I believe, `sane-airscan` + `ipp-usb` combination will work correctly in 99.9% cases and can easily be fixed in remaining 00.1% cases as soon as I'll receive a bug report. But there still can be problems with printing, if for some device legacy driver works better, that "driverless" stack.

Ideally, USB interfaces should be claimed only when actually used, but it requires run-time cooperation between CUPS, SANE and ipp-usb, so it is hard to implement.

debiantriage, 2021-01-09:
Here's to the next 100, Alex. Thank you for your work and your commitment to fixing bugs.

I am not familiar with Ubuntu's release process but, to be fair, `sane-airscan` and `ipp-usb` may have come along too late to be included in 20.04. However, AFAICT, there wasn't any documentation to point to `ippusbxd` as the new manager of a USB bus. The fact that it did not work for many users didn't help the situation either.

The next edition of Mint will not use `ippusbxd`, but it will not be replaced with `ipp-usb`. I argued against that. However they do document the new situation:

https://forums.linuxmint.com/viewtopic.php?f=51&t=339269

> ...I believe that moving forward is strategically better, 

Agreed. So much so that my support for users wanting to use legacy printer/scanner drivers with a modern device is very, very limited.

Cheers,

Brian.

alexpevzner, 2021-01-09:
`ippusbxd` was existent for years (probably, since Ubuntu 18.04 or so), but it used to be an optional package. Ubuntu 20.04 began installing it by default. It was a mistake (ipp-usb was already in existence at that time, but not yet included into official Ubuntu repo).

Current (git) version of `sane-escl` is relatively stable (though `sane-airscan` is better anyway :)), but it comes with SANE bundle and logistics is so slow, so most of `sane-escl` users run very old (and buggy) version. While logistics of `sane-airscan`, that comes as independent package, works much faster. On Fedora, I typically have `sane-airscan` updated from the system repo with ~2 weeks delay after my own release.

This combination of events made a bad reputation of Linux driverless printing/scanning stack, which will delay propagation of this stack to mass users for year or two. Which is not good, because in most (but not all) cases this stack works MUCH better, that legacy drivers.

BTW, Fedora never shipped `ippusbxd` even as an optional package, and Fedora's build of SANE comes without `sane-escl`, so Fefora users got driverless stack a bit later that Ububtu users, but with right components from the very beginning.