alexpevzner/sane-airscan #114: Allow setting the brightness and the contrast

bertogg, 2021-01-12:
Hi,

I have an HP OfficeJet Pro 9010 series. The scanner works out of the box with sane-airscan, but I noticed when scanning photos that the resulting images are too bright and the colors are often burned.

The scanner allows adjusting the brightness level but it seems that the default value is too high. Lowering it produces better results in my case. Unfortunately this backend does not seem to allow setting that value.

It is possible however to do it using eSCL, ScannerCapabilities reports this:

```
	<scan:BrightnessSupport>
		<scan:Min>0</scan:Min>
		<scan:Max>2000</scan:Max>
		<scan:Normal>1000</scan:Normal>
		<scan:Step>1</scan:Step>
	</scan:BrightnessSupport>
	<scan:ContrastSupport>
		<scan:Min>0</scan:Min>
		<scan:Max>2000</scan:Max>
		<scan:Normal>1000</scan:Normal>
		<scan:Step>1</scan:Step>
	</scan:ContrastSupport>
	<scan:ThresholdSupport>
		<scan:Min>0</scan:Min>
		<scan:Max>255</scan:Max>
		<scan:Normal>128</scan:Normal>
		<scan:Step>1</scan:Step>
	</scan:ThresholdSupport>
```

And I verified that manually adding e.g. `<scan:Brightness>500</scan:Brightness>` to the query does the job.

alexpevzner, 2021-01-29:
Hi,

sorry for long delay in response.

sane-airscan implements brightness/contrast options sinse release 0.99.20, though it does it in software, If it doesn't work for you, please give me more detailed explanation, what is wrong with these options.

bertogg, 2021-01-29:
I'm attaching two scans of the same image, one with the brightness level set to 500 and the other set to 1000 (the default). I'm setting the level on the scanner itself, I'm not doing any post-processing in software.

For context, what you see in the image is the sea with some scattered clouds.

With brigthness=1000 the clouds are far too bright and the details are lost, you can also see that in the histogram. There's no amount of post-processing that will restore that.

The default brightness level is however ok for letters and other documents, so I don't think there's a good default. I change the brightness level depending on what I'm going to scan.

[image]
[image]

bertogg, 2021-01-29:
Another issue is that this scanner can send the results in either raw format or with jpeg compression. sane-airscan uses always the latter but that can produce visible artifacts (see the 'T' letter on the images that I uploaded earlier). If you think that selecting the compression type is a feature worth adding I can open another ticket.

alexpevzner, 2021-01-30:
OK, I understood. Will see what can I do with it. As my device doesn't support brightness/contrast settings in hardware, I will have to rely on you in testing.

> Another issue is that this scanner can send the results in either raw format or with jpeg compression

Can you try to manually select raw format (similar to how you've manually set brightness)? sane-airscan will not unpack the returned image (will complain I/O error), but it can save it into the protocol trace file. I want to see this image file, to understand, how to unpack it.

To enable protocol trace, uncomment the following lines in the `/etc/sane.d/airscan.conf` file:

```
[debug]
trace = ~/airscan/trace
enable = true
```

Protocol traces will be saved to the `~/airscan/trace` directory, which will be created automatically. `.log` files contains a very detailed protocol log (much more detailed that log, that goes to console), and `.tar` files contains images.

And I want to see the whole ScannerCapabilities too, not only the fragment you've attached before.

bertogg, 2021-01-30:
Hi, I haven't had the time to rebuild sane-airscan, but I did a quick test by simply using CURL to send this to the scanner:

```
<?xml version="1.0" encoding="UTF-8"?>
<scan:ScanSettings xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03">
  <pwg:Version>2.0</pwg:Version>
  <pwg:ScanRegions>
    <pwg:ScanRegion>
      <pwg:ContentRegionUnits>escl:ThreeHundredthsOfInches</pwg:ContentRegionUnits>
      <pwg:XOffset>1320</pwg:XOffset>
      <pwg:YOffset>450</pwg:YOffset>
      <pwg:Width>300</pwg:Width>
      <pwg:Height>500</pwg:Height>
    </pwg:ScanRegion>
  </pwg:ScanRegions>
  <pwg:InputSource>Platen</pwg:InputSource>
  <scan:ColorMode>RGB24</scan:ColorMode>
  <pwg:DocumentFormat>application/octet-stream</pwg:DocumentFormat>
  <scan:DocumentFormatExt>application/octet-stream</scan:DocumentFormatExt>
  <scan:XResolution>150</scan:XResolution>
  <scan:YResolution>150</scan:YResolution>
  <scan:Brightness>700</scan:Brightness>
</scan:ScanSettings>
```

I can turn the output file into a Netpbm image simply by adding a header (the image is 150x250 pixels):

`echo 'P6 150 250 255' > image.pnm ; cat image.bin >> image.pnm`

I'm attaching the raw image data and the capabilities.

[files.zip](https://github.com/alexpevzner/sane-airscan/files/5898242/files.zip)