alexpevzner/sane-airscan #116: HP localhost quirk handling

fabled, 2021-01-18:
Commit cc19e6c57d3001e7c9baf7aff401f972322a1c5e whitelisted HP Color LaserJet FlowMFP M578  for the "Host: localhost" hack. However, I am hopeful that this could be generalized with looking the HTTP reply's `Server: HP_Compact_Server` header, or perhaps make the MakeAndModel list configurable.

I have tested the following HP devices so far (I hope to get access to lab with more devices, but not sure if it works due to covid restrictions):
                                                             
Model | Server |Status
--- | --- | ---
Color LaserJet Pro MFP M479fdn | nginx | OK
HP Color LaserJet FlowMFP M578 | HP_Compact_Server | OK-with-quirk
HP ScanJet Pro 4500 fn1 | localhost:60000 | OK

Currently it would seem that the Server response is fairly reliable way to detect it, but this is of course very small subset.

Alternatively, perhaps for HP devices, the `PUT` request could be automatically retried with host quirk applied if `409` response is received?

fabled, 2021-01-27:
Visited lab today. The Main comment's table is updated now. It seems that on all "enterprise" and/or "managed" devices the hack is needed (so far with one exception). This included some models that are still "pre-release".

After speaking to HP person, the one model to detect these printers would be to match model to include "HP", but *not* include "Pro". The "Pro" indicates different product series with completely different firmware base. Would this kind of check be acceptable?

Alternately since it seems most new devices need the hack, I'd appreciate a way to configure whether or not to enable this.

Any ideas?

alexpevzner, 2021-01-28:
Added check for `Server: HP_Compact_Server`. Please, test.

P.S. It would be **very** interesting to know, for what reason HP firmware requires this hack...

fabled, 2021-01-28:
> Added check for `Server: HP_Compact_Server`. Please, test.

Thank you very much for this. Looks good! It will be some time until I get access again to these devices, but will confirm once possible.

> P.S. It would be **very** interesting to know, for what reason HP firmware requires this hack...

I have inquired about this. Let's see if we get any response back.

alexpevzner, 2021-01-28:
> Thank you very much for this. Looks good!

So you think its good enough to issue the next release, from the enterprise HP support perspective?

> I have inquired about this. Let's see if we get any response back.

Honestly, I don't think we will get an answer :-(

It looks like very naive and incorrect attempt to limit eSCL functionality to IPP over USB only (IPP over USB sets `Host: localhost` for its own reasons). There are may be even some option behind many levels of menus of the web interface, that enables eSCL access from the network, and this requirement goes away. I don't have an enterprise HP device to test this hypothesis.

alexpevzner, 2021-01-29:
Fixed in 0.99.24

closed 2021-01-29 by alexpevzner