alexpevzner/sane-airscan #117: Does not build on clang/i386

HenryHu, 2021-01-18:
Line 233 of airscan-bmp.c says:
    if (bmp->next_line == labs(bmp->info_header.biHeight)) {

Since next_line is unsigned int, and labs(biHeight) is long, they have different signedness, and this triggers a warning in clang (-Wsign-compare, compare between signed and unsigned). 
Since -Wall -Werror is specified, all warnings are considered as errors. Converting next_line to long first can resolve the issue.

alexpevzner, 2021-01-19:
Fixed. Strange that it was never reported before...

closed 2021-01-19 by alexpevzner

HenryHu, 2021-01-20:
Thanks. I currently include a patch for this for the port for 0.99.23. If you release a new version later, I can update the port to remove the patch.

alexpevzner, 2021-01-20:
OK.

Next version will bring workaround for HP Color LaserJet FlowMFP M578. There are related issues in the `ipp-usb` project (https://github.com/OpenPrinting/ipp-usb), I plan to release both projects together, when these issues will be fixed.

I'll notify you, when it will happen.

alexpevzner, 2021-01-29:
Hi,

I've just released 0.99.24, which contains this fix.

commit 25cd6410 (2021-01-19): Fixed clang/i386 build (fixes #117)