alexpevzner/sane-airscan #120: blacklisting facilities

motorBMW, 2021-01-26:
I remember you were asking for blacklisting facilities. I was thinking some time about it and can probably add something like this:
```
[blacklist]
model = "Brother*"
name = "*LaserJet*"
ip = 192.168.0.1/24
```

_Originally posted by @alexpevzner in https://github.com/alexpevzner/sane-airscan/issues/97#issuecomment-725979875_

motorBMW, 2021-01-26:
Hi, Alexander. I created the issue to highlight the need to create a blacklist. I would be very grateful for your help!

alexpevzner, 2021-01-29:
OK. Sorry for delay in answering. Probably, I'll add it in 0.99.25

motorBMW, 2021-03-04:
Hi, Alexander!

I really need your help! Please create a blacklist!

alexpevzner, 2021-03-04:
You may test it for now, but please note, that it is not well-tested on my side yet :-)

The syntax is as proposed above. model/name supports glob-style wildcards. Netmask is optional, if omitted, exact address match is assumed.

motorBMW, 2021-03-05:
The filter works fine, the necessary devices are excluded from the list of available devices. Now I look forward to your assembled package to distribute to the devices. Thank you very much for your work, Alexander!

alexpevzner, 2021-04-13:
Fixed in 0.99.25, sorry for delay with release

closed 2021-04-13 by alexpevzner