alexpevzner/sane-airscan #121: A Samsung M2070

debiantriage, 2021-01-31:
https://forums.linuxmint.com/viewtopic.php?f=51&t=341527

> ty=Samsung M2070 Series

Nicely clearcut, but why two IPs for the device?

Incidentally, at some point in the past you wondered about the effort needed to implement WSD support. I would rather deal with sane-airscan than the ULD! Thanks for the effort.

Regards,

Brian.

closed 2021-01-31 by debiantriage

alexpevzner, 2021-01-31:
> Nicely clearcut, but why two IPs for the device?

Because this device was discovered twice, on two different IP addresses.

These addresses were made visible upon request of people who uses sane-airscan in a large cooperative network with a lot of similar scanners around, to help identify the particular device among many others of the same model and similar auto-assigned name. See #97 for details (though half of discussion is on Russian there).

>  I would rather deal with sane-airscan than the ULD!

Nice to head, but... what is ULD? :-)

debiantriage, 2021-01-31:
ULD - Unified Linux Driver. Non-free software offered by Samsung for printing and scanning.