alexpevzner/sane-airscan #122: ipp-usb and Debian 10

debiantriage, 2021-02-03:
https://download.opensuse.org/repositories/home:/pzz/Debian_10/

offers ipp-usb binaries for Debian 10. In order to work, they need a version of avahi that is not in buster (it has 0.7-4+b1). I was reminded of this when helping a user:

https://answers.launchpad.net/hplip/+question/695329

ipp-usb on Ubuntu bionic works because avahi has been patched:

> avahi (0.7-3ubuntu2) bionic; urgency=low

  * local-only-services-support.patch: Added support for advertising
    local-only services on the loopback ("lo", "localhost") device. This
    is especially needed for making CUPS and cups-browsed auto-discover
    IPP-over-USB printers using ippusbxd (LP: #1736757, Upstream issue #125).

 -- Till Kamppeter <till.kamppeter@gmail.com>  Tue,  9 Jan 2017 21:35:51 -0200

alexpevzner, 2021-02-03:
`ipp-usb` will work with unpatched Avahi, with the following change on its configuration file (`/etc/ipp-usb/ipp-usb.conf`):
```
interface = all
```

With this change, it will expose device to the network, not only to `localhost`, and will announce it appropriately, so unpatched Avahi will accept it.

debiantriage, 2021-02-03:
I was aware of that option, so really should have been able to work it out for myself. Many thanks.

I will return to the user I am dealing with and see how we go on. Anyway, it works for me.

debiantriage, 2021-02-08:
The user solved his issue by using a wireless. He has since not returned. AFAIAC, my query has been addressed.

closed 2021-02-08 by debiantriage