alexpevzner/sane-airscan #126: Exposing Additional Scanner Capabilities

kyvmo, 2021-03-11:
We need access to additional scanner capability information for Epson scanners. The ADF for Epson (and potentially others) can be justified in multiple ways, rather than the currently assumed top-left justification. 
Can the justification capability under the ADF tag be exposed? Get Capabilities from the scanner should include this information, is it possible to add it as a loadable option? 
Sane driver can return informational parameters- is that a reasonable way to go about it?

Here's an example of the xml returned by the scanner that contains the necessary info (with the other options excluded):
<scan:Adf>
 .
.
.
    <scan:Justification>
      <pwg:XImagePosition>Right</pwg:XImagePosition>
      <pwg:YImagePosition>Top</pwg:YImagePosition>
    </scan:Justification>
 .
.
.
  </scan:Adf>

alexpevzner, 2021-03-12:
Hi,

I have few questions.

Is it read-only information, or it is assumed to be settable?

The natural way to expose some information from SANE driver is to implement an additional read-only option. If Epson proprietary driver already implements something similar, I'd prefer to be compatible with it in this aspect. Do you know, does Epson driver implements something like this?

I'd like to to see the whole Capabilities XML, not just this small fragment.

kyvmo, 2021-03-12:
Thank you for your fast response!

It's read-only information. It appears that the Epson driver does implement something similar - when requesting all options on the device (`scanimage -A`), it lists `--adf-orientation <int>` under `Buttons`. 

Here's the entire Capabilities XML:
<?xml version="1.0" encoding="UTF-8"?>
<scan:ScannerCapabilities xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm">
  <pwg:Version>2.63</pwg:Version>
  <pwg:MakeAndModel>EPSON XP-7100 Series</pwg:MakeAndModel>
  <pwg:SerialNumber>583544353135303853</pwg:SerialNumber>
  <scan:UUID>cfe92100-67c4-11d4-a45f-389d92b27db4</scan:UUID>
  <scan:AdminURI>http://100.107.108.190:80/PRESENTATION/BONJOUR</scan:AdminURI>
  <scan:IconURI>https://100.107.108.190:443/PRESENTATION/AIRPRINT/PRINTER_128.PNG</scan:IconURI>
  <scan:Platen>
    <scan:PlatenInputCaps>
      <scan:MinWidth>96</scan:MinWidth>
      <scan:MaxWidth>2550</scan:MaxWidth>
      <scan:MinHeight>96</scan:MinHeight>
      <scan:MaxHeight>3510</scan:MaxHeight>
      <scan:MaxScanRegions>1</scan:MaxScanRegions>
      <scan:SettingProfiles>
        <scan:SettingProfile>
          <scan:ColorModes>
            <scan:ColorMode>BlackAndWhite1</scan:ColorMode>
            <scan:ColorMode>Grayscale8</scan:ColorMode>
            <scan:ColorMode>RGB24</scan:ColorMode>
          </scan:ColorModes>
          <scan:DocumentFormats>
            <pwg:DocumentFormat>application/pdf</pwg:DocumentFormat>
            <pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat>
            <scan:DocumentFormatExt>application/pdf</scan:DocumentFormatExt>
            <scan:DocumentFormatExt>image/jpeg</scan:DocumentFormatExt>
          </scan:DocumentFormats>
          <scan:SupportedResolutions>
            <scan:DiscreteResolutions>
              <scan:DiscreteResolution>
                <scan:XResolution>100</scan:XResolution>
                <scan:YResolution>100</scan:YResolution>
              </scan:DiscreteResolution>
              <scan:DiscreteResolution>
                <scan:XResolution>200</scan:XResolution>
                <scan:YResolution>200</scan:YResolution>
              </scan:DiscreteResolution>
              <scan:DiscreteResolution>
                <scan:XResolution>300</scan:XResolution>
                <scan:YResolution>300</scan:YResolution>
              </scan:DiscreteResolution>
              <scan:DiscreteResolution>
                <scan:XResolution>600</scan:XResolution>
                <scan:YResolution>600</scan:YResolution>
              </scan:DiscreteResolution>
              <scan:DiscreteResolution>
                <scan:XResolution>1200</scan:XResolution>
                <scan:YResolution>1200</scan:YResolution>
              </scan:DiscreteResolution>
            </scan:DiscreteResolutions>
          </scan:SupportedResolutions>
          <scan:ColorSpaces>
            <scan:ColorSpace scan:default="true">sRGB</scan:ColorSpace>
          </scan:ColorSpaces>
          <scan:CcdChannels>
            <scan:CcdChannel scan:default="true">NTSC</scan:CcdChannel>
          </scan:CcdChannels>
          <scan:BinaryRenderings>
            <scan:BinaryRendering>Threshold</scan:BinaryRendering>
            <scan:BinaryRendering scan:default="true">Halftone</scan:BinaryRendering>
          </scan:BinaryRenderings>
          <scan:ContentTypes>
            <pwg:ContentType>Photo</pwg:ContentType>
            <pwg:ContentType>Text</pwg:ContentType>
[... 173 lines cut ...]
  <scan:CompressionFactorSupport>
    <scan:Min>1</scan:Min>
    <scan:Max>3</scan:Max>
    <scan:Normal>2</scan:Normal>
    <scan:Step>1</scan:Step>
  </scan:CompressionFactorSupport>
  <scan:ContrastSupport>
    <scan:Min>-100</scan:Min>
    <scan:Max>100</scan:Max>
    <scan:Normal>0</scan:Normal>
    <scan:Step>1</scan:Step>
  </scan:ContrastSupport>
  <scan:SharpenSupport>
    <scan:Min>0</scan:Min>
    <scan:Max>3</scan:Max>
    <scan:Normal>2</scan:Normal>
    <scan:Step>1</scan:Step>
  </scan:SharpenSupport>
  <scan:ThresholdSupport>
    <scan:Min>0</scan:Min>
    <scan:Max>255</scan:Max>
    <scan:Normal>186</scan:Normal>
    <scan:Step>1</scan:Step>
  </scan:ThresholdSupport>
</scan:ScannerCapabilities>

alexpevzner, 2021-03-13:
Hi,

`adf-orientation` is documented, for example, in `man sane-pixma`. And according to it, it is a read-only integer parameter with values: 1 = Portrait, 2 = Landcape.

However, I don't quite understand, how the following:
```
<scan:Justification>
    <pwg:XImagePosition>Right</pwg:XImagePosition>
    <pwg:YImagePosition>Top</pwg:YImagePosition>
</scan:Justification>
```
should be mapped to `adf-orientation` in all cases.

In another words, what another possible values can be found there?

kyvmo, 2021-03-15:
For Epson specifically, there are 3 possible values for the XImagePosition Justification: 'Left', 'Right', and 'Center'. 
This 'Justification' targets how the ADF interprets where the content falls on the page it's scanning, as opposed to the orientation of portrait/landscape. 
'Left' is the standard justification value, but for some scanners (including the one we are working on), the value is set to one of the two other options. We need to be able to check that specifically (I don't think the YImagePosition is important at the moment, but if it's possible to access all of the justification data that may be useful in the future).

alexpevzner, 2021-03-15:
When does this information change?

kyvmo, 2021-03-15:
I think it comes preset with the scanner, I can't find any information on modifying the values at the moment.

kyvmo, 2021-03-22:
Hello!

Just checking in, is this a feature that's able to be exposed by sane-airscan? Any other questions I can answer for you? 

Thanks!

alexpevzner, 2021-06-01:
Hi @kyvmo,

sorry for being slow with this issue, but I have finally incorporated your patch, refactored/fixed it here and there, and added an `adf-justification-y` option just for consistency.

Please, retest the current version, so I can proceed with the next release.