alexpevzner/sane-airscan #128: An EPSON Stylus SX535WD and a Canon TS6400 series

debiantriage, 2021-03-14:
https://forums.linuxmint.com/viewtopic.php?f=51&t=344803

> ty=EPSON Epson Stylus SX535WD

I nearly slipped up on this one (forgetting about WSD) but the user rescued me! No URI given yet but surely the device can only be scanning with WSD?

=============================================

https://forums.linuxmint.com/viewtopic.php?f=51&t=344922

> ty=Canon TS6400 series

`simple-scan "airscan:e0:Canon TS6400 series"` works. Is my idea to substitute w1 for e0 invalid for testing for WSD working?

=============================================

BTW. In #125 there are two devices. Did you miss the second one? Perhaps I shouldn't have added it at a later date?

Regards,

Brian.

alexpevzner, 2021-03-14:
> Is my idea to substitute w1 for e0 invalid for testing for WSD working?

No, it doesn't work this way. To test WSD, user should enable manual protocol selection in the `/etc/sane.d/airscan.conf`:
```
[options]
  protocol = manual
```

It will cause multi-protocol devices to appear twice in the list, once for ESCL and once for WSD, so user can choose.

> Did you miss the second one?

Looks so. Now fixed.

debiantriage, 2021-03-14:
> No, it doesn't work this way. To test WSD, user should enable manual protocol selection in the `/etc/sane.d/airscan.conf`:
>
> ```
> [options]
>   protocol = manual
> ```
>
> It will cause multi-protocol [...]

Thanks. I can now confirm my HP ENVY 4500 also scans with WSD.

debiantriage, 2021-03-14:
Just for completeness for the EPSON Epson Stylus SX535WD the user reports:

> $ scanimage -L
device `airscan:w0:EPSON Epson Stylus SX535WD' is a WSD EPSON Epson Stylus SX535WD ip=192.168.0.8

> $ airscan-discover
[devices]
  Epson Stylus SX535WD = http://192.168.0.80:80/WDP/SCAN, WSD

This is a device from about 2011. Incorporating WSD into sane-airscan was a good decision. Thanks for the work involved.

closed 2021-03-14 by debiantriage