alexpevzner/sane-airscan #130: sane-airscan no longer finds devices?

Xavron, 2021-03-16:
Really confused here. Another OS shows scanimage -L finding the mfc device but there doesn't seem to be airscan as a package. This other one **was** finding the mfc scanner before and no longer is but if I try to remove airscan there's no longer anything as far as network capability.

So I assume some problem is happening with airscan. Either way, airscan & airscan-discover can not longer find the mfc device with this OS.

Same version of just about everything sane & sane-airscan but one month it worked, another month it doesn't. So either another package broke it or another package changed and airscan broke?

zeroconf.tar is empty.

zeroconf.log
00:00:00.000: zeroconf: zeroconf configuration:
00:00:00.000: zeroconf:   discovery    = enable
00:00:00.000: zeroconf:   model        = network
00:00:00.000: zeroconf:   protocol     = auto
00:00:00.000: zeroconf:   ws-discovery = fast
00:00:00.001: MDNS: AVAHI_CLIENT_S_RUNNING
00:00:00.003: WSDD: bind(0.0.0.0:3702): Address already in use
00:00:00.003: zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
00:00:00.003: zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
00:00:00.004: zeroconf: ZEROCONF_MDNS_HINT: initial scan finished

Maybe you have an idea as to what is going on?

Is this a thing or is it normal? WSDD: bind(0.0.0.0:3702): Address already in use

Thanks

Xavron, 2021-03-16:
Also to add that the mfc device is set in the OS printer section and prints fine from this OS. The scanner by all means as far as I can see should still be visible and working.

alexpevzner, 2021-03-16:
Looks like the problem is in network connection. Can you ping your printer from this computer?

alexpevzner, 2021-03-16:
Also, what `ifconfig -a` prints?

Xavron, 2021-03-16:
I think that looks fine.

ping -c 2 192.168.1.3
PING 192.168.1.3 (192.168.1.3) 56(84) bytes of data.
64 bytes from 192.168.1.3: icmp_seq=1 ttl=255 time=0.470 ms
64 bytes from 192.168.1.3: icmp_seq=2 ttl=255 time=0.462 ms

--- 192.168.1.3 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1033ms
rtt min/avg/max/mdev = 0.462/0.466/0.470/0.004 ms

ifconfig -a
enp37s0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.183  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::e9ed:1fb6:4575:2eed  prefixlen 64  scopeid 0x20<link>
        ether 4c:cc:6a:fc:ee:d7  txqueuelen 1000  (Ethernet)
        RX packets 7072  bytes 4833345 (4.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6090  bytes 820129 (800.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 84  bytes 8907 (8.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 84  bytes 8907 (8.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
        ether 52:54:00:e7:80:81  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0-nic: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether 5e:5e:e3:ac:87:53  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

alexpevzner, 2021-03-16:
What OS do you use?

Xavron, 2021-03-16:
Fedora 33 is having the problem

Xavron, 2021-03-16:
Stepped through airscan code using gdb and it almost looks like airscan is being blocked from finding it though I'm not familiar with the code here and may have missed something.

Didn't do anything directly related to airscan or even indirectly to the printer previously. But I did do a bunch of things not related but that touched the firewall for example but then again I disabled that and still nothing.

Running low on ideas.

alexpevzner, 2021-03-16:
Can you somehow guess, what process can be bound to UDP port 3702?

alexpevzner, 2021-03-16:
Probably, `netstat -plun|grep 3702`

Xavron, 2021-03-16:
Stepping through airscan with gdb I believe reveals that it cannot find it hence the empty device list.

> Can you somehow guess, what process can be bound to UDP port 3702?

I can try to find out.

For the moment I ran test-zeroconf out of curiosity and it only shows example
zeroconf:     https://192.168.1.102:9095/eSCL/
zeroconf:   device:    0001 (found)

Is that what I'm thinking it is? Don't even see a .102 on the network.

alexpevzner, 2021-03-16:
test-zeroconf only tests internal algorithms used for processing discovery results. It doesn't perform any network activity, so all addresses it "discovers" come from the test data

Xavron, 2021-03-16:
> test-zeroconf only tests internal algorithms used for processing discovery results. It doesn't perform any network activity, so all addresses it "discovers" come from the test data

okay thanks. I'm trying but learning :-)

Xavron, 2021-03-16:
> test-zeroconf only tests internal algorithms used for processing discovery results. It doesn't perform any network activity, so all addresses it "discovers" come from the test data

okay thanks. I'm trying but learning :-)

udp        0      0 192.168.122.1:3702      0.0.0.0:*                           1427/python3        
udp        0      0 239.255.255.250:3702    0.0.0.0:*                           1427/python3        
udp        0      0 192.168.1.183:3702      0.0.0.0:*                           1427/python3        
udp        0      0 239.255.255.250:3702    0.0.0.0:*                           1427/python3        
udp6       0      0 fe80::e9ed:1fb6:45:3702 :::*                                1427/python3        
udp6       0      0 ff02::c:3702            :::*                                1427/python3

Wonder why python is attached to that?

alexpevzner, 2021-03-16:
Now this is interesting to know what script does it run. Probably, `ps ax | grep 1427` will give us an answer (1427 is the PID from the previous command)

Xavron, 2021-03-16:
1427 ?          Ss     0:00 python3 /usr/bin/wsdd --shortlog
6737 pts/0    S+     0:00 grep --color=auto 1427

So this was used to get Windows to show up on networking in Linux. I will try disabling it. Didn't even imagine that would happen lol.

Xavron, 2021-03-16:
Yes, that did it. Conflicts :-(

Thanks a ton :+1:  :+1:

alexpevzner, 2021-03-16:
Seems that its author forgot to tell `SO_REUSEADDR` on listening multicast socket. Its better to fix there.

Xavron, 2021-03-16:
> Seems that its author forgot to tell `SO_REUSEADDR` on listening multicast socket. Its better to fix there.

I might let them know. Right now just want to shake the frustration :-)

Thanks for hanging in there with me. In a way its good to know other projects kill your work.

Xavron, 2021-03-16:
Happily solved :-)

closed 2021-03-16 by Xavron

alexpevzner, 2021-03-16:
If you'll file a bug against wsdd, please mention me there, so I will receive notifications from github

Xavron, 2021-03-17:
> If you'll file a bug against wsdd, please mention me there, so I will receive notifications from github

You got it. Giving you two notices. Starting it right now :-)

Xavron, 2021-03-17:
Think I figured it out. Looks like the tutorial I followed might be the cause. wsdd has a distro provided package that I didn't see previously and I believe is working alright that way.