alexpevzner/sane-airscan #131: Various defects found by Coverity

alexmurray, 2021-03-17:
As part of a security review of sane-airscan for Ubuntu, I ran Coverity across the current `0.99.24-1` version in Ubuntu 21.04 and it found a number of defects - for each defect, the line containing the actual defect is listed first, with a `Type` then below that is an explanation of how this is believed to occur on preceding lines.

```
airscan-device.c:1781
  Type: Dereference before null check (REVERSE_INULL)

airscan-device.c:1770:
  deref_ptr: Directly dereferencing pointer "len_out".
airscan-device.c:1781:
  check_after_deref: Null-checking "len_out" suggests that it may be null, but it has already been dereferenced on all paths leading to the check.

airscan-escl.c:450
  Type: Branch past initialization (PW.BRANCH_PAST_INITIALIZATION)

airscan-escl.c:450:
  1. branch_past_initialization: transfer of control bypasses initialization of:
airscan-escl.c:450:
  2. name_at_decl_position: variable "src_ok" (declared at line 506)

airscan-escl.c:455
  Type: Branch past initialization (PW.BRANCH_PAST_INITIALIZATION)

airscan-escl.c:455:
  1. branch_past_initialization: transfer of control bypasses initialization of:
airscan-escl.c:455:
  2. name_at_decl_position: variable "src_ok" (declared at line 506)

airscan-escl.c:500
  Type: Branch past initialization (PW.BRANCH_PAST_INITIALIZATION)

airscan-escl.c:500:
  1. branch_past_initialization: transfer of control bypasses initialization of:
airscan-escl.c:500:
  2. name_at_decl_position: variable "src_ok" (declared at line 506)

airscan-http.c:1275
  Type: Uninitialized pointer read (UNINIT)

airscan-http.c:1261:
  1. var_decl: Declaring variable "parser" without initializer.
airscan-http.c:1265:
  2. path: Condition "skip_line", taking true branch.
airscan-http.c:1267:
  3. path: Condition "s", taking true branch.
airscan-http.c:1275:
  4. uninit_use_in_call: Using uninitialized value "parser.data" when calling "http_parser_init".
http_parser.c:2220:
  4.1. read_parm_fld: Reading a parameter field.

airscan-netif.c:293
  Type: Infinite loop (INFINITE_LOOP)

airscan-netif.c:293:
  loop_top: Top of the loop.
airscan-netif.c:310:
  loop_bottom: Bottom of the loop.
airscan-netif.c:293:
  loop_condition: If "halves[1] != NULL" is initially true then it will remain true.

airscan-os.c:39
  Type: Large stack use (STACK_USE)

airscan-os.c:39:
  stack_use_local_overflow: Local variable "buf" uses 16384 bytes of stack space, which exceeds the maximum single use of 10000 bytes.

airscan-os.c:109
  Type: String not null terminated (STRING_NULL)

airscan-os.c:79:
  1. string_null_argument: Function "readlink" does not terminate string "*os_progname_buf". [Note: The source code implementation of the function has been overridden by a builtin model.]
[... 394 lines cut ...]
airscan-http.c:610:
  2.16.13. path: Falling through to end of if statement.
airscan-http.c:614:
  2.16.14. return_alloc: Returning allocated memory "uri".
airscan-http.c:837:
  2.17. return_alloc_fn: Directly returning storage allocated by "http_uri_new".
test-uri.c:138:
  3. var_assign: Assigning: "uri_rel" = storage returned from "http_uri_new_relative(uri_base, ref, false, false)".
test-uri.c:139:
  4. path: Condition "uri_rel == NULL", taking false branch.
test-uri.c:143:
  5. noescape: Resource "uri_rel" is not freed or pointed-to in "http_uri_str".
airscan-http.c:858:
  5.1. noescape: "http_uri_str(http_uri *)" does not free or save its parameter "uri".
test-uri.c:144:
  6. path: Condition "strcmp(s, expected)", taking false branch.
test-uri.c:147:
  7. leaked_storage: Variable "uri_rel" going out of scope leaks the storage it points to.

test.c:51
  Type: Large stack use (STACK_USE)

test.c:51:
  stack_use_local_overflow: Local variable "buf" uses 65536 bytes of stack space, which exceeds the maximum single use of 10000 bytes.
```

alexpevzner, 2021-03-17:
Hi Alex,

thank you for your report. I've reviewed all these defects. Though none of them are critical (i.e., none of them could really lead to unsafe behavior), I've fixed as much as possible.

Now few comments.

All `CHECKED_RETURN` warnings come from places where return value is purposely ignored. Either because the called function can never fail, or because error is not important for some reason. I've added explicit cast of returned values to `void` in hope it will silence these warnings; I also wrote short comment everywhere when error is ignored, explaining while error is ignored here.

The `STACK_USE` warnings come from a couple of places where I use 16K temporary buffer on a stack, once in initialization path and once in test. I believe there is no real problem here, user space thread stack is much larger in Linux, so it's better just to tune Coverity stack size threshold.

Errors like these (repeated in few similar places):
```
airscan-netif.c:293
  Type: Infinite loop (INFINITE_LOOP)
```
are false positive. Seems that loop invariant is a bit more complex that the static analyzer can understand.

This is false positive:
```
airscan-wsd.c:385
  Type: Unused value (UNUSED_VALUE)

airscan-wsd.c:400:
  value_overwrite: Overwriting previous write to "tmp" with value from "max_hei".
airscan-wsd.c:385:
  assigned_value: Assigning value from "min_wid" to "tmp" here, but that stored value is overwritten before it can be used.

```

Please, rerun your tests on a newest github version and update the report.

alexmurray, 2021-03-18:
Thanks for the updates - with current git master we now see only 9 defects compared to the previous 29 \o/.

Of these I think the STRING_NULL one should be addressed since this still seems like a real bug to me as according to the `readlink()` man page it does not nul terminate the buffer - so an easy fix for this would be to simply `memset()` the `os_progname_buf` buffer to 0 before use, or declare it as zero-initialised.

```
airscan-eloop.c:476
  Type: Dead default in switch (DEADCODE)

airscan-eloop.c:469:
  dead_error_condition: The "switch" governing value "mask & ELOOP_FDPOLL_BOTH" cannot reach the "default" case.
airscan-eloop.c:476:
  dead_error_line: Execution cannot reach this statement: "return "{??}";".

airscan-netif.c:293
  Type: Infinite loop (INFINITE_LOOP)

airscan-netif.c:293:
  loop_top: Top of the loop.
airscan-netif.c:310:
  loop_bottom: Bottom of the loop.
airscan-netif.c:293:
  loop_condition: If "halves[1] != NULL" is initially true then it will remain true.

airscan-os.c:39
  Type: Large stack use (STACK_USE)

airscan-os.c:39:
  stack_use_local_overflow: Local variable "buf" uses 16384 bytes of stack space, which exceeds the maximum single use of 10000 bytes.

airscan-os.c:112
  Type: String not null terminated (STRING_NULL)

airscan-os.c:80:
  1. string_null_argument: Function "readlink" does not terminate string "*os_progname_buf".
airscan-os.c:83:
  2. path: Condition "rc < 0", taking false branch.
airscan-os.c:112:
  3. string_null: Passing unterminated string "os_progname_buf" to "strrchr", which expects a null-terminated string.

airscan-pollable.c:95
  Type: Ignoring number of bytes read (CHECKED_RETURN)

airscan-pollable.c:95:
  1. check_return: "read(int, void *, size_t)" returns the number of bytes read, but it is ignored.

airscan-rand.c:30
  Type: Ignoring number of bytes read (CHECKED_RETURN)

airscan-rand.c:26:
  1. path: Condition "!(rand_fp != NULL)", taking false branch.
airscan-rand.c:30:
  2. check_return: "fread(void * restrict, size_t, size_t, FILE * restrict)" returns the number of bytes read, but it is ignored.

airscan-wsd.c:385
  Type: Unused value (UNUSED_VALUE)

airscan-wsd.c:400:
  value_overwrite: Overwriting previous write to "tmp" with value from "max_hei".
airscan-wsd.c:385:
  assigned_value: Assigning value from "min_wid" to "tmp" here, but that stored value is overwritten before it can be used.

airscan-zeroconf.c:700
  Type: Infinite loop (INFINITE_LOOP)

airscan-zeroconf.c:700:
  loop_top: Top of the loop.
airscan-zeroconf.c:717:
  loop_bottom: Bottom of the loop.
airscan-zeroconf.c:700:
  loop_condition: If "halves[1] != NULL" is initially true then it will remain true.

test.c:51
  Type: Large stack use (STACK_USE)

test.c:51:
  stack_use_local_overflow: Local variable "buf" uses 65536 bytes of stack space, which exceeds the maximum single use of 10000 bytes.
```

alexpevzner, 2021-03-18:
> Of these I think the STRING_NULL one should be addressed since this still seems like a real bug to me as according to the readlink()

It reads into the statically allocated buffer, which is zero-filled by compiler. Plus, on a previous iteration, I've added explicit `memset` to zero-fill the buffer. So I simply don't know how to silence this warning, but it is definitely false positive.

```
airscan-eloop.c:476
  Type: Dead default in switch (DEADCODE)
```

It wasn't on previous iteration, though code didn't change. Formally, Coverity is correct here, but if I remove the final `return`, some compilers will complain about missed return from function with value.

```
airscan-pollable.c:95
  Type: Ignoring number of bytes read (CHECKED_RETURN)
```

Casting returned value to `void` doesn't silence this warning, unfortunately

alexmurray, 2021-03-19:
Ok no worries - thanks again for looking at these and addressing them.

alexpevzner, 2021-04-09:
FIXED (as much as possible)

closed 2021-04-09 by alexpevzner

alexmurray, 2021-04-09:
Thanks again for your efforts on these @alexpevzner :)