alexpevzner/sane-airscan #133: driverless scaning does not work: "No connection to the scanner."

Golddouble, 2021-03-26:
My system: MX Linux 19.3 with Xfce (and Debian 10).
My printer: HP Envy 5544connected via usb

I have installed this:
https://download.opensuse.org/repositories/home:/pzz/Debian_10/amd64/ipp-usb_0.9.17-1+48.2_amd64.deb

After I have made this entry in `/etc/rc.local` file:
`ipp-usb standalone -bg`

-> Yes, driverless printing works now.

As I also want to scan driverless I have ...
... installed "sane-airscan 0.99.19-1-MX19+1" from test repository.

Then I have rebooted.
Then I have started the program "Simple Scan" and tried to make a scan.

Message: "No connection to the scanner."

So I can not scan. Is there something I miss? Thank you.

alexpevzner, 2021-03-26:
Hi,

please show me output of the following command: `lsusb -v` (you need to run it as root)

Golddouble, 2021-03-26:
```
Bus 004 Device 002: ID 8087:8000 Intel Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x8087 Intel Corp.
  idProduct          0x8000 
  bcdDevice            0.04
  iManufacturer           0 
  iProduct                0 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval              12
Hub Descriptor:
  bLength              11
  bDescriptorType      41
  nNbrPorts             8
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood        0 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x00
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
Device Qualifier (for other device speed):
[... 780 lines cut ...]
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00 0x06
  PortPwrCtrlMask    0xff 0xff
 Hub Port Status:
   Port 1: 0000.0303 lowspeed power enable connect
   Port 2: 0000.0100 power
   Port 3: 0000.0303 lowspeed power enable connect
   Port 4: 0000.0100 power
   Port 5: 0000.0100 power
   Port 6: 0000.0100 power
   Port 7: 0000.0100 power
   Port 8: 0000.0100 power
   Port 9: 0000.0100 power
   Port 10: 0000.0100 power
   Port 11: 0000.0503 highspeed power enable connect
   Port 12: 0000.0100 power
   Port 13: 0000.0100 power
   Port 14: 0000.0100 power
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

alexpevzner, 2021-03-26:
OK, thanks. And now, the following commands, please:
```
avahi-browse -rt _uscan._tcp
scanimage -L
```

Golddouble, 2021-03-27:
```
Goldouble@mx:~
$ avahi-browse -rt _uscan._tcp
bash: avahi-browse: Command not found.
Goldouble@mx:~
$ sudo avahi-browse -rt _uscan._tcp
[sudo] Passwort für Goldouble: 
sudo: avahi-browse: Command not found
Goldouble@mx:~
$ scanimage -L
device `airscan:e0:HP ENVY 5540 series [341375] (USB)' is a eSCL HP ENVY 5540 series [341375] (USB) eSCL network scanner
device `hpaio:/usb/ENVY_5540_series?serial=TH58Q1N1SQ0671' is a Hewlett-Packard ENVY_5540_series all-in-one
Goldouble@mx:~
$ 
```

alexpevzner, 2021-03-27:
What driver do you use when scan with simple-scan? Are you sure you use airscan, not hplip?

You may try to scan with xsane, it always ask what driver to use if there are more that one of them.

alexpevzner, 2021-04-09:
ping

Golddouble, 2021-04-09:
I have asked the MX Linux packers to make a newer simple-scan and sane-airscan version available. As soon as it is ready (in some weeks) I will try again.

ferdnyc, 2021-04-28:
@Golddouble Another tool you could try is [`gscan2pdf`](http://gscan2pdf.sf.net), which is [installed by default in MX](https://mxlinux.org/wiki/applications/gscan2pdf/) and, as they document on that wiki page, lets you specify the device on the command line if necessary.

(You can also select the device in the preferences GUI, which usually works fine, but if there's any doubt about device support or selection, then as a test and to be 100% sure you can launch it with the device specified in the `--device` argument.)

For your scanner:
```sh
gscan2pdf --device='airscan:e0:HP ENVY 5540 series [341375] (USB)'
```
should launch gscan2pdf connected directly to the ENVY via airscan.

Golddouble, 2021-04-29:
Thank you @ferdnyc for the hint.

I was not aware of gscan2pdf.
I have tried it out now.

1. Version 2.3.0-1 from stable repository:
Result:
[image]

2. I will try the same with Version 2.8.1-1 mx19+1 from test repository: later.

Golddouble, 2021-09-23:
MX Linux packers have backported simple-scan version 3.38.1

It has worked with this version.

closed 2021-09-23 by Golddouble