alexpevzner/sane-airscan #135: assertion error while scanning

akumria, 2021-04-08:
```
$ simple-scan 
file airscan-http.c: line 269 (http_uri_field_replace_len): assertion failed: (uri2 != NULL)
Aborted
```

The device in question is a HP OfficeJet 4620; scanning works fine via the

```
$ airscan-discover
[devices]
  Officejet 4620 series [58DDA5] = http://111.222.333.444:53048, WSD
  Officejet 4620 series [58DDA5] = http://[FE80::3AEA:A7FF:FE58:DDA5%252]:53048, WSD
```

(I've obscured the IPv4 address)

akumria, 2021-04-08:
Just discovered the `scanimage` command:

```
$ scanimage --test --verbose --progress
Output format is not set, using pnm as a default.
file airscan-http.c: line 269 (http_uri_field_replace_len): assertion failed: (uri2 != NULL)
Aborted
```
and

```
$ scanimage --verbose --progress
Output format is not set, using pnm as a default.
file airscan-http.c: line 269 (http_uri_field_replace_len): assertion failed: (uri2 != NULL)
Aborted
```

akumria, 2021-04-08:
Even just asking for the options to be set, results in the same assertion:

```
$ scanimage --verbose --progress -n
Output format is not set, using pnm as a default.
file airscan-http.c: line 269 (http_uri_field_replace_len): assertion failed: (uri2 != NULL)
Aborted
```

akumria, 2021-04-08:
Also found some debug options in `/etc/sane.d/airscan.conf`, which I enabled, re-running the above:

```
$ scanimage --verbose --progress --test
Output format is not set, using pnm as a default.
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 111.222.333.400
WSDD:  + fe80::1278:d2ff:fe7b:afdc%2
WSDD: 111.222.333.400: started discovery, UDP port=45214
WSDD: fe80::1278:d2ff:fe7b:afdc: started discovery, UDP port=46680
API: sane_get_devices(): called
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6 "Officejet 4620 series [58DDA5]": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv4 "Brother HL-3170CDW series": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv4 "Officejet 4620 series [58DDA5]": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6 "Officejet 4620 series [58DDA5]": AVAHI_BROWSER_NEW _scanner._tcp
MDNS: browse/ipv4 "Officejet 4620 series [58DDA5]": AVAHI_BROWSER_NEW _scanner._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
MDNS: resolve/ipv6 "Officejet 4620 series [58DDA5]": AVAHI_RESOLVER_FOUND _scanner._tcp
WSDD: directed probe: trying if=enp0s25, addr=111.222.333.444
WSDD: HTTP POST http://111.222.333.444/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 111.222.333.444 80
WSDD: HTTP trying 111.222.333.444:80
MDNS: resolve/ipv4 "Officejet 4620 series [58DDA5]": AVAHI_RESOLVER_FOUND _scanner._tcp
WSDD: directed probe: trying if=enp0s25, addr=111.222.333.444
WSDD: directed probe: already in progress, skipping
WSDD: HTTP 758 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP 1766 bytes received
WSDD: HTTP POST http://111.222.333.444/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: got response headers (200)
WSDD: HTTP 861 bytes received
WSDD: HTTP done response reception
WSDD: HTTP POST http://111.222.333.444/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: 200 OK
WSDD: HTTP POST http://111.222.333.444/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: OK
WSDD: HTTP POST http://111.222.333.444:3911/
WSDD: HTTP POST http://[FE80::3AEA:A7FF:FE58:DDA5%252]:3911/
WSDD: HTTP resolving 111.222.333.444 3911
WSDD: HTTP trying 111.222.333.444:3911
WSDD: HTTP resolving FE80::3AEA:A7FF:FE58:DDA5%2 3911
WSDD: HTTP trying [fe80::3aea:a7ff:fe58:dda5%2]:3911
WSDD: HTTP 630 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP 645 bytes sent
WSDD: HTTP done request sending
zeroconf: found urn:uuid:e3248000-80ce-11db-8000-30055c115e1b
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp0s25)
zeroconf:   name:      -
zeroconf:   model:     -
zeroconf:   addresses:
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: DNS-SD not finished...
MDNS: resolve/ipv4 "Brother HL-3170CDW series": AVAHI_RESOLVER_FOUND _ipp._tcp
WSDD: directed probe: trying if=enp0s25, addr=203.7.227.26
WSDD: directed probe: device already contacted, skipping
WSDD: HTTP 2896 bytes received
WSDD: HTTP POST http://111.222.333.444:3911/: got response headers (200)
[... 61 lines cut ...]
API: sane_open("w2:Officejet 4620 series [58DDA5]"): called
zeroconf: device_list wait: requested
zeroconf: device_list wait: OK
Officejet 4620 series [58DDA5]: device created
Officejet 4620 series [58DDA5]: DEVICE_STM_OPENED->DEVICE_STM_PROBING
Officejet 4620 series [58DDA5]: using protocol "WSD"
Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048
Officejet 4620 series [58DDA5]: HTTP using timeout: 5000 ms
Officejet 4620 series [58DDA5]: HTTP resolving 111.222.333.444 53048
Officejet 4620 series [58DDA5]: HTTP trying 111.222.333.444:53048
Officejet 4620 series [58DDA5]: HTTP 969 bytes sent
Officejet 4620 series [58DDA5]: HTTP done request sending
Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: got response headers (200)
Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
Officejet 4620 series [58DDA5]: HTTP 1206 bytes received
Officejet 4620 series [58DDA5]: HTTP done response reception
Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: 200 OK
Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: OK
Officejet 4620 series [58DDA5]: scanner capabilities: no supported image formats
file airscan-http.c: line 269 (http_uri_field_replace_len): assertion failed: (uri2 != NULL)
Aborted
```

similar IPv4 obfuscation done as earlier.

akumria, 2021-04-08:
And here is the output from the airscan trace log

```
00:00:02.518: Officejet 4620 series [58DDA5]: device created
00:00:02.518: Officejet 4620 series [58DDA5]: DEVICE_STM_OPENED->DEVICE_STM_PROBING
00:00:02.518: Officejet 4620 series [58DDA5]: using protocol "WSD"
00:00:02.519: Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048
00:00:02.519: Officejet 4620 series [58DDA5]: HTTP using timeout: 5000 ms
00:00:02.519: Officejet 4620 series [58DDA5]: HTTP resolving 111.222.333.444 53048
00:00:02.519: Officejet 4620 series [58DDA5]: HTTP trying 111.222.333.444:53048
00:00:02.529: Officejet 4620 series [58DDA5]: HTTP 969 bytes sent
00:00:02.529: Officejet 4620 series [58DDA5]: HTTP done request sending
00:00:02.570: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:02.570: Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: got response headers (200)
00:00:02.574: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:02.578: Officejet 4620 series [58DDA5]: HTTP 1206 bytes received
00:00:02.578: Officejet 4620 series [58DDA5]: HTTP done response reception
00:00:02.578: Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: 200 OK
==============================
POST http://111.222.333.444:53048
Connection: close
Content-Type: application/soap+xml
Cache-Control: no-cache
Pragma: no-cache
User-Agent: WSDAPI
Host: 111.222.333.444:53048
Content-Length: 786

<soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:sca="http://schemas.microsoft.com/windows/2006/08/wdp/scan">
  <soap:Header>
    <wsa:MessageID>urn:uuid:3a4b75ff-6feb-b0ea-8023-1751e4f92cd0</wsa:MessageID>
    <wsa:To>http://111.222.333.444:53048</wsa:To>
    <wsa:ReplyTo>
      <wsa:Address>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:Address>
    </wsa:ReplyTo>
    <wsa:Action>http://schemas.microsoft.com/windows/2006/08/wdp/scan/GetScannerElements</wsa:Action>
  </soap:Header>
  <soap:Body>
    <sca:GetScannerElementsRequest>
      <sca:RequestedElements>
        <sca:Name>sca:ScannerConfiguration</sca:Name>
      </sca:RequestedElements>
    </sca:GetScannerElementsRequest>
  </soap:Body>
</soap:Envelope>

Status: 200 OK
Server: gSOAP/2.7
Content-Type: application/soap+xml; charset=utf-8
Content-Length: 3972
Connection: close

<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:addressing="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wscn="http://schemas.microsoft.com/windows/2006/08/wdp/scan" xmlns:xop="http://www.w3.org/2004/08/xop/include">
  <SOAP-ENV:Header>
    <addressing:To>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</addressing:To>
    <addressing:Action>http://schemas.microsoft.com/windows/2006/08/wdp/scan/GetScannerElementsResponse</addressing:Action>
    <addressing:MessageID>uuid:1c860619-bba9-1f08-b2d0-38eaa758dda5</addressing:MessageID>
    <addressing:RelatesTo>urn:uuid:3a4b75ff-6feb-b0ea-8023-1751e4f92cd0</addressing:RelatesTo>
  </SOAP-ENV:Header>
  <SOAP-ENV:Body>
    <wscn:GetScannerElementsResponse>
      <wscn:ScannerElements>
        <wscn:ElementData Valid="true" Name="wscn:ScannerConfiguration">
          <wscn:ScannerConfiguration>
            <wscn:DeviceSettings>
              <wscn:FormatsSupported>
                <wscn:FormatValue>tiff-single-uncompressed</wscn:FormatValue>
              </wscn:FormatsSupported>
              <wscn:CompressionQualityFactorSupported>
                <wscn:MinValue>0</wscn:MinValue>
[... 79 lines cut ...]
                </wscn:ADFResolutions>
                <wscn:ADFColor>
                  <wscn:ColorEntry>Grayscale8</wscn:ColorEntry>
                  <wscn:ColorEntry>RGB24</wscn:ColorEntry>
                </wscn:ADFColor>
                <wscn:ADFMinimumSize>
                  <wscn:Width>26</wscn:Width>
                  <wscn:Height>26</wscn:Height>
                </wscn:ADFMinimumSize>
                <wscn:ADFMaximumSize>
                  <wscn:Width>8500</wscn:Width>
                  <wscn:Height>14000</wscn:Height>
                </wscn:ADFMaximumSize>
              </wscn:ADFFront>
            </wscn:ADF>
          </wscn:ScannerConfiguration>
        </wscn:ElementData>
      </wscn:ScannerElements>
    </wscn:GetScannerElementsResponse>
  </SOAP-ENV:Body>
</SOAP-ENV:Envelope>

00:00:02.578: Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: OK
00:00:02.579: Officejet 4620 series [58DDA5]: scanner capabilities: no supported image formats
```

alexpevzner, 2021-04-08:
Hi @akumria,

there are actually 2 problems here:
1. assertion in IPV6 URL handling
2. your device only supports TIFF image formats, while sane-airscan doesn't have TIFF image decoder

The first is simple and I've just fixed it. The second will require more time to fix. The problem is that `libtiff` is ugly, and I don't know any alternatives to it.

I will update this issue when I will have some fix for the second problem.

alexpevzner, 2021-04-09:
Hi @akumria,

I've added TIFF support (currently to https://github.com/alexpevzner/sane-airscan-unstable, but it will be merged to the stable `sane-airscan` soon).

You may test by downloading and installing appropriate package from here: https://download.opensuse.org/repositories/home:/pzz/)

Note, for now you need `sane-airscan-unstable`, not `sane-airscan_0.99.24`

akumria, 2021-04-10:
Hi @alexpevzner 

Thanks for this, I've given this a try and I now get a scan dialog.

However during the middle of the scan, something seems to crash the scanner and the only way to recover is to physically power off / power on the device.

The last lines of the protocol trace are:

```
00:00:04.670: Officejet 4620 series [58DDA5]: HTTP POST http://111.222.333.444:53048: 200 OK
==============================
POST http://111.222.333.444:53048
Connection: close
Content-Type: application/soap+xml
Cache-Control: no-cache
Pragma: no-cache
User-Agent: WSDAPI
Host: 111.222.333.444:53048
Content-Length: 1663

<soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:sca="
http://schemas.microsoft.com/windows/2006/08/wdp/scan">
  <soap:Header>
    <wsa:MessageID>urn:uuid:81864597-b729-7cc1-7e46-d7e250abb923</wsa:MessageID>
    <wsa:To>http://111.222.333.444:53048</wsa:To>
    <wsa:ReplyTo>
      <wsa:Address>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</wsa:Address>
    </wsa:ReplyTo>
    <wsa:Action>http://schemas.microsoft.com/windows/2006/08/wdp/scan/CreateScanJob</wsa:Action>
  </soap:Header>
  <soap:Body>
    <sca:CreateScanJobRequest>
      <sca:ScanTicket>
        <sca:JobDescription>
          <sca:JobName>sane-airscan request</sca:JobName>
          <sca:JobOriginatingUserName>sane-airscan</sca:JobOriginatingUserName>
          <sca:JobInformation>sane-airscan</sca:JobInformation>
        </sca:JobDescription>
        <sca:DocumentParameters>
          <sca:Format>tiff-single-uncompressed</sca:Format>
          <sca:ImagesToTransfer>0</sca:ImagesToTransfer>
          <sca:InputSize>
            <sca:InputMediaSize>
              <sca:Width>8500</sca:Width>
              <sca:Height>11693</sca:Height>
            </sca:InputMediaSize>
          </sca:InputSize>
          <sca:InputSource>Platen</sca:InputSource>
          <sca:MediaSides>
            <sca:MediaFront>
              <sca:ColorProcessing>RGB24</sca:ColorProcessing>
              <sca:Resolution>
                <sca:Width>300</sca:Width>
                <sca:Height>300</sca:Height>
              </sca:Resolution>
              <sca:ScanRegion>
                <sca:ScanRegionXOffset>0</sca:ScanRegionXOffset>
                <sca:ScanRegionYOffset>0</sca:ScanRegionYOffset>
                <sca:ScanRegionWidth>8500</sca:ScanRegionWidth>
                <sca:ScanRegionHeight>11693</sca:ScanRegionHeight>
              </sca:ScanRegion>
            </sca:MediaFront>
          </sca:MediaSides>
        </sca:DocumentParameters>
      </sca:ScanTicket>
    </sca:CreateScanJobRequest>
  </soap:Body>
</soap:Envelope>

Status: 200 OK
[... 94 lines cut ...]
00:00:09.561: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:09.581: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:09.588: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:09.588: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:09.602: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:09.654: Officejet 4620 series [58DDA5]: HTTP 1904 bytes received
[ ... ]
00:00:56.769: Officejet 4620 series [58DDA5]: HTTP 8 bytes received
00:00:56.772: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.782: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.784: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.787: Officejet 4620 series [58DDA5]: HTTP 1432 bytes received
00:00:56.791: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.793: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.794: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.804: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.805: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.812: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.820: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
00:00:56.821: Officejet 4620 series [58DDA5]: HTTP 472 bytes received
00:00:56.821: Officejet 4620 series [58DDA5]: HTTP 8 bytes received
00:00:56.828: Officejet 4620 series [58DDA5]: HTTP 1448 bytes received
```

And it appears that `scanimage` is spinning on a futux and not progressing.

alexpevzner, 2021-04-10:
How long did you wait? Has the HTTP response receiving eventually stopped, you've just interrupted it? Please, try with lower resolutions.

Your printer can only return uncompressed TIFF images, it can be huge, so image reception can take a while.

akumria, 2021-04-10:
Hi @alexpevzner,

I started off the scan job and did some other things, so I can't put an exact time on it.

But it turns out the printer idles itself after 15 minutes of no communication from a computer anyway.

I was not until I went into the next room that I discovered this.

I re-tried the scan a few times (well 2, actually) and after 5 mins cancelled each of them. I could understand if there was something happening with the process, but as I said an strace indicated that it was spinning on a futex.

For reference with another machine running the Windows software, each page scan is about 2 mins.

akumria, 2021-04-10:
Note - happy to take the scans not completing to another issue, since the assertion error is now fixed if that is helpful.

alexpevzner, 2021-04-10:
`scanimage` is sleeping on a futex, because all I/O is performed from a separate thread, while main thread is waiting for results (yes, on futex). The I/O thread is most likely sleeping on `poll`, waiting for data to arrive on a socket. You may check with `strace -f`

From logs I see that communication has reached the image downloading state and hung somewhere during image downloading. There is no timeout in this state, because it is hard to guess, what timeout value would be reasonable. Some devices are VERY slow.

I would appreciate if you could capture scanning from Windows with Wireshark, though Windows driver may use some proprietary protocol, not WSD, when communicating with scanner.

alexpevzner, 2021-04-15:
ping

akumria, 2021-04-16:
sorry, it will take me a few days to set things up to capture since the machines involved all use wifi.

alexpevzner, 2021-04-16:
OK.

I also want to ask you to send me the whole log file, not only interesting parts, as before. You can attach it here.

akumria, 2021-04-18:
9531bdfb1eaac01bc0d76eae53f7c880df0306ef  hp-scanner.pcapng.gz

Sent to your email address

alexpevzner, 2021-04-21:
Hi,

HP uses its own protocol, similar to eSCL, but not equal to it.

May I ask you to send me a whole log of failed WSD scan?

akumria, 2021-04-26:
Yes, I will do.

akumria, 2021-04-28:
I did the following:

```
scanimage -v -p -o test.pnm 
```

And have sent three captures to you.

```
shasum scanimage-*
f4b6269034afeffaf8a1933f7d4bcbd056a35820  scanimage-one.pcapng.gz
5e9de9e093ee30d53da92f140d926b6b1d584c70  scanimage-three.pcapng.gz
bd9cfac681ba4a2554e82e6edaff9822578d39df  scanimage-two.pcapng.gz
```

In the first two scanimage exited without actually completing the scan.

When it failed, the scanner was in a weird state and I reset it (removed power, reinserted power) to get it working again.

On the third try, the scan worked (resulting in a 26M .pnm file)

Let me know if you need further details, or I should try something else.

alexpevzner, 2021-04-28:
Hi,

I've fixed some HTTP-related bug. May I ask you to retest against latest `sane-airscan-unstable`?

akumria, 2021-04-29:
Hi,

I installed `sane-airscan-unstable 1619610080-1+20.1` for Debian from your openbuild respository. I did:

```
scanimage -v -p -o test3.pnm 
```

The first invocation failed. I reset the scanner. The second failed. I also reset the scanner.

With the third one, after resetting it, I was distracted, and tried the command but I noticed the front-panel had gone into "idle" (i.e. was not lit up) mode. It also failed. When I reset it again, I then did scan four -- which worked -- as soon as the front-panel was lit.

I wonder if perhaps the scanner is taking some time to wake from idle, before it is ready to scan.

```
$ shasum unstable-*
5e8dfde81ac37709695c08862191eef29f74e2f8  unstable-four.pcapng.gz
8994aa12e2daf7f80614336bc6c540607eae4c1b  unstable-one.log
1e2fdca5d93868e5056c377fcc384c344459c30d  unstable-one.pcapng.gz
e5b802efc7ff4567ba74fefe01843e9b926d26cf  unstable-three.log
9fe0048f9a16e0bbebfd832cfb8f8c65df9ede0f  unstable-three.pcapng.gz
2b8661f1ca20f8816f41d9f7459718d54452241a  unstable-two.log
b1fdbf4edefd6cb0176f46beadc896f178d46b04  unstable-two.pcapng.gz
```

As the files are large, they have been sent to your email address.

akumria, 2021-04-29:
Oh, `unstable-two` is particularly interesting as the scan started but then the scanimage program seemed to get stuck waiting in a loop; I had to Ctrl-C it and restart the scanner.

akumria, 2021-05-12:
Is there anything further I can do to assist here?

alexpevzner, 2021-05-12:
Yes. Are you able to build/install `sane-airscan-unstable` by yourself?

akumria, 2021-05-12:
Yes - is there a branch you would like me to try?

alexpevzner, 2021-05-12:
https://github.com/alexpevzner/sane-airscan-unstable

Please, enable protocol traces with hex dumps:
```
[debug]
trace = ~/airscan/trace
enable = true
hexdump = true
```

And send me `*.log` files from the `~/airscan/trace` directory

closed 2024-11-18 by alexpevzner