alexpevzner/sane-airscan #136: Port needed in Host header for some scanners

yetamrra, 2021-04-09:
We have a report from Epson that some of their models require the port to be included in the Host header for eSCL even if it matches the default value.  When the port isn't included, the steps up to the scan job creation work, but an invalid redirect will be sent from the response to /eSCL/ScanJobs and then the scan job will never finish.

I saw from the commit history that unconditionally adding the port to the Host header will break other devices.  Do you think this could be added as a quirk for just Epson devices?  They have confirmed to us that all their models will work correctly if the port is present, so it shouldn't introduce any new compatibility breaks.  I'm happy to take a stab at adding a quirk if you think that's a viable approach.

alexpevzner, 2021-04-09:
Yes, it sounds reasonable. If Epson people can be 100% sure that all Epson devices will work with such a quirk, we can apply this quirk to all Epson devices.

yetamrra, 2021-04-09:
OK, I'll put something together and send a PR in a few days.

alexpevzner, 2021-04-09:
OK, agreed.

PR #137 (2021-04-15): Add quirk to force port in Host header
Some devices rely on finding the port in the Host HTTP header to
construct their response to the /eSCL/ScanJobs request.  When the port
isn't included, the steps up to the scan job creation work, but an
invalid redirect will be sent in the response to /eSCL/ScanJobs and then
the scan job will never finish.

Add a new quirk that forces the port to always be included in the Host
header.  This is known to affect some Epson models, and Epson has
indicated that this should be safe to use with all of their devices, so
enable it for eSCL whenever the manufacture [... cut]
commit 5d175bb0 (2021-04-15): Add quirk to force port in Host header

Some devices rely on finding the port in the Host HTTP header to
construct their response to the /eSCL/ScanJobs request.  When the port
isn't included, the steps up to the scan job creation work, but an
invalid redirect will be sent in the response to /eSCL/ScanJobs and then
the scan job will never finish.

Add a new quirk that forces the port to always be included in the Host
header.  This is known to affect some Epson models, and Epson has
indicated that this should be safe to use with all of their devices, so
enable it for eSCL whenever the manufactur [... cut]
closed 2021-04-15 by alexpevzner