alexpevzner/sane-airscan #139: Canon i-SENSYS MF411dw.  Scan quality.

aleksey1831981, 2021-04-15:
Приветствую, Александр.

    Ранее мы уже общались по настройке нашего сканера, и это прекрасно работает до сих пор).
    Но, буквально сегодня, мы обратили внимание, что при настройке разрешения сканирования в Simple Scan, наш сканер все равно сканирует с разрешением 300 dpi. Независимо от того, уменьшаем значение, или увеличиваем.
   Так-же мы пытались сканировать через Scan2PDF. Но тут мы даже не могли выбрать другое значение. Т.е. из доступных разрешений, есть только 300 dpi.
  
     В драйвер sane у нас закомментированы все строки.
     В sane-airscan принудительно указан eSCL режим.

    Является -ли это проблемой sane-airscan?

alexpevzner, 2021-04-15:
Здравствуйте, Алексей,

Я посмотрел логи с прошлого раза.

В режиме eSCL сканер предлагает единственное разрешение, 300 DPI. В режиме WSD он предлагает 150 и 300. 

Simple Scan - довольно глюкавая программа, она может выдать список разрешений из своей собственной головы, не спросив у драйвера, так что она не показатель.

Возможно, кэнноновский "родной" драйвер умеет уговаривать этот сканер работать в других разрешениях, но он, скорее всего, использует не eSCL/WSD, а свой собственный протокол. И эти другие разрешения могут оказаться, на самом деле, програмной эмуляцией.

aleksey1831981, 2021-04-15:
И, кстати, если мы сканируем с Виндовс компьютера, то мы можем выбрать любое нужное разрешение. И сканер корректно сканирует с указанным разрешением.

Завтра я еще попробую сканировать в WSD, и после этого сообщу, что у меня получилось.

alexpevzner, 2021-04-15:
А какие разрешения умеет венда?

aleksey1831981, 2021-04-16:
Потестировали Винду. 

Сканировали в JPG через фирменную утилиту от Canon.
Доступны след разрешения:
75, 100, 150, 200, 300, 400, 600.

После сохранения сканов,  в свойствах файлов все разрешения соответствуют заявленным.
Размеры файлов тоже все разные.

alexpevzner, 2021-04-16:
С этим я вряд ли что-то смогу сделать. Некоторые устройства, действительно, не отдают по eSCL/WSD полный набор своей функциональности.

Странно, что он предлагает под виндой 400/600. В eSCL'ном ответе написано, что OpticalResolution равен 300. Т.е., больше 300-т он реально сканировать не может, может только додумывать.

Я тогда закрою этот issue за неимением идей, что я могу с ним сделать.

closed 2021-04-16 by alexpevzner

aleksey1831981, 2021-04-16:
Ок. Спасибо.