alexpevzner/sane-airscan #143: Ricoh SP C261SFNw - Error during device I/O

heinrichat, 2021-04-26:
I finally managed to replace my printer with a SANE compatible one, but I have troubles getting it to work. Running Debian 10 with backports enabled. I've read through the Debian and Arch Linux SANE Wiki without any success. Further I've found no option in the configuration webinterface to set a timeout or not, but my energy saving mode is active. I've tried scanning while the MFP was not in energy saving mode, but that didn't work either. 

`
$ airscan-discover
[devices]
  RICOH SP C261SFNw (da9638) = http://192.168.1.124:631/eSCL/, eSCL
  RICOH SP C261SFNw (da9638) = https://192.168.1.124:443/eSCL/, eSCL
`

`$ export SANE_DEBUG_DLL=255 && scanimage --format=png > test.png
[sanei_debug] Setting debug level of dll to 255.
[dll] sane_init: SANE dll backend version 1.0.13 from sane-backends 1.0.27
[dll] sane_init/read_dlld: attempting to open directory `./dll.d'
[dll] sane_init/read_dlld: attempting to open directory `/etc/sane.d/dll.d'
[dll] sane_init/read_dlld: using config directory `/etc/sane.d/dll.d'
[dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/airscan
[dll] sane_init/read_config: reading dll.d/airscan
[dll] add_backend: adding backend `airscan'
[dll] sane_init/read_dlld: done.
[dll] sane_init/read_config: reading dll.conf
[dll] add_backend: adding backend `airscan'
[dll] add_backend: `airscan' is already there
[dll] sane_get_devices
[dll] load: searching backend `airscan' in `/usr/lib/x86_64-linux-gnu/sane:/usr/lib/sane'
[dll] load: trying to load `/usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1'
[dll] load: dlopen()ing `/usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1'
[dll] init: initializing backend `airscan'
[dll] init: backend `airscan' is version 1.0.0
[dll] sane_get_devices: found 1 devices
[dll] sane_open: trying to open `airscan:e0:RICOH SP C261SFNw (da9638)'
[dll] sane_open: open successful
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=0)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=0,action=0,value=0x7ffda979494c,info=(nil))
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=0)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=0,action=0,value=0x7ffda9794850,info=(nil))
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=1)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=2)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=3)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=4)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=5)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=6)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=7)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=8)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=9)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=10)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=11)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=12)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=13)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=14)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=15)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=16)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=8,action=0,value=0x55f8a7577380,info=(nil))
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=6,action=0,value=0x7ffda9794854,info=(nil))
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=9,action=0,value=0x55f8a7577384,info=(nil))
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=7,action=0,value=0x7ffda9794854,info=(nil))
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=6,action=0,value=0x7ffda9794980,info=(nil))
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=8)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=8,action=1,value=0x7ffda97949a0,info=0x7ffda9794864)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=7,action=0,value=0x7ffda9794980,info=(nil))
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=9)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=9,action=1,value=0x7ffda97949a0,info=0x7ffda9794864)
[dll] sane_get_option_descriptor(handle=0x55f8a7c6b7b0,option=2)
[dll] sane_control_option(handle=0x55f8a7c6b7b0,option=2,action=0,value=0x7ffda9794860,info=(nil))
[dll] sane_start(handle=0x55f8a7c6b7b0)
scanimage: sane_start: Error during device I/O
[dll] sane_cancel(handle=0x55f8a7c6b7b0)
[dll] sane_close(handle=0x55f8a7c6b7b0)
[dll] sane_exit: exiting
[dll] sane_exit: calling backend `airscan's exit function
[dll] sane_exit: finished`

alexpevzner, 2021-04-26:
Hi,

I need a protocol trace. To obtain it, please uncomment the following lines in the `/etc/sane.d/airscan.conf`:
```
[debug]
trace = ~/airscan/trace
enable = true
```

Trace files will be in the `~/airscan/trace` directory, which will be created automatically. Currently I want to look to the `.log` files only`.tar` files are not needed for now.

heinrichat, 2021-04-26:
```
00:00:00.001: zeroconf: zeroconf configuration:
00:00:00.001: zeroconf:   discovery    = e00:00:00.001: zeroconf: zeroconf configuration:
00:00:00.001: zeroconf:   discovery    = enable
00:00:00.001: zeroconf:   model        = network
00:00:00.001: zeroconf:   protocol     = auto
00:00:00.001: zeroconf:   ws-discovery = fast
00:00:00.006: MDNS: AVAHI_CLIENT_S_RUNNING
00:00:00.011: WSDD: netif addresses update:
00:00:00.011: WSDD:  + 192.168.1.145
00:00:00.011: WSDD:  + 192.168.1.155
00:00:00.011: WSDD:  + fe80::f2de:f1ff:fea6:41c%2
00:00:00.011: WSDD: 192.168.1.145: started discovery, UDP port=38715
00:00:00.011: WSDD: probe sent: 192.168.1.145:38715->239.255.255.250:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:264d576c-c8ff-f4f4-5034-aecbeb70b99c</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.012: WSDD: setsockopt(AF_INET,IP_ADD_MEMBERSHIP): Address already in use
00:00:00.012: WSDD: 192.168.1.155: started discovery, UDP port=34271
00:00:00.012: WSDD: probe sent: 192.168.1.155:34271->239.255.255.250:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:5fe28f22-2d79-2a25-06c9-0b94b79252e8</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.012: WSDD: fe80::f2de:f1ff:fea6:41c: started discovery, UDP port=51608
00:00:00.012: WSDD: probe sent: [fe80::f2de:f1ff:fea6:41c%2]:51608->[ff02::c]:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:db6fb44c-1712-13fd-f981-18cd2b3aac20</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.013: MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_NEW
00:00:00.013: MDNS: browse-ipv6("_ipp._tcp"): "RICOH SP C261SFNw (da9638)"
00:00:00.014: MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
00:00:00.014: MDNS: browse-ipv4("_ipp._tcp"): "RICOH SP C261SFNw (da9638)"
00:00:00.015: MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
00:00:00.015: MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
00:00:00.015: MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_NEW
00:00:00.015: MDNS: browse-ipv6("_ipps._tcp"): "RICOH SP C261SFNw (da9638)"
00:00:00.016: MDNS: browse-ipv4("_ipps._tcp"): AVAHI_BROWSER_NEW
00:00:00.016: MDNS: browse-ipv4("_ipps._tcp"): "RICOH SP C261SFNw (da9638)"
00:00:00.017: MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
00:00:00.017: MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
[... 1898 lines cut ...]
00:01:48.759: RICOH SP C261SFNw (da9638): HTTP fdpoll: {--} -> {-w}
00:01:48.760: RICOH SP C261SFNw (da9638): HTTP 82 bytes sent
00:01:48.760: RICOH SP C261SFNw (da9638): HTTP done request sending
00:01:48.760: RICOH SP C261SFNw (da9638): HTTP fdpoll: {-w} -> {rw}
00:01:48.760: RICOH SP C261SFNw (da9638): HTTP fdpoll: {rw} -> {r-}
00:01:48.762: RICOH SP C261SFNw (da9638): HTTP 155 bytes received
00:01:48.762: RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/101: got response headers (200)
00:01:48.762: RICOH SP C261SFNw (da9638): HTTP done response reception
00:01:48.762: RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/101: 200 OK
==============================
DELETE http://192.168.1.124:631/eSCL/ScanJobs/101
Connection: close
Host: 192.168.1.124:631

Status: 200 OK
Date: Thu Jan  1 09:06:30 1970
Cache-Control: no-cache
Server: bsa/1.2
Connection: close
Content-Type: text/xml
Content-Length: 0

00:01:48.762: RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/101: OK
00:01:48.762: RICOH SP C261SFNw (da9638): DEVICE_STM_CANCEL_SENT->DEVICE_STM_CANCEL_REQ_DONE
```

alexpevzner, 2021-04-26:
I see you've has cancelled scan after ~1.5 minutes of waiting. Did scanner actually started scanning process (according to scanning noise etc)? Can it be that it was waiting for confirmation from the front panel?

heinrichat, 2021-04-26:
That time I heard a scanning noise, yes. Also this is the content of test.png
I've checked there is no cofrimation I can do at the unit. It only informs me with something like "Device busy. Scanning... (eSCL)."

```
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.1.145
WSDD:  + 192.168.1.155
WSDD:  + fe80::f2de:f1ff:fea6:41c%2
WSDD: 192.168.1.145: started discovery, UDP port=49738
WSDD: setsockopt(AF_INET,IP_ADD_MEMBERSHIP): Address already in use
WSDD: 192.168.1.155: started discovery, UDP port=48155
WSDD: fe80::f2de:f1ff:fea6:41c: started discovery, UDP port=58615
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv6("_ipp._tcp"): "RICOH SP C261SFNw (da9638)"
API: sane_get_devices(): called
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv6("_ipps._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv4("_ipps._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipps._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv6("_uscan._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv4("_uscan._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_uscan._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv6("_uscans._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv4("_uscans._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_uscans._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv6("_scanner._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv4("_scanner._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_scanner._tcp"): "RICOH SP C261SFNw (da9638)"
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: resolve-ipv6("_ipp._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv6("_ipp._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:631
WSDD: directed probe: trying if=enp0s25, addr=192.168.1.124
WSDD: HTTP POST http://192.168.1.124/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.1.124 80
WSDD: HTTP trying 192.168.1.124:80
WSDD: HTTP fdpoll: {--} -> {-w}
MDNS: resolve-ipv4("_ipp._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:631
WSDD: directed probe: trying if=enp0s25, addr=192.168.1.124
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv6("_ipps._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv6("_ipps._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:443
WSDD: directed probe: trying if=enp0s25, addr=192.168.1.124
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipps._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipps._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:443
WSDD: directed probe: trying if=enp0s25, addr=192.168.1.124
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv6("_uscan._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv6("_uscan._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:631
WSDD: directed probe: trying if=enp0s25, addr=192.168.1.124
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_uscan._tcp", "RICOH SP C261SFNw (da9638)"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_uscan._tcp", "RICOH SP C261SFNw (da9638)"): 192.168.1.124:631
zeroconf: found urn:uuid:6861d9b0-a100-11e0-8264-002673da9638
zeroconf:   method:    ZEROCONF_USCAN_TCP
[... 207 lines cut ...]
WSDD: 192.168.1.145: done discovery
WSDD: fe80::f2de:f1ff:fea6:41c: done discovery
WSDD: 192.168.1.155: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: initial scan timer expired
RICOH SP C261SFNw (da9638): cancel processing started
RICOH SP C261SFNw (da9638): JOB status=Operation was cancelled
RICOH SP C261SFNw (da9638): DEVICE_STM_CANCEL_REQ->DEVICE_STM_CANCEL_SENT
RICOH SP C261SFNw (da9638): HTTP using timeout: 10000 ms
RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/103
RICOH SP C261SFNw (da9638): HTTP using timeout: 30000 ms
RICOH SP C261SFNw (da9638): HTTP resolving 192.168.1.124 631
RICOH SP C261SFNw (da9638): HTTP trying 192.168.1.124:631
RICOH SP C261SFNw (da9638): HTTP fdpoll: {--} -> {-w}
RICOH SP C261SFNw (da9638): HTTP 82 bytes sent
RICOH SP C261SFNw (da9638): HTTP done request sending
RICOH SP C261SFNw (da9638): HTTP fdpoll: {-w} -> {rw}
RICOH SP C261SFNw (da9638): HTTP fdpoll: {rw} -> {r-}
RICOH SP C261SFNw (da9638): HTTP 155 bytes received
RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/103: got response headers (200)
RICOH SP C261SFNw (da9638): HTTP done response reception
RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/103: 200 OK
RICOH SP C261SFNw (da9638): HTTP DELETE http://192.168.1.124:631/eSCL/ScanJobs/103: OK
RICOH SP C261SFNw (da9638): DEVICE_STM_CANCEL_SENT->DEVICE_STM_CANCEL_REQ_DONE
```

alexpevzner, 2021-04-26:
Have you tried to wait for some longer time?

On logs I see that scan job was successfully created on a scanner, then driver initiates `HTTP GET` to obtain the image, and this `HTTP GET` stays pending without any progress until cancelled. Until that point everything looks very reasonable and clear.

Some scanners take a very noticeable amount of time to return image.

heinrichat, 2021-04-26:
I've waited until I could scan a new image again. Which takes about 5 minutes. But now I don't hear any scanning noise. This is so random.

alexpevzner, 2021-04-28:
Hi,

may I ask you to manually configure device, using HTTPS (not HTTP) URL and retest.

To manually configure device, add the following lines to the `/etc/sane.d/airscan.conf`:

```
[devices]
  RICOH SP C261SFNw (da9638) = https://192.168.1.124:443/eSCL/, eSCL
```

heinrichat, 2021-09-24:
Hi,

sorry for the late replay. The printers USB port suddenly doesn't work anymore. I think the device might be faulty. You can close the issue. Thank you for your help :) I appreciate it.

In any case do you have a recommendation for a fairly recent color laser multi function printer which you *know* works very well with sane-airscan?

alexpevzner, 2021-09-24:
Recent Kyocera MFP printers work very well. The devices are quite expensive, but toner is **very** reasonably priced, especially on highest-speed models.

heinrichat, 2021-09-24:
Okay thank you for the suggestion. I went for the Kyocera Ecosys M5521cdw. Hope it will work well :+1:

alexpevzner, 2021-09-24:
ECOSYS M5526cdw will cost almost the same, but toner will be ~2 times cheaper, per page.

heinrichat, 2021-09-25:
> ECOSYS M5526cdw will cost almost the same, but toner will be ~2 times cheaper, per page.

That one is too expensive for private use. 300€ vs 600€.
I was replacing the toner powder myself with the Ricoh anyways, so I think cost per page doesn't won't matter in this case.

I might repair a ECOSYS M5526cdw for cheap. Will post here if SANE works ;-)

heinrichat, 2021-10-01:
I can't resolve the issue with the Ricoh. It just might not work. I'm curious if sane-airscan works with Ricoh M C250fwb. This would be the latest and greatest from Ricoh. However it seems like Ricoh (the company) will be gone soon so... I'm happy with my Kyocera.

closed 2021-10-01 by heinrichat

alexpevzner, 2021-10-04:
Hm. In Russia, M5521cdw vs M5526cdw is EUR 437 vs 563. Not as much as 300 vs 600.

I've begun writting sane-airscan after purchase of Kyocera ECOSYS M2040d, which was not supported by SANE at that time. I've chosen this printer due to it's reputation of very reliably device (my previous Xerox WorkCentre was broken completely at mechanical level after printing about 1500 pages during 5 years) and due to very low cost of original toner, per page.

heinrichat, 2021-10-04:
I repaired the Kyocera ECOSYS M5526cdw. Two things weren't working. Printouts had color stripes and not grabbing paper via ADf while scanning. First problem I tackeled with deep clean and repair of the auger. A plastic piece broke off. And for the ADF I just soaked the paper pickup rollers in the ADF with WD40. Now the rubber has more grip and is able to grab the paper sheets. And for the toner I bought aftermarket toner cartridges with a fill hole ontop.

Pretty awesome machine. I can attest that nearly everything works. Skookum machine!! And build quality is really good and easily repairable. While disassembly only two different kind of screws where used. One metric machine screw and one for plastic parts. Kudos Kyocera.

Only issue I had was with Xsane. It doesn't want to combine multiple scanned sheets to a PDF. Xsane only saves one page.... but that is purely a Xsane problem.