alexpevzner/sane-airscan #147: Brother ADS-2700W

pschichtel, 2021-05-21:
Hi I just got an ADS-2700W today and immediately tried to use airscan with it, sadly it's not being detected.

```$ airscan-discover -d -t
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 2001:16b8:247a:a201:4ab3:b87c:ae4d:3246
WSDD:  + 192.168.1.10
WSDD:  + fe80::4e11:4d5:9c02:ed20%3
WSDD: 2001:16b8:247a:a201:4ab3:b87c:ae4d:3246: started discovery, UDP port=50643
WSDD: 192.168.1.10: started discovery, UDP port=34096
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fe80::4e11:4d5:9c02:ed20: started discovery, UDP port=44220
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
WSDD: HTTP POST http://192.168.1.154:3911/
WSDD: HTTP resolving 192.168.1.154 3911
WSDD: HTTP trying 192.168.1.154:3911
WSDD: HTTP fdpoll: {--} -> {-w}
WSDD: HTTP 631 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP fdpoll: {-w} -> {rw}
WSDD: HTTP fdpoll: {rw} -> {r-}
WSDD: HTTP 1887 bytes received
WSDD: HTTP POST http://192.168.1.154:3911/: got response headers (400)
WSDD: HTTP done response reception
WSDD: HTTP POST http://192.168.1.154:3911/: 400 Bad Request
WSDD: HTTP POST http://192.168.1.154:3911/: Bad Request
WSDD: "(null)": IPv4 address expected but not yet discovered
WSDD: "(null)": IPv6 address expected but not yet discovered
WSDD: "(null)": IPv4 address expected but not yet discovered
WSDD: "(null)": IPv6 address expected but not yet discovered
WSDD: "(null)": IPv4 address expected but not yet discovered
WSDD: "(null)": IPv6 address expected but not yet discovered
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: WSDD not finished...
WSDD: "(null)": IPv4 address expected but not yet discovered
WSDD: "(null)": IPv6 address expected but not yet discovered
WSDD: "(null)": publish-delay timer expired
zeroconf: found urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 3 (eno1)
zeroconf:   name:      -
zeroconf:   model:     -
zeroconf:   addresses:
zeroconf:   protocol:  WSD
zeroconf:   endpoints:
zeroconf:   device:    0000 (created)
zeroconf: device_list wait: WSDD not finished...
WSDD: 192.168.1.10: done discovery
WSDD: fe80::4e11:4d5:9c02:ed20: done discovery
WSDD: 2001:16b8:247a:a201:4ab3:b87c:ae4d:3246: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb (0): can: none, use: none
zeroconf: urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb (0): skipping, none of supported protocols discovered
zeroconf: zeroconf_device_list_get: resulting list:
[devices]
zeroconf: device gone urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 3 (eno1)
```

Wireshark showed me this SOAP request http://192.168.1.154:3911/ with method POST:

```xml
<?xml version="1.0"?>
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:s="http://www.w3.org/2003/05/soap-envelope">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2004/09/transfer/Get</a:Action>
    <a:MessageID>urn:uuid:1be51ddb-b394-fe33-b618-1294796eb78a</a:MessageID>
    <a:To>urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb</a:To>
    <a:ReplyTo>
      <a:Address>http://schemas.xmlsoap.org/ws/2004/08/addressing/role/anonymous</a:Address>
    </a:ReplyTo>
  </s:Header>
  <s:Body/>
</s:Envelope>
```
and the scanner responds with this response with `HTTP/1.1 400 Bad Request`:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wse="http://schemas.xmlsoap.org/ws/2004/08/eventing" xmlns:xop="http://www.w3.org/2004/08/xop/include" xmlns:mex="http://schemas.xmlsoap.org/ws/2004/09/mex" xmlns:wxf="http://schemas.xmlsoap.org/ws/2004/09/transfer" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof" xmlns:wsa="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:wscn="http://schemas.microsoft.com/windows/2006/08/wdp/scan" xmlns:UNS1="http://www.microsoft.com/windows/test/testdevice/11/2005" xmlns:PNPX="http://schemas.microsoft.com/windows/pnpx/2005/10" xmlns:wstransfer="http://schemas.xmlsoap.org/ws/2004/09/transfer" xmlns:df="http://schemas.microsoft.com/windows/2008/09/devicefoundation" xmlns:addressing="http://schemas.xmlsoap.org/ws/2004/08/addressing">
  <SOAP-ENV:Header>
    <wsa:MessageID>urn:uuid:1be51ddb-b394-fe33-b618-1294796eb78a</wsa:MessageID>
    <wsa:To>urn:uuid:3ae3095c-02c9-4ca8-bf94-50e7eb3784fb</wsa:To>
    <wsa:Action>http://schemas.xmlsoap.org/ws/2004/09/transfer/Get</wsa:Action>
  </SOAP-ENV:Header>
  <SOAP-ENV:Body>
    <SOAP-ENV:Fault>
      <SOAP-ENV:Code>
        <SOAP-ENV:Value>SOAP-ENV:Sender</SOAP-ENV:Value>
        <SOAP-ENV:Subcode>
          <SOAP-ENV:Value>addressing:InvalidMessageInformationHeader</SOAP-ENV:Value>
        </SOAP-ENV:Subcode>
      </SOAP-ENV:Code>
      <SOAP-ENV:Reason>
        <SOAP-ENV:Text xml:lang="en">ReplyTo header element is not valid.</SOAP-ENV:Text>
      </SOAP-ENV:Reason>
    </SOAP-ENV:Fault>
  </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```

Apparently it doesn't like the ReplyTo element, but I'm not sure how I could find out how a correct element would look like.

Do you know any tools or documentation I could checkout?

alexpevzner, 2021-05-21:
May I ask you to gather protocol trace, as it is much more informative, that console log?

To do so, please uncomment the following lines in the `/etc/sane.d/airscan.conf` file:
```
[debug]
trace = ~/airscan/trace
```

Traces will be in the `~/airscan/trace` directory, which will be created automatically. The interesting files for this case are `*zeroconf.log`

pschichtel, 2021-05-21:
This was generated when I ran airscan-discover with -t, seems to be what you are looking for.

[airscan-discover-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/6525324/airscan-discover-zeroconf.log)

alexpevzner, 2021-05-21:
ReplyTo is actually correct: https://docs.microsoft.com/en-us/windows/win32/wsdapi/get--metadata-exchange--http-request-and-message

But some firmwares erroneously require XML namespace prefixes to be exactly `soap` and `wsa`, failing to handle `xmlns` correctly. I'm a way too sleepy today (it's about 2.00am here in Moscow), so will try to patch it tomorrow. If you want to try by yourself, this is `wsdd_get_metadata_template` variable in the `airscan-wsdd.c` file.

closed 2021-05-22 by alexpevzner

pschichtel, 2021-05-22:
I did some scan tests yesterday with my patched version. Discovery works well and so does scanning. The scanner supports 1200dpi over USB, but seems to only expose 600dpi over WSD. I guess there is nothing that can be done about that here. Apart from that scanning seems to work perfectly fine (tested 600dpi duplex color scanning, worked as expected), so I I guess the model could be added to the list of know good devices with no for eSCL and yes for WSD

alexpevzner, 2021-05-23:
I've updated the README, thanks!

PR #148 (2021-05-22): Use usual namespace names in wsdd
This closes #147 

thanks @alexpevzner !