alexpevzner/sane-airscan #152: Help with Canon_iR-ADV_C3520

x09, 2021-06-24:
Hi i have Canon_iR-ADV_C3520 in different network.
My ip is a.b.c.d , canon ip address is a.x.y.z

open ports on canon below
```
PORT     STATE SERVICE
25/tcp   open  smtp
80/tcp   open  http
443/tcp  open  https
515/tcp  open  printer
8000/tcp open  http-alt
8443/tcp open  https-alt
9007/tcp open  unknown
9014/tcp open  unknown
9015/tcp open  unknown
9017/tcp open  unknown
9018/tcp open  unknown
9019/tcp open  unknown
9021/tcp open  panagolin-ident
9022/tcp open  paragent
9100/tcp open  jetdirect
```

airsan-dicovery not help me. I tried edit config manually

[image]

but no success..

x09, 2021-06-24:
WSD scan enabled

[image]

/etc/sane.d/airscan.conf

```
[devices]
#"1C3520" = http://10.61.50.200:9007/eSCL,eSCL
#"2C3520" = http://10.61.50.200:9014/eSCL,eSCL
#"3C3520" = http://10.61.50.200:9015/eSCL,eSCL
#"4C3520" = http://10.61.50.200:9017/eSCL,eSCL
#"5C3520" = http://10.61.50.200:9018/eSCL,eSCL
#"6C3520" = http://10.61.50.200:9019/eSCL,eSCL
#"7C3520" = http://10.61.50.200:9021/eSCL,eSCL
#"8C3520" = http://10.61.50.200:9022/eSCL,eSCL
#"c35200" = http://10.61.50.200:53048/, WSD
#"c35201" = http://10.61.50.200:53214/, WSD
"c35201" = http://10.61.50.200:19100/, WSD
"c35202" = http://10.61.50.200:47545/, WSD
"c35203" = http://10.61.50.200:47575/, WSD
"c35204" = http://10.61.50.200:53214/, WSD

[options]
discovery=disable

[debug]
enable = true
```

sane-find-scanners say nothing again..
what i can do wrong?

alexpevzner, 2024-11-18:
This is very hard to guess the proper WSD URL, because you need to guess not only the TCP port number, but HTTP path as well. Unfortunately, unlike eSCL, it is not standardized.

Devices manuals also rarely contain this information.

So my recommendation is to use `airscan-discover` (or sniff the Windows traffic, which is harder). You may need to temporary connect your PC to the local network where your scanner is connected.

I understand, that in the corporative environment it may be hard to do and may require coordination with the IT department...

closed 2024-11-18 by alexpevzner