alexpevzner/sane-airscan #154: Brother  MFC-L2700DW WSD not working

degenkolb, 2021-06-25:
Hi, 

in your Readme it says WSD is working for the MFC-L2700DW series but I don't get it to work. 

airscan-discover -d mentions the IP of my scanner but apparently fails to find the WSD port. 

Can you tell me whats wrong? On the device in the admin web interface under Network > Protocol "Network Scan" is checked. 

An nmap on the device shows the following open ports 

```
$ nmap -p 1-65000 192.168.1.11
Starting Nmap 7.80 ( https://nmap.org ) at 2021-06-25 22:08 CEST
Nmap scan report for 192.168.1.11
Host is up (0.060s latency).
Not shown: 64991 closed ports
PORT      STATE SERVICE
23/tcp    open  telnet
80/tcp    open  http
443/tcp   open  https
515/tcp   open  printer
631/tcp   open  ipp
9100/tcp  open  jetdirect
54921/tcp open  unknown
54922/tcp open  unknown
54923/tcp open  unknown

Nmap done: 1 IP address (1 host up) scanned in 45.51 seconds
```
and the content of the airscan-discover-zeroconf.log is 

```
00:00:00.001: zeroconf: zeroconf configuration:
00:00:00.001: zeroconf:   discovery    = enable
00:00:00.001: zeroconf:   model        = network
00:00:00.001: zeroconf:   protocol     = manual
00:00:00.001: zeroconf:   ws-discovery = full
00:00:00.010: MDNS: AVAHI_CLIENT_S_RUNNING
00:00:00.019: WSDD: netif addresses update:
00:00:00.019: WSDD:  + 192.168.1.104
00:00:00.019: WSDD:  + fe80::4e15:d315:b09c:fa6d%2
00:00:00.020: WSDD: 192.168.1.104: started discovery, UDP port=53725
00:00:00.020: WSDD: probe sent: 192.168.1.104:53725->239.255.255.250:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:103e1687-c769-dc01-0005-4f075083b56c</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.022: WSDD: fe80::4e15:d315:b09c:fa6d: started discovery, UDP port=49422
00:00:00.023: WSDD: probe sent: [fe80::4e15:d315:b09c:fa6d%2]:49422->[ff02::c]:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:72b29276-85cd-2ecc-199d-28e59c47e5f7</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>
[... 431 lines cut ...]
```
$ apt show sane-airscan 
Package: sane-airscan
Version: 0.99.24-1
Priority: optional
Section: graphics
Maintainer: Debian Printing Team <debian-printing@lists.debian.org>
Installed-Size: 371 kB
Depends: libavahi-client3 (>= 0.6.16), libavahi-common3 (>= 0.6.16), libc6 (>= 2.29), libgnutls30 (>= 3.7.0), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libxml2 (>= 2.7.4)
Recommends: ipp-usb, sane-utils
Homepage: https://github.com/alexpevzner/sane-airscan
Download-Size: 97,8 kB
APT-Manual-Installed: yes
APT-Sources: http://mirrordirector.raspbian.org/raspbian bullseye/main armhf Packages
```
I already tried different device strings in the form of 
```
[devices]
"MFC-L2700DW 54921" = http://192.168.1.11:54921, WSD
"MFC-L2700DW 54922" = http://192.168.1.11:54922, WSD
"MFC-L2700DW 54923" = http://192.168.1.11:54923, WSD
```
but nothing worked. 

Let me know what else I should check.

degenkolb, 2021-06-25:
An update

with 
```
$ apt show sane-airscan 
Package: sane-airscan
Version: 0.99.26-1+86.6
Status: install ok installed
Priority: optional
Section: graphics
Maintainer: Alexander Pevzner <pzz@apevzner.com>
Installed-Size: 362 kB
Depends: libavahi-client3 (>= 0.6.16), libavahi-common3 (>= 0.6.16), libc6 (>= 2.17), libgnutls30 (>= 3.6.6), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libxml2 (>= 2.7.4), zlib1g (>= 1:1.1.4)
Homepage: https://github.com/alexpevzner/sane-airscan
Download-Size: unknown
APT-Manual-Installed: yes
APT-Sources: /var/lib/dpkg/status
Description: SANE backend for MFP and document scanners that
 implements either eSCL (AirScan/AirPrint scanning) or WSD "driverless"
 scanning protocol
```

I get the following airscan-discover-zeroconf.log
```
00:00:00.001: zeroconf: zeroconf configuration:
00:00:00.001: zeroconf:   discovery    = enable
00:00:00.001: zeroconf:   model        = network
00:00:00.001: zeroconf:   protocol     = manual
00:00:00.001: zeroconf:   ws-discovery = full
00:00:00.010: MDNS: AVAHI_CLIENT_S_RUNNING
00:00:00.019: WSDD: netif addresses update:
00:00:00.019: WSDD:  + 192.168.1.104
00:00:00.019: WSDD:  + fe80::4e15:d315:b09c:fa6d%2
00:00:00.019: WSDD: 192.168.1.104: started discovery, UDP port=49210
00:00:00.019: WSDD: probe sent: 192.168.1.104:49210->239.255.255.250:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:aad42b22-9a02-899e-9060-322526a7d2bf</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.021: WSDD: fe80::4e15:d315:b09c:fa6d: started discovery, UDP port=34594
00:00:00.021: WSDD: probe sent: [fe80::4e15:d315:b09c:fa6d%2]:34594->[ff02::c]:3702
<s:Envelope xmlns:a="http://schemas.xmlsoap.org/ws/2004/08/addressing" xmlns:d="http://schemas.xmlsoap.org/ws/2005/04/discovery" xmlns:s="http://www.w3.org/2003/05/soap-envelope" xmlns:wsdp="http://schemas.xmlsoap.org/ws/2006/02/devprof">
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:c0caf00f-b96e-9b28-f514-4353e38249cf</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:00.025: zeroconf: zeroconf_device_list_get: requested
00:00:00.025: zeroconf: device_list wait: requested
00:00:00.025: zeroconf: device_list wait: DNS-SD not finished...
00:00:00.026: MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
00:00:00.026: MDNS: browse-ipv4("_ipp._tcp"): "Brother MFC-L2700DW series"
00:00:00.032: MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
00:00:00.033: MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
[... 392 lines cut ...]
  <s:Header>
    <a:Action>http://schemas.xmlsoap.org/ws/2005/04/discovery/Probe</a:Action>
    <a:MessageID>urn:uuid:2f5684c7-3140-47f2-d3a9-6cf57ec66af2</a:MessageID>
    <a:To>urn:schemas-xmlsoap-org:ws:2005:04:discovery</a:To>
  </s:Header>
  <s:Body>
    <d:Probe>
      <d:Types>wsdp:Device</d:Types>
    </d:Probe>
  </s:Body>
</s:Envelope>

00:00:02.535: WSDD: 192.168.1.104: done discovery
00:00:02.536: WSDD: fe80::4e15:d315:b09c:fa6d: done discovery
00:00:02.536: zeroconf: ZEROCONF_WSD: initial scan finished
00:00:02.536: zeroconf: device_list wait: OK
00:00:02.536: zeroconf: zeroconf_device_list_get: building list of devices
00:00:02.537: zeroconf: Brother MFC-L2700DW series (0): can: none, use: none
00:00:02.537: zeroconf: Brother MFC-L2700DW series (0): skipping, none of supported protocols discovered
00:00:02.537: zeroconf: zeroconf_device_list_get: resulting list:
00:00:02.537: WSDD: HTTP POST http://192.168.1.11/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: Cancelled
00:00:02.543: zeroconf: device gone urn:uuid:e3248000-80ce-11db-8000-30055c5a88dd
00:00:02.543: zeroconf:   method:    ZEROCONF_MDNS_HINT
00:00:02.543: zeroconf:   interface: 2 (enxb827eb50fe0f)
```

degenkolb, 2021-06-25:
And with 

```
$ apt show sane-airscan-unstable
Package: sane-airscan-unstable
Version: 1620822232-1+21.6
Status: install ok installed
Priority: optional
Section: graphics
Maintainer: Alexander Pevzner <pzz@apevzner.com>
Installed-Size: 371 kB
Depends: libavahi-client3 (>= 0.6.16), libavahi-common3 (>= 0.6.16), libc6 (>= 2.17), libgnutls30 (>= 3.6.6), libjpeg62-turbo (>= 1.3.1), libpng16-16 (>= 1.6.2-1), libtiff5 (>= 4.0.3), libxml2 (>= 2.7.4), zlib1g (>= 1:1.1.4)
Homepage: https://github.com/alexpevzner/sane-airscan-unstable

```

this

jomority, 2023-04-21:
Hey, I have the Brother MFC-L2700DN (which is similar except it doesn't have WiFi), and I had the same problem.

What solved it for me was enabling _Web Services_ in the printer's web console under Network -> Protocol and the restarting the printer.