alexpevzner/sane-airscan #159: Brother MFC-8710DW results and observations, resolutions and features

markosjal, 2021-07-25:
I have been testing with a Brother MFC-8710DW which is picked up as a WSD device.

Unfortunately this driver provides me only a maximum of 300 dpi 
`Options specific to device `airscan:w2:Brother MFC-8710DW':
  Standard:
    --resolution 100|200|300dpi [300]
        Sets the resolution of the scanned image.
    --mode Color|Gray [Color]
        Selects the scan mode (e.g., lineart, monochrome, or color).
    --source Flatbed|ADF [Flatbed]
        Selects the scan source (such as a document-feeder).
  Geometry:
    -l 0..215.9mm [0]
        Top-left x position of scan area.
    -t 0..355.6mm [0]
        Top-left y position of scan area.
    -x 0..215.9mm [215.9]
        Width of scan-area.
    -y 0..355.6mm [355.6]
        Height of scan-area.
  Enhancement:
    --brightness -100..100% (in steps of 1) [0]
        Controls the brightness of the acquired image.
    --contrast -100..100% (in steps of 1) [0]
        Controls the contrast of the acquired image.
    --shadow 0..100% (in steps of 1) [0]
        Selects what radiance level should be considered "black".
    --highlight 0..100% (in steps of 1) [100]
        Selects what radiance level should be considered "white".
    --analog-gamma 0.0999908..4 [1]
        Analog gamma-correction
    --negative[=(yes|no)] [no]
        Swap black and white
`
I have confirmed this in the XSane GUI as well

There is a Brother Proprietary driver (that could easily go obsolete with the next version of linux I install) , that shows the following

`Options specific to device `brother4:net1;dev0':
  Mode:
    --mode Black & White|Gray[Error Diffusion]|True Gray|24bit Color|24bit Color[Fast] [24bit Color[Fast]]
        Select the scan mode
    --resolution 100|150|200|300|400|600|1200|2400|4800|9600dpi [200]
        Sets the resolution of the scanned image.
    --source FlatBed|Automatic Document Feeder(left aligned)|Automatic Document Feeder(centrally aligned) [Automatic Document Feeder(left aligned)]
        Selects the scan source (such as a document-feeder).
    --brightness -50..50% (in steps of 1) [inactive]
        Controls the brightness of the acquired image.
    --contrast -50..50% (in steps of 1) [inactive]
        Controls the contrast of the acquired image.
  Geometry:
    -l 0..211.9mm (in steps of 0.0999908) [0]
        Top-left x position of scan area.
    -t 0..355.6mm (in steps of 0.0999908) [0]
        Top-left y position of scan area.
    -x 0..211.9mm (in steps of 0.0999908) [211.881]
        Width of scan-area.
    -y 0..355.6mm (in steps of 0.0999908) [355.567]
        Height of scan-area.`

another item worth noting and not sure how bug of a deal it is but notice the naming difference in "ADF" vs "Automatic Document Feeder"

schdro, 2023-03-19:
Same for Brother _MFC-L2750DW_.

For that latter:
- Both WSD+eSCL supported. Toggling via config.
- WSD offers only max 300dpi, while eSCL offers at least max 600dpi. From official spec, at least flatbed should support 1200dpi.
- According to enabled trace log: Already the `<scan:ScannerCapabilities>...<scan:SupportedResolutions>` XML tree, as retrieved from the printer, publishes those max discrete scan resolutions as supported. Meanwhile, `<scan:MaxOpticalXResolution>` publishes a maximum optical scanner resolution of 2400dpi.

Bottomline:

1. Better try all modes via config (e.g. also **eSCL**).
2. Might need some network logging to check how proprietary Brother driver drives higher resolutions (like special format encoding). On _printer_ side, IMHO the proprietary driver encodes an BRScript binary stream before passing via IPP -- maybe something comparable for scanner interface. But currently, from trace log, unfortunately would doubt that higher resolutions are possible with "universal" driver.

schdro, 2023-03-19:
After further short checking of my own MFC-L2750DW:

Proprietary driver does not seem to use eSCL but dedicated TCP port 54921, for sending proprietary communication with some plain string providing higher resolution to fetch e.g. JPEG scan. Port 54921 is also listed on Brother help page for requiring firewall exception sometimes, e.g. if the Brother proprietary scan tool has problems.

Looks as if certain Brother MFCs exhibit highest possible scan resolutions only via proprietary (closed?) protocol, not the ones (eSCL, WSD) implemented by open source sane-airscan.

schdro, 2023-03-19:
Parallel git works like [github/mfc-j430w](https://github.com/v0lp3/mfc-j430w) and forks exist  for trying to reverse engineer port 54921 protocol. Setup strings there (e.g. for setting resolution like 1200dpi) seem to match what is captured in traffic logs of proprietary driver.

Restricted Brother resolution issue for eSCL/WSD here out of scope? Should be either requested to printer vendor for fully fledged open protocol support, or tracked in separate protocol driver.

schdro, 2023-03-19:
Another small addon:

sane has an builtin backend "escl", which is also designated for recent Brother MFC devices. It works e.g. for scanning with MFC-L2750, but obviously also uses eSCL, and therefore as expected suffers from similar restrictions as sane-airprint.

alexpevzner, 2023-03-19:
Hi guys,

this is known thing, that some devices expose restricted functionality to eSCL/WDS interfaces in comparison with their "native" (proprietary) protocols.

This may be either because some vendors a quite "lazy" to implement fill functionality driverlessly, considering these interfaces as a second class while proprietary protocol considered the first class, or because the proprietary driver implements some nontrivial processing which is hard to implement on a device due to hardware limitations (especially for Brother - their hardware is **very** limited).

I believe that eSCL scanning will eventually win and all proprietary interfaces sooner or later will go away. Apple don't want to deal with printer/scanners proprietary drivers and Apple's users is the important market for printer vendors. I believe this is a matter of several (few) years.

But for now we have whatever we have and have to live with it.