alexpevzner/sane-airscan #162: Using WSD seems to always request the entire scan surface regardless of selected scan region (at least on HP DeskJet 5000)

cipriancraciun, 2021-08-02:
I have an wireless "HP DeskJet All-in-One 5000 Series" (actually 5075) that supports both WSD and eSCL, and from my limited experiments I've seen that when using WSD, `airscan` seems to **always explicitly request the entire scan surface** regardless of the requested scan region.

I base my conclusions on the following:
* I've used `scanimage` with and without `-x 100 -y 100` (or some other values) at various DPI's (300, 600, 1200);
* I've used Wireshark to inspect the SOAP exchanges with the printer, and it seems that `airscan` always requests for the entire scan surface (i.e. larger than my requested region, and regardless of it);
* the scanning time is the same (at the same DPI) regardless of the requested scan region (including different heights);
* there is a "bug" in this HP model (when using eSCL or `hpaio`) so that when one asks anything less than the entire scan area, it will try to do edge detection, and thus apply an extra crop followed by an internal software resize to match the expected pixel size;  in comparison all images scanned via WSD seem not to have this bug;

alexpevzner, 2021-08-02:
Hi @cipriancraciun,

my Kyocera ECOSYS M2040dn incorrectly handles scan region in the WSD mode, so really, sane-airscan in the WSD mode always requests the entire scan surface and then performs image cropping in software. In the eSCL mode all known so far devices handle scan region correctly, so in this mode sane-airscan relies on hardware cropping,

I guess, your problem will be fixed, if on the eSCL mode sane-airscan also will rely on software cropping, at least for your device, correct?

cipriancraciun, 2021-08-04:
> I guess, your problem will be fixed, if on the eSCL mode sane-airscan also will rely on software cropping, at least for your device, correct?

@alexpevzner, I understand there are thousands of scanners out there, each with its own quirks and bugs, thus I don't think there is an "optimal" / "best" solution that would cover all.

However I personally would take the following approach:
* have `airscan` always do the "correct" thing by default, according to the documentation (i.e. always use scan regions both in WSD and eSCL);
* however, have some configuration items (per device, and per global) that allows one to "tweak" the behaviour for broken devices;  (in this case for example with your Kyocera in WSD, and my HP in any mode, we would always scan the entire region and do cropping in software;)

closed 2024-11-18 by alexpevzner