alexpevzner/sane-airscan #163: Kyocera ECOSYS M6526cdn : I/O Error due to error `The TLS connection was non-properly terminated`

ericschdt, 2021-08-05:
### Setup:

- System: Fedora 34
- sane-airscan: 0.99.26 (release 1.fc34)
- sane-backends 1.0.32
- Printer: Kyocera ECOSYS M6526cdn

### What did I do
Both my laptop and my printer/scanner are on the same subnet. I tried to scan via
```
SANE_DEBUG_DLL=3 scanimage -x 100 -y 100 --format=tiff >image.tiff
```

### Result
The scanner received the scan job, it turns on, the scanner light starts moving and the display states
> Functions are used remotely...

Then scanimage finally breaks up saying
```
00:00:03.073: Kyocera ECOSYS M6526cdn: cancelling job due to error: The TLS connection was non-properly terminated.
00:00:03.073: Kyocera ECOSYS M6526cdn: JOB status=Error during device I/O
```

and the scanner remains in its `Functions are used remotely...` state for a while until finally going into standby again.

### Traces and logs

[scanimage-Kyocera-ECOSYS-M6526cdn.log](https://github.com/alexpevzner/sane-airscan/files/6938532/scanimage-Kyocera-ECOSYS-M6526cdn.log)
[scanimage-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/6938533/scanimage-zeroconf.log)
[sane_command_debug3.txt](https://github.com/alexpevzner/sane-airscan/files/6938534/sane_command_debug3.txt)

ericschdt, 2021-08-05:
It works fine* when disabling `eSCL over SSL` and use plain `eSCL` at the scanner instead at least with `XSane` and `simple-scan`.

alexpevzner, 2022-05-05:
Hi @ericschdt,

sorry for long delay in reply.

I've attempted to create a work around for the problem with TLS that you've reported. If you are able to build `sane-airscan` from sources and retest, I will appreciate your feedback.

ericschdt, 2022-05-07:
> sorry for long delay in reply.

Don't worry. It's working with WSD fine so far. :)

> I've attempted to create a work around for the problem with TLS that you've reported. If you are able to build `sane-airscan` from sources and retest, I will appreciate your feedback.

Sure, but this will take some weeks until I will have the chance to test it again, but the chance will definitely come.

alexpevzner, 2022-05-07:
OK, thanks.

At this case, I'll close the issue for now. Feel free to reopen it, if the problem actually remains.

closed 2022-05-07 by alexpevzner