alexpevzner/sane-airscan #169: Samsung SCX3400 Compatibility

Offpics, 2021-08-16:
I've tested `scanimage` on Samsung SCX3400, it works. I would like to add it to the compatibility table, but I'm not sure how to test in what mode it operates.

alexpevzner, 2021-10-04:
Please, show me an output of `scanimage -L` command

Offpics, 2021-10-04:
```device `airscan:w0:Samsung Electronics Co., Ltd. SCX-3400 Series' is a WSD Samsung Electronics Co., Ltd. SCX-3400 Series ip=192.168.0.30```

Thermi, 2021-11-21:
What firmware version is it? I have an SCX-3400FW and sane-airscan reports errors while trying to discover its properties, making it unusable. I also have a pcap of the complete communication with it. I filtered for the conversation between my host and the printer. A zipped pcapng file of it is attached, I hope it helps in finding out why this particular mfp doesn't actually work for me, but for you. The output of `airscan-discover -d -t` is below. 

[airscan-scx-3400.pcapng.zip](https://github.com/alexpevzner/sane-airscan/files/7575668/airscan-scx-3400.pcapng.zip)

My SCX-3400FW is flashed with the following firmwares:
Main Firmware Version V3.00.01.08 NOV-21-2011
Network Firmware Version V6.01.00(SCX-3400 Series)
IP Core Software Version 22.01.NW Stack Version
Engine Firmware Version V1.00.06 11-14-2011
SPL Firmware Version SPL 5.51 03-28-2011

```
└──╼ airscan-discover -d -t
MDNS: AVAHI_CLIENT_CONNECTING
WSDD: netif addresses update
WSDD:  + 192.168.178.20
WSDD:  + 192.168.120.1
WSDD:  + 172.17.0.1
WSDD: 192.168.178.20: started discovery, UDP port=47351
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: 192.168.120.1: started discovery, UDP port=54456
WSDD: 172.17.0.1: started discovery, UDP port=44440
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
WSDD: HTTP POST http://192.168.178.28:8018/wsd
WSDD: HTTP resolving 192.168.178.28 8018
WSDD: HTTP trying 192.168.178.28:8018
WSDD: HTTP fdpoll: {--} -> {-w}
WSDD: HTTP 675 bytes sent
WSDD: HTTP done request sending
WSDD: HTTP fdpoll: {-w} -> {rw}
WSDD: HTTP fdpoll: {rw} -> {r-}
WSDD: HTTP 1460 bytes received
WSDD: HTTP POST http://192.168.178.28:8018/wsd: got response headers (200)
WSDD: HTTP 1460 bytes received
WSDD: HTTP 41 bytes received
WSDD: HTTP done response reception 
WSDD: HTTP POST http://192.168.178.28:8018/wsd: 200 OK
WSDD: HTTP POST http://192.168.178.28:8018/wsd: OK
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv4 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": IPv6 address expected but not yet discovered
WSDD: "Samsung Electronics Co., Ltd. SCX-3400 Series": publish-delay timer expired
zeroconf: found urn:uuid:16a65700-007c-1000-bb49-0015999e4025
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp1s0
zeroconf:   name
zeroconf:   model:     Samsung Electronics Co., Ltd. SCX-3400 Series
zeroconf:   addresses
zeroconf:   protocol:  WSD
zeroconf:   endpoints
zeroconf:   device:    0000 (created
zeroconf: device_list wait: DNS-SD not finished
WSDD: 172.17.0.1: done discovery
WSDD: 192.168.120.1: done discovery
WSDD: 192.168.178.20: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
zeroconf: initial scan timer expired
zeroconf: device_list wait: DNS-SD not finished...
zeroconf: device_list wait: timeout
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: Samsung Electronics Co., Ltd. SCX-3400 Series (0): can: none, use: none
zeroconf: Samsung Electronics Co., Ltd. SCX-3400 Series (0): skipping, none of supported protocols discovered
zeroconf: zeroconf_device_list_get: resulting list:
[devices
zeroconf: device gone urn:uuid:16a65700-007c-1000-bb49-0015999e4025
zeroconf:   method:    ZEROCONF_WSD
zeroconf:   interface: 2 (enp1s0
```

alexpevzner, 2026-03-10:
Samsung SCX3400 already in the list...