alexpevzner/sane-airscan #170: Samsung SL M2875FD Error during device I/O

diabolusss, 2021-08-21:
Hello!
I'm trying to get this mfd printer to work over lan, but i'm completely stuck with a bunch of errors.
First, to get my printer recognized i had to start avahi-daemon
`systemctl start avahi-daemon`

After this step it gets recognized as escl device, which, as i assume, is wrong, because it selects Sync Thru WebService port :80 and fails with out of memory error:
```
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.1.218
WSDD:  + fe80::aa5e:45ff:fe38:3f1d%2
WSDD:  + 10.160.195.129%2
WSDD:  + fe80::216:3eff:fec7:3df0%4
WSDD: 192.168.1.218: started discovery, UDP port=46003
WSDD: fe80::aa5e:45ff:fe38:3f1d: started discovery, UDP port=56955
WSDD: 10.160.195.129: started discovery, UDP port=35267
WSDD: fe80::216:3eff:fec7:3df0: started discovery, UDP port=33995
API: sane_get_devices(): called
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: WSDD not finished...
WSDD: fe80::216:3eff:fec7:3df0: done discovery
WSDD: fe80::aa5e:45ff:fe38:3f1d: done discovery
WSDD: 10.160.195.129: done discovery
WSDD: 192.168.1.218: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   WSD   "w0:Samsung SL M2875FD"
API: sane_get_devices(): done
file.xml:1: parser error : Space required after the Public Identifier
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:1: parser error : SystemLiteral " or ' expected
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:12: parser error : Opening and ending tag mismatch: CENTER line 0 and BODY
</BODY>
       ^
file.xml:14: parser error : Opening and ending tag mismatch: BODY line 0 and HTML
</HTML>
       ^
file.xml:16: parser error : EndTag: '</' not found

^
scanimage: open of device escl:http://192.168.1.111:80 failed: Out of memory
Type ``scanimage --help -d DEVICE'' to get list of all options for DEVICE.

zeroconf: initial scan timer expired
[... 129 lines cut ...]
Port Number:    8018
Service Location Protocol (SLP)
SLP Protocol:    Enabled
Port Number:    427
Simple Service Discovery Protocol (SSDP)
SSDP Protocol:    Enabled
Port Number:    1900
Friendly Name:    Samsung M267x 287x Series (192.168.1.111)
Multicast DNS (mDNS)
mDNS Protocol:   Enabled
Port Number:    5353
mDNS Name:    SEC84251.local.
Friendly Name:    Samsung M267x 287x Series (SEC84251)
Simple Network Management Protocol (SNMP)
SNMPv1/v2 Protocol:    Enabled
SNMPv3 Protocol:    Disabled
HTTP
HTTP Protocol:    Enabled
Port Number:    80
Proxy
HTTP Proxy:    Disabled
HTTP Proxy Server Address:    Not Configured
Port Number:    Not Configured
```
Any help will be appreciated.

diabolusss, 2021-08-23:
Xerox_mfp driver and sane debug log:
```
$ export SANE_DEBUG_DLL=255 && export SANE_DEBUG_XEROX_MFP=4 && scanimage -L
[09:22:12.865644] [sanei_debug] Setting debug level of dll to 255.
[09:22:12.865721] [dll] sane_init: SANE dll backend version 1.0.13 from sane-backends 1.0.32
[09:22:12.865736] [dll] sane_init/read_dlld: attempting to open directory `./dll.d'
[09:22:12.865749] [dll] sane_init/read_dlld: attempting to open directory `/etc/sane.d/dll.d'
[09:22:12.865763] [dll] sane_init/read_dlld: using config directory `/etc/sane.d/dll.d'
[09:22:12.865788] [dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/utsushi
[09:22:12.865807] [dll] sane_init/read_config: reading dll.d/utsushi
[09:22:12.865828] [dll] add_backend: adding backend `utsushi'
[09:22:12.865841] [dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/airscan
[09:22:12.865855] [dll] sane_init/read_config: reading dll.d/airscan
[09:22:12.865868] [dll] add_backend: adding backend `airscan'
[09:22:12.865881] [dll] sane_init/read_dlld: done.
[09:22:12.865898] [dll] sane_init/read_config: reading dll.conf
[09:22:12.865919] [dll] add_backend: adding backend `escl'
[09:22:12.865928] [dll] add_backend: adding backend `hp'
[09:22:12.865935] [dll] add_backend: adding backend `hp3500'
[09:22:12.865945] [dll] add_backend: adding backend `hp3900'
[09:22:12.865954] [dll] add_backend: adding backend `hp4200'
[09:22:12.865962] [dll] add_backend: adding backend `hp5400'
[09:22:12.865971] [dll] add_backend: adding backend `hp5590'
[09:22:12.865993] [dll] add_backend: adding backend `hpljm1005'
[09:22:12.866002] [dll] add_backend: adding backend `hpsj5s'
[09:22:12.866011] [dll] add_backend: adding backend `hs2p'
[09:22:12.866027] [dll] add_backend: adding backend `pint'
[09:22:12.866042] [dll] add_backend: adding backend `xerox_mfp'
[09:22:12.866051] [dll] add_backend: adding backend `smfp'
[09:22:12.866070] [dll] sane_get_devices
[09:22:12.866080] [dll] load: searching backend `smfp' in `/usr/lib/sane'
[09:22:12.866088] [dll] load: trying to load `/usr/lib/sane/libsane-smfp.so.1'
[09:22:12.866105] [dll] load: dlopen()ing `/usr/lib/sane/libsane-smfp.so.1'
[09:22:12.867982] [dll] init: initializing backend `smfp'
[09:22:12.880831] [dll] init: backend `smfp' is version 1.0.1
[09:22:13.882132] [dll] load: searching backend `xerox_mfp' in `/usr/lib/sane'
[09:22:13.882196] [dll] load: trying to load `/usr/lib/sane/libsane-xerox_mfp.so.1'
[09:22:13.882219] [dll] load: dlopen()ing `/usr/lib/sane/libsane-xerox_mfp.so.1'
[09:22:13.882651] [dll] init: initializing backend `xerox_mfp'
[09:22:13.882681] [sanei_debug] Setting debug level of xerox_mfp to 4.
[09:22:13.882690] [xerox_mfp] sane_init: Xerox backend (build 13), version != null, authorize != null
[09:22:13.891292] [dll] init: backend `xerox_mfp' is version 1.0.13
[09:22:13.891320] [xerox_mfp] sane_xerox_mfp_get_devices: 0x7ffeab904220, 0
[09:22:13.891428] [xerox_mfp] list_one_device: tcp 192.168.1.111
[09:22:13.891446] [xerox_mfp] tcp_dev_open: open tcp 192.168.1.111
[09:22:13.894179] [xerox_mfp] :: dev_command(INQUIRY[0x12], 70)
[09:22:13.894281] [xerox_mfp] tcp_dev_request: wait for 1024 bytes
[09:22:14.971689] [xerox_mfp] tcp_dev_request: error Resource temporarily unavailable, bytes requested: 1024, bytes read: 0
[09:22:14.971755] [xerox_mfp] dev_command: illegal response len 0, need 70
[09:22:14.971766] [xerox_mfp] tcp_dev_close: closing dev 0x5649ead50220
[09:22:14.971884] [xerox_mfp] list_one_device: dev_inquiry(tcp 192.168.1.111): Error during device I/O
[09:22:14.971975] [dll] load: searching backend `pint' in `/usr/lib/sane'
[09:22:14.971992] [dll] load: trying to load `/usr/lib/sane/libsane-pint.so.1'
[09:22:14.972013] [dll] load: couldn't open `/usr/lib/sane/libsane-pint.so.1' (No such file or directory)
[09:22:14.972023] [dll] load: couldn't find backend `pint' (No such file or directory)
[09:22:14.972042] [dll] load: searching backend `hs2p' in `/usr/lib/sane'
[09:22:14.972051] [dll] load: trying to load `/usr/lib/sane/libsane-hs2p.so.1'
[09:22:14.972073] [dll] load: dlopen()ing `/usr/lib/sane/libsane-hs2p.so.1'
[09:22:14.972361] [dll] init: initializing backend `hs2p'
[09:22:14.972927] [dll] init: backend `hs2p' is version 1.0.0
[09:22:14.972960] [dll] load: searching backend `hpsj5s' in `/usr/lib/sane'
[09:22:14.972970] [dll] load: trying to load `/usr/lib/sane/libsane-hpsj5s.so.1'
[09:22:14.972985] [dll] load: dlopen()ing `/usr/lib/sane/libsane-hpsj5s.so.1'
[09:22:14.973359] [dll] init: initializing backend `hpsj5s'
[09:22:14.973554] [dll] init: backend `hpsj5s' is version 1.0.3
[09:22:14.973575] [dll] load: searching backend `hpljm1005' in `/usr/lib/sane'
[09:22:14.973584] [dll] load: trying to load `/usr/lib/sane/libsane-hpljm1005.so.1'
[09:22:14.973599] [dll] load: dlopen()ing `/usr/lib/sane/libsane-hpljm1005.so.1'
[09:22:14.973895] [dll] init: initializing backend `hpljm1005'
[09:22:14.974028] [dll] init: backend `hpljm1005' is version 1.0.1
[... 86 lines cut ...]
[09:22:18.550034] [dll] init: backend `utsushi' is version 1.1.0
[09:22:18.663406] [dll] sane_get_devices: found 3 devices
device `escl:http://192.168.1.111:80' is a Samsung M267x 287x Series (192.168.1.111) flatbed or ADF scanner
device `airscan:w0:Samsung SL M2875FD:8018' is a WSD Samsung SL M2875FD:8018 ip=192.168.1.111
device `airscan:e1:Samsung SL M2875FD:9400' is a eSCL Samsung SL M2875FD:9400 ip=192.168.1.111
[09:22:18.663450] [dll] sane_exit: exiting
[... cut ...]
[09:22:18.664161] [dll] sane_exit: calling backend `hp3500's exit function
[09:22:18.664199] [dll] sane_exit: calling backend `hp's exit function
[09:22:18.664238] [dll] sane_exit: calling backend `escl's exit function
[09:22:18.664290] [dll] sane_exit: calling backend `airscan's exit function
API: sane_exit(): called
API: sane_exit(): OK
[09:22:18.668680] [dll] sane_exit: calling backend `utsushi's exit function
[09:22:18.668736] [dll] sane_exit: finished
```
I suppose it just doesn't know how to wake printer from sleep!? i'll check this later...

diabolusss, 2021-08-24:
Ful log while scanning:
```
$ export SANE_DEBUG_DLL=255 && export SANE_DEBUG_XEROX_MFP=4 && scanimage -d "xerox_mfp:tcp 192.168.1.111" --format=png -p -v --resolution=100 --mode=Lineart --source=ADF  -x 216 -y 297 --batch='./SCANx%d.png'
[12:49:41.747628] [sanei_debug] Setting debug level of dll to 255.
[12:49:41.747689] [dll] sane_init: SANE dll backend version 1.0.13 from sane-backends 1.0.32
[12:49:41.747704] [dll] sane_init/read_dlld: attempting to open directory `./dll.d'
[12:49:41.747718] [dll] sane_init/read_dlld: attempting to open directory `/etc/sane.d/dll.d'
[12:49:41.747733] [dll] sane_init/read_dlld: using config directory `/etc/sane.d/dll.d'
[12:49:41.747758] [dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/utsushi
[12:49:41.747776] [dll] sane_init/read_config: reading dll.d/utsushi
[12:49:41.747795] [dll] add_backend: adding backend `utsushi'
[12:49:41.747808] [dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/airscan
[12:49:41.747823] [dll] sane_init/read_config: reading dll.d/airscan
[12:49:41.747833] [dll] add_backend: adding backend `airscan'
[12:49:41.747847] [dll] sane_init/read_dlld: done.
[12:49:41.747861] [dll] sane_init/read_config: reading dll.conf
[12:49:41.747881] [dll] add_backend: adding backend `escl'
[12:49:41.747891] [dll] add_backend: adding backend `hp'
[12:49:41.747899] [dll] add_backend: adding backend `hp3500'
[12:49:41.747908] [dll] add_backend: adding backend `hp3900'
[12:49:41.747916] [dll] add_backend: adding backend `hp4200'
[12:49:41.747923] [dll] add_backend: adding backend `hp5400'
[12:49:41.747930] [dll] add_backend: adding backend `hp5590'
[12:49:41.747939] [dll] add_backend: adding backend `hpljm1005'
[12:49:41.747948] [dll] add_backend: adding backend `hpsj5s'
[12:49:41.747956] [dll] add_backend: adding backend `hs2p'
[12:49:41.747969] [dll] add_backend: adding backend `pint'
[12:49:41.747983] [dll] add_backend: adding backend `xerox_mfp'
[12:49:41.747991] [dll] add_backend: adding backend `smfp'
[12:49:41.748012] [dll] sane_open: trying to open `xerox_mfp:tcp 192.168.1.111'
[12:49:41.748023] [dll] load: searching backend `xerox_mfp' in `/usr/lib/sane'
[12:49:41.748032] [dll] load: trying to load `/usr/lib/sane/libsane-xerox_mfp.so.1'
[12:49:41.748047] [dll] load: dlopen()ing `/usr/lib/sane/libsane-xerox_mfp.so.1'
[12:49:41.748241] [dll] init: initializing backend `xerox_mfp'
[12:49:41.748256] [sanei_debug] Setting debug level of xerox_mfp to 4.
[12:49:41.748267] [xerox_mfp] sane_init: Xerox backend (build 13), version != null, authorize != null
[12:49:41.756988] [dll] init: backend `xerox_mfp' is version 1.0.13
[12:49:41.757028] [xerox_mfp] sane_xerox_mfp_open: 'tcp 192.168.1.111'
[12:49:41.757037] [xerox_mfp] sane_xerox_mfp_get_devices: (nil), 1
[12:49:41.757145] [xerox_mfp] list_one_device: tcp 192.168.1.111
[12:49:41.757156] [xerox_mfp] tcp_dev_open: open tcp 192.168.1.111
[12:49:41.759789] [xerox_mfp] :: dev_command(INQUIRY[0x12], 70)
[12:49:41.759887] [xerox_mfp] tcp_dev_request: wait for 1024 bytes
[12:49:42.951490] [xerox_mfp] tcp_dev_request: error Resource temporarily unavailable, bytes requested: 1024, bytes read: 70
[12:49:42.951560] [xerox_mfp] dev_inquiry: found Samsung/M267x 287x Series
[12:49:42.951575] [xerox_mfp] tcp_dev_close: closing dev 0x55aa3757caf0
[12:49:42.951732] [xerox_mfp] tcp_dev_open: open tcp 192.168.1.111
[12:49:42.953173] [dll] sane_open: open successful
[12:49:42.953244] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=0)
[12:49:42.953255] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 0
[12:49:42.953264] [dll] sane_control_option(handle=0x55aa37580a50,option=0,action=0,value=0x7ffe8e6b13cc,info=(nil))
[12:49:42.953272] [xerox_mfp] sane_xerox_mfp_control_option: 0x55aa3757caf0, 0, <0>, 0x7ffe8e6b13cc, (nil)
[12:49:42.953281] [xerox_mfp] sane_xerox_mfp_control_option: 0, <0> => 0000000b, 0
[12:49:42.953292] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=0)
[12:49:42.953300] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 0
[12:49:42.953312] [dll] sane_control_option(handle=0x55aa37580a50,option=0,action=0,value=0x7ffe8e6b12d0,info=(nil))
[12:49:42.953320] [xerox_mfp] sane_xerox_mfp_control_option: 0x55aa3757caf0, 0, <0>, 0x7ffe8e6b12d0, (nil)
[12:49:42.953328] [xerox_mfp] sane_xerox_mfp_control_option: 0, <0> => 0000000b, 0
[12:49:42.953341] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=1)
[12:49:42.953355] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 1
[12:49:42.953366] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=2)
[12:49:42.953376] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 2
[12:49:42.953392] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=3)
[12:49:42.953402] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 3
[12:49:42.953419] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=4)
[12:49:42.953431] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 4
[12:49:42.953442] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=5)
[12:49:42.953449] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 5
[12:49:42.953456] [dll] sane_get_option_descriptor(handle=0x55aa37580a50,option=6)
[12:49:42.953464] [xerox_mfp] sane_xerox_mfp_get_option_descriptor: 0x55aa3757caf0, 6
[... 243 lines cut ...]
[... cut ...]
[12:50:17.431823] [xerox_mfp] sane_xerox_mfp_cancel: 0x55aa3757caf0
[12:50:17.431833] [dll] sane_close(handle=0x55aa37580a50)
[12:50:17.431840] [xerox_mfp] sane_xerox_mfp_close: 0x55aa3757caf0 (tcp 192.168.1.111)
[12:50:17.431848] [xerox_mfp] tcp_dev_close: closing dev 0x55aa3757caf0
[12:50:17.431949] [dll] sane_exit: exiting
[12:50:17.431964] [dll] sane_exit: calling backend `xerox_mfp's exit function
[12:50:17.432106] [dll] sane_exit: finished
```
Still can't figure it out, how to make xerox_mfp to appear in scanner list...

alexpevzner, 2021-08-24:
Hi @diabolusss,

first of all, I see the following lines:
```
List of available devices:
    escl:http://192.168.1.111:80 escl:http://192.168.1.111:80
    airscan:w0:Samsung SL M2875FD
```

It means that your printer was recognized by 2 drivers: `sane-escl`, which supports only eSCL protocol, and `sane-airscan`, which supports eSCL and WSD. And sane-airscan recognizes your printer as WSD device.

Then, `scanimage` fails, trying to open device, exposed by `sane-escl`:
```
scanimage: open of device escl:http://192.168.1.111:80 failed: Out of memory
```

It is hard to say, why `scanimage` prefers device, exposed by `sane-escl`, not by `sane-airscan`. Probably, it happens because that device name appears first in the list. So you may try to give device's name to the `scanimage` directly (`w0:Samsung SL M2875FD`) or use some GUI app that allows to choose device out of the list.

Then about your changes in the `airscan.conf` file. Guessing WSD port is exceptionally hard. Typically, it is not 80, not 8080 and not printer's port. There are `airscan-discover` utility that comes with `sane-airscan`, it does it automatically and prints all the discovered devices in the format, suitable to adding to the `airscan.conf` file. But as I see in the first log that device was discovered, probably this step is not needed.

Also, I see a lot of error messages from the XML parser. Most likely, these messages come from the `sane-escl` driver, because `sane-airscan` attempts to hook `libxml` error messages, but here I cannot be 100% sure, because this stuff in `libxml` is quite messy.

It is hard to say, why `sane-airscan` doesn't discover the device as eSCL, while `sane-escl` does, without seeing the `sane-airscan` trace files (it is much more detailed log that log which goes to console). To enable protocol trace, the following lines in the `airscan.conf` file needs to be uncommented:
```
[debug]
trace = ~/airscan/trace
enable = true
```

Protocol trace files will be written to the `~/airscan/trace` directory, which will be created automatically.

So, my recommendations are following:
1, With unmodified `airscan.conf`, try to choose the `w0:Samsung SL M2875FD` device explicitly
2. If you got "device not found" error, write the following lines in the `airscan.conf` file and retry:
```
[options]
protocol = manual
```
Note, setting `protocol = manual` will expose WSD side of the device, even if shadowed by its eSCL instance (by default, if device supports both protocols, `sane-airscan` automatically prefers eSCL).
4. If more investigations are required, please enable protocol trace and come back with these files.

diabolusss, 2021-08-25:
Hello, @alexpevzner!
First of all, thenk you for checking this issue.

> It means that your printer was recognized by 2 drivers: `sane-escl`, which supports only eSCL protocol, and `sane-airscan`, which supports eSCL and WSD. And sane-airscan recognizes your printer as WSD device.

> 1. With unmodified `airscan.conf`, try to choose the `w0:Samsung SL M2875FD` device explicitly
> 2. If you got "device not found" error, write the following lines in the `airscan.conf` file and retry:

~~I don't know about escl side~~, (i've remembered, that it's static record, too) but second part, related to sane-aiscan, wasn't automatically recognized - i have statically added this line to airscan.conf, as i've previously [mentioned](https://github.com/alexpevzner/sane-airscan/issues/170#issue-976064963). What i see from tests, that it doesn't matter what i write there - it will be listed (and checked) anyway. airscan-discover finds nothing in any case.

> Guessing WSD port is exceptionally hard.

I believe there is nothing to guess, because it's listed in printer settings as [8018](https://github.com/alexpevzner/sane-airscan/issues/170#issue-976064963). But it just doesn't work. I suppose service can be hidden behind some url path, like in example "Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL, or like http://192.168.1.111:8018/ScannerCapabilities. But of course, this obvious examples doesn't work. Maybe we can find an answer in [Samsung mobile print app](https://play.google.com/store/apps/details?id=com.sec.print.mobileprint&hl=ru&gl=US) which perfectly works on android...

[scanimage-zeroconf.log](https://pastebin.com/raw/JwaZbkKL)
[scanimage-Samsung-SL-M2875FD:8018.log](https://pastebin.com/raw/K3MRArX5)
[scanimage-Samsung-SL-M2875FD:test.log](https://pastebin.com/raw/xGkTXU0r)

> It is hard to say, why `scanimage` prefers device, exposed by `sane-escl`, not by `sane-airscan`

Sorry for that,  i've forgot that i made changes in escl.conf, too. I've explicitly added "device http://192.168.1.111:80 "Samsung M267x 287x Series (192.168.1.111)" \"hack=localhost\"", which actually means that only xerox_mfp is kind of working out of the box (magically triggered, only once and only for 1 page)... 

Checking escl debug we see that it just return invalid request html page.
export SANE_DEBUG_DLL=255 && export SANE_DEBUG_XEROX_MFP=4 && export SANE_DEBUG_ESCL=255 && scanimage -d "escl:http://192.168.1.111:80"

```
[sanei_debug] Setting debug level of escl to 255.
[07:46:44.653294] [escl] escl sane_init
[07:46:44.655191] [dll] init: backend `escl' is version 1.0.0
[07:46:44.655227] [escl] escl sane_open
[07:46:44.655236] [escl] escl_parse_name
[07:46:44.655278] [escl] escl_curl_url: URL: http://192.168.1.111:80/eSCL/ScannerCapabilities
[07:46:44.655303] [escl] Before use hack
[07:46:44.655310] [escl] After use hack
[07:46:44.660937] [escl] XML Capabilities[
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
<HTML>
<HEAD>
<TITLE>Invalid Request</TITLE>
</HEAD>

<BODY BGCOLOR="#FFFFFF" TEXT="#000000">
<CENTER>
<FONT SIZE="+2" COLOR="#FFFFFF" ALIGN="Center">
</FONT>
<B>Invalid Request. Some Error</B>
</BODY>

</HTML>

]
file.xml:1: parser error : Space required after the Public Identifier
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:1: parser error : SystemLiteral " or ' expected
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:1: parser error : SYSTEM or PUBLIC, the URI is missing
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2//EN">
                                                ^
file.xml:12: parser error : Opening and ending tag mismatch: CENTER line 0 and BODY
</BODY>
       ^
file.xml:14: parser error : Opening and ending tag mismatch: BODY line 0 and HTML
</HTML>
       ^
file.xml:16: parser error : EndTag: '</' not found

^
scanimage: open of device escl:http://192.168.1.111:80 failed: Out of memory
[07:46:44.661415] [dll] sane_exit: exiting
[07:46:44.661438] [dll] sane_exit: calling backend `escl's exit function
[07:46:44.661447] [escl] escl sane_exit
[07:46:44.661851] [dll] sane_exit: finished

```

markosjal, 2021-09-01:
I have been testing with a Brother MFP8710 . This scanner pre-dates eSCL on Brother.  I have the choice of WSD or Brother drivers on this device. Honestly I found the WSD driver here to be spotty. It does not provide full resolutions on most scanners and does not work consistently on others . I strongly suggest you install the Brother Linux drivers and call it good.   In fact I have now seen that if Windows installs a generic WSD driver like it did on Brother and on an HP, WSD did not offer the full resolutions offered by proprietary ( or in the case of the HP driverless escl) drivers . I mean on the HP if I compare mopria Android resolutions and features available over eSCL it is superior to any WSD support I have seen for it. Same goes for the HP drivers,  which also are way better than WSD.

The Brother is no different it just seems to be a limited feature set on WSD.

markosjal, 2021-09-04:
Sorry My bad about Brother but Ditto for Samsung. I have one of those too . Mine is a M2885 so I guess I confused them at time of writing.  The Samsung ULD driver works well  https://support.hp.com/in-en/drivers/selfservice/samsung-xpress-sl-m2875-laser-multifunction-printer-series/16462699/model/16462703.  I know there is a debian ULD repo somewhere but if all else fails download the ULD (Universal Linux Driver) driver from Samsung . Once extracted you can choose to install only the scanner if you are Okay with the driverless printing option, but I think you will get more options using the Samsung Print driver.

This will allow you to connect over a LAN to the Samsung MFP. It also offers more options than the WSD driver . I did finally have some success also with SimulPiscator/AirSane to share the Samsung with Android using Mopria but came to find out the hard way that it does not support the feeder, which defeats the whole purpose in my case.

runningman84, 2021-11-26:
maybe you should try this config:
```
[devices]
"Samsung M2070" = http://10.0.1.111:8018/wsd/scan, wsd
```
in my arch setup this works just fine!

alexpevzner, 2022-05-06:
Hi!

sorry for disappearing for very long time.

Is this issue still relevant?

diabolusss, 2022-05-12:
Hi. I believe it's not, i don't use scanner very often, so for my needs scanimage via tcp is enough right now.
It's stable for few consequential scans even for ADF source, but i haven't tested it thoroughly.
Of course, it's sad that i couldn't find any UI app for user-friendly management, but like i've said, for a rare use case it's acceptable right now.

`device xerox_mfp:tcp 192.168.1.111' is a Samsung M267x 287x Series multi-function peripheral`

```
$ cat /usr/bin/scanimage_adf
#!/bin/bash
scanimage --batch='scan_out%d.png' --source ADF -d 'xerox_mfp:tcp 192.168.1.111 9400'
```

alexpevzner, 2023-07-07:
The issue looks obsoleted - closing it for now

closed 2023-07-07 by alexpevzner