alexpevzner/sane-airscan #175: Scanning with eSCL client to custom server fails for unknown reason.

markosjal, 2021-09-04:


markosjal, 2021-09-08:
If anyone is interested in following this , now tested with OSX Mojave Imaging app to find similar results. It seems this may have to do with Keep-alives so since my server code is running on XAMPP on Windows I posted it here  https://community.apachefriends.org/f/viewtopic.php?f=16&t=81270&p=274294#p274294 . Oddly enough the Mopria Scan App does not have this issue!

https://community.apachefriends.org/f/viewtopic.php?f=16&t=81270&p=274294#p274294

markosjal, 2021-09-13:
Okay, I have more info on this that seems relevant.......

I acquired an HP Color Laserjet M283fdw. After some testing I note the following

JPGs are transferred Immediately after each page is scanned on a sheetfeeder.  The SimpleScan (whatever it is now called) Client itself , does any and all processing if a PDF is the desired format. This is not the PDF defined in the Mopria Spec. The Mopria Spec would have it generated on the scanner.

So,
The eSCL format clearly stipulates that a REQUIRED format is PDF. In the case of a Multi-page PDF the PDF can not be generated until ALL pages have been scanned. This could mean that the client (SimpleScan with Airscan) needs to wait until ALL pages are scanned and the PDF is generated.SimpleScann with AirScan instead times out after about 2 minutes.

Have a look at the behavior of Mopria Scan on Android. It expects the PDF to be generated in the scanner and does not have a problem waiting up to about 300 seconds for that to happen under my current testing scenario. Simplescan with Airscan however seems to want to scan each page one by one , thus violating the conformity of eSCL by not allowing multi page PDFs to be generated on the host scanner.

Now in my case,  a Windows eSCL Scan server,  no JPG or PDF is generated until all pages are scanned. This works perfectly well on Mopria Android App with most any sheetfeeder scanner (except a few). 

A Multi-page tiff might be another reason that Airscan would need to "Hold on". The TIFF can not be generated until all pages are scanned.

This would be a non issue if PDFs were not part of the Mopria Spec but they are. An eSCL compatible client may need to "Hold on" till all pages are scanned and the file is generated.

Airscan times out after about 2 minutes, no matter how I try to manipulate server config or headers. I do not believe this is correct behavior for the eSCL protocol and comparing it to the most "official" eSCL client I have, Mopria Scan for Android, is incorrect behavior.

alexpevzner, 2023-07-06:
Hi @markosjal,

SANE requires its drivers to return unpacked images. So adding PDF support will require `sane-airscan` to be able to unpack this format.

Existing PDF rendering libraries are heavyweigh and hard to use. Also, PDF doesn't guarantee lossless image. Actually, PDF may contain either lossless image in some PDF-specific format or just to be a container for JPEG-compressed image.

So for now I don't want to add PDF support to `sane-airscan`.

When single-image format, like JPEG, is used for ADF scan, `sane-airscan` generates a series of  `NextDocument` requests, and each request returns a single image. Even if internally the server scans a whole series of pages at once, it must return one image per request at this case.

closed 2024-11-18 by alexpevzner