alexpevzner/sane-airscan #176: Brother MFC-J430W works only with limited dpi

Kirk1984, 2021-09-09:
`device `airscan:w1:Brother MFC-J430W' is a WSD Brother MFC-J430W ip=xxx.xxx.xxx.xxx`

But i only have 3 dpi settings (100,200,300). The Device is listed by Brother for (up to) 1200 × 2400 dpi for Flatbed and 1200 × 600 dpi

debiantriage, 2021-09-12:
Please see [here](https://github.com/alexpevzner/sane-airscan/issues/145). There isn't anything the sane-airscan author can do about this. I have a bash script that can deal with all the resolutions using WSD.

alexpevzner, 2021-10-04:
Unfortunately, this device exposes only 100/200/300 DPI via WSD. It sometimes happen, there are few devices that can do more using the proprietary protocol rather that using eSCL/WSD protocols. Here I cannot help, I only expose the functionality offered by device.

closed 2021-10-04 by alexpevzner