alexpevzner/sane-airscan #179: Kyocera ECOSYS M5526cdw - Device is busy

heinrichat, 2021-09-27:
Hello!

New printer, same-ish problem? :(
I had a similar issue with my old Ricoh SP C 261sfnw printer. Maybe its not the issue with airscan-sane?

`
$ export SANE_DEBUG_DLL=255 && scanimage --format=jpeg > test.jpeg
[sanei_debug] Setting debug level of dll to 255.
[dll] sane_init: SANE dll backend version 1.0.13 from sane-backends 1.0.27
[dll] sane_init/read_dlld: attempting to open directory `./dll.d'
[dll] sane_init/read_dlld: attempting to open directory `/etc/sane.d/dll.d'
[dll] sane_init/read_dlld: using config directory `/etc/sane.d/dll.d'
[dll] sane_init/read_dlld: considering /etc/sane.d/dll.d/airscan
[dll] sane_init/read_config: reading dll.d/airscan
[dll] add_backend: adding backend `airscan'
[dll] sane_init/read_dlld: done.
[dll] sane_init/read_config: reading dll.conf
[dll] add_backend: adding backend `net'
[dll] add_backend: adding backend `ricoh'
[dll] sane_get_devices
[dll] load: searching backend `ricoh' in `/usr/lib/x86_64-linux-gnu/sane:/usr/lib/sane'
[dll] load: trying to load `/usr/lib/x86_64-linux-gnu/sane/libsane-ricoh.so.1'
[dll] load: dlopen()ing `/usr/lib/x86_64-linux-gnu/sane/libsane-ricoh.so.1'
[dll] init: initializing backend `ricoh'
[dll] init: backend `ricoh' is version 1.0.0
[dll] load: searching backend `net' in `/usr/lib/x86_64-linux-gnu/sane:/usr/lib/sane'
[dll] load: trying to load `/usr/lib/x86_64-linux-gnu/sane/libsane-net.so.1'
[dll] load: dlopen()ing `/usr/lib/x86_64-linux-gnu/sane/libsane-net.so.1'
[dll] init: initializing backend `net'
[dll] init: backend `net' is version 1.0.27
[dll] load: searching backend `airscan' in `/usr/lib/x86_64-linux-gnu/sane:/usr/lib/sane'
[dll] load: trying to load `/usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1'
[dll] load: dlopen()ing `/usr/lib/x86_64-linux-gnu/sane/libsane-airscan.so.1'
[dll] init: initializing backend `airscan'
[dll] init: backend `airscan' is version 1.0.0
[dll] sane_get_devices: found 1 devices
[dll] sane_open: trying to open `airscan:e1:Kyocera ECOSYS M5526cdw'
[dll] sane_open: open successful
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=0)
[dll] sane_control_option(handle=0x555f227b7c10,option=0,action=0,value=0x7ffdb7309b8c,info=(nil))
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=0)
[dll] sane_control_option(handle=0x555f227b7c10,option=0,action=0,value=0x7ffdb7309a90,info=(nil))
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=1)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=2)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=3)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=4)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=5)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=6)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=7)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=8)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=9)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=10)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=11)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=12)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=13)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=14)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=15)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=16)
[dll] sane_control_option(handle=0x555f227b7c10,option=8,action=0,value=0x555f22793380,info=(nil))
[dll] sane_control_option(handle=0x555f227b7c10,option=6,action=0,value=0x7ffdb7309a94,info=(nil))
[dll] sane_control_option(handle=0x555f227b7c10,option=9,action=0,value=0x555f22793384,info=(nil))
[dll] sane_control_option(handle=0x555f227b7c10,option=7,action=0,value=0x7ffdb7309a94,info=(nil))
[dll] sane_control_option(handle=0x555f227b7c10,option=6,action=0,value=0x7ffdb7309bc0,info=(nil))
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=8)
[dll] sane_control_option(handle=0x555f227b7c10,option=8,action=1,value=0x7ffdb7309be0,info=0x7ffdb7309aa4)
[dll] sane_control_option(handle=0x555f227b7c10,option=7,action=0,value=0x7ffdb7309bc0,info=(nil))
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=9)
[dll] sane_control_option(handle=0x555f227b7c10,option=9,action=1,value=0x7ffdb7309be0,info=0x7ffdb7309aa4)
[dll] sane_get_option_descriptor(handle=0x555f227b7c10,option=2)
[dll] sane_control_option(handle=0x555f227b7c10,option=2,action=0,value=0x7ffdb7309aa0,info=(nil))
[... 1413 lines cut ...]
<scan:ScannerStatus xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm">
  <pwg:Version>2.62</pwg:Version>
  <pwg:State>Idle</pwg:State>
  <scan:AdfState>ScannerAdfProcessing</scan:AdfState>
</scan:ScannerStatus>

00:00:11.777: Kyocera ECOSYS M5526cdw: HTTP GET http://192.168.1.153:9095/eSCL/ScannerStatus: OK
00:00:11.777: Kyocera ECOSYS M5526cdw: PROTO_OP_CHECK: decoding
00:00:11.777: Kyocera ECOSYS M5526cdw: PROTO_OP_CHECK: device status: Success
00:00:11.777: Kyocera ECOSYS M5526cdw: PROTO_OP_CHECK: ADF status: Document feeder out of documents
00:00:11.777: Kyocera ECOSYS M5526cdw: PROTO_OP_CHECK: job status: Success
00:00:11.777: Kyocera ECOSYS M5526cdw: PROTO_OP_CHECK: decoded: status="Device busy" next=PROTO_OP_FINISH delay=0
00:00:11.777: Kyocera ECOSYS M5526cdw: JOB status=Device busy
00:00:11.777: Kyocera ECOSYS M5526cdw: DEVICE_STM_SCANNING->DEVICE_STM_DONE
00:00:11.777: Kyocera ECOSYS M5526cdw: device_start_wait: Device busy
00:00:11.777: Kyocera ECOSYS M5526cdw: DEVICE_STM_DONE->DEVICE_STM_IDLE
00:00:11.777: Kyocera ECOSYS M5526cdw: API: sane_start(): Device busy
00:00:11.777: Kyocera ECOSYS M5526cdw: API: sane_close(): called
00:00:11.777: Kyocera ECOSYS M5526cdw: DEVICE_STM_IDLE->DEVICE_STM_CLOSED
00:00:11.777: Kyocera ECOSYS M5526cdw: removed from device table
00:00:11.777: Kyocera ECOSYS M5526cdw: closed protocol "eSCL"
00:00:11.777: Kyocera ECOSYS M5526cdw: closed decoder: jpeg
00:00:11.777: Kyocera ECOSYS M5526cdw: device destroyed
00:00:11.777: Kyocera ECOSYS M5526cdw: API: sane_close(): done
`

alexpevzner, 2021-09-27:
Hi!

I have the very similar device (ECOSYS M2040dn) with very similar firmware, and it works especially good and reliable with sane-airscan.

> Also currently there are no toner cartridges installed in the unit, but I hope that in 2021 this isn't the root problem heh.

It actually can.

P.S. I've opened the toner cartridge door, and it was enough to reproduce you problem. Probably, If I'll remove toner cartridge and close the door, the problem will remain, but I didn't try. So looks that without toner, device just doesn't want to scan.

heinrichat, 2021-09-30:
Hey!

So I bought replacement toner for it. I was detected as genuine. Scanning works but the output can't be viewed with any image viewer? I tried JPEG, PNG and TIFF. I also tried scanning via flatbed and ADF.

Here are the contents of the logs when trying all three formats. What am I doing wrong?

Attached are the images scanned with the flatbed.
test.jpg https://file.io/9opFupqW3J0I
test.png https://file.io/5JUgMT4hToNB
test.tiff https://file.io/5U2AvjIsxfjQ

```
00:00:01.006: Kyocera ECOSYS M5526cdw: device created
00:00:01.006: Kyocera ECOSYS M5526cdw: DEVICE_STM_OPENED->DEVICE_STM_PROBING
00:00:01.006: Kyocera ECOSYS M5526cdw: using protocol "eSCL"
00:00:01.006: Kyocera ECOSYS M5526cdw: HTTP GET http://192.168.1.153:9095/eSCL/ScannerCapabilities
00:00:01.006: Kyocera ECOSYS M5526cdw: HTTP using timeout: 5000 ms
00:00:01.006: Kyocera ECOSYS M5526cdw: HTTP resolving 192.168.1.153 9095
00:00:01.006: Kyocera ECOSYS M5526cdw: HTTP trying 192.168.1.153:9095
00:00:01.006: Kyocera ECOSYS M5526cdw: HTTP fdpoll: {--} -> {-w}
00:00:01.009: Kyocera ECOSYS M5526cdw: HTTP 87 bytes sent
00:00:01.009: Kyocera ECOSYS M5526cdw: HTTP done request sending
00:00:01.009: Kyocera ECOSYS M5526cdw: HTTP fdpoll: {-w} -> {rw}
00:00:01.009: Kyocera ECOSYS M5526cdw: HTTP fdpoll: {rw} -> {r-}
00:00:01.024: Kyocera ECOSYS M5526cdw: HTTP 1623 bytes received
00:00:01.024: Kyocera ECOSYS M5526cdw: HTTP GET http://192.168.1.153:9095/eSCL/ScannerCapabilities: got response headers (200)
00:00:01.024: Kyocera ECOSYS M5526cdw: HTTP 5792 bytes received
00:00:01.035: Kyocera ECOSYS M5526cdw: HTTP 1420 bytes received
00:00:01.035: Kyocera ECOSYS M5526cdw: HTTP done response reception
00:00:01.035: Kyocera ECOSYS M5526cdw: HTTP GET http://192.168.1.153:9095/eSCL/ScannerCapabilities: 200 OK
==============================
GET http://192.168.1.153:9095/eSCL/ScannerCapabilities
Connection: close
Host: 192.168.1.153:9095

Status: 200 OK
Content-Length: 8660
Upgrade: TLS/1.0
Accept-Encoding: identity
Date: Fri, 01 Oct 2021 18:26:27 GMT
Server: KM-MFP-http/V0.0.1
Content-Type: text/xml

<scan:ScannerCapabilities xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm">
  <pwg:Version>2.62</pwg:Version>
  <pwg:MakeAndModel>Kyocera ECOSYS M5526cdw</pwg:MakeAndModel>
  <pwg:SerialNumber>VE96Y02346</pwg:SerialNumber>
  <pwg:ModelName>ECOSYS M5526cdw</pwg:ModelName>
  <scan:UUID>4509a320-0060-00d0-006a-0025074fc1a3</scan:UUID>
  <scan:AdminURI>https://printer.local/airprint</scan:AdminURI>
  <scan:IconURI>https://printer.local/printer-icon/machine_128.png</scan:IconURI>
  <scan:Platen>
    <scan:PlatenInputCaps>
      <scan:MinWidth>118</scan:MinWidth>
      <scan:MaxWidth>2551</scan:MaxWidth>
      <scan:MinHeight>118</scan:MinHeight>
      <scan:MaxHeight>3508</scan:MaxHeight>
      <scan:SettingProfiles>
        <scan:SettingProfile>
          <scan:ColorModes>
            <scan:ColorMode>BlackAndWhite1</scan:ColorMode>
            <scan:ColorMode>Grayscale8</scan:ColorMode>
            <scan:ColorMode>RGB24</scan:ColorMode>
          </scan:ColorModes>
          <scan:DocumentFormats>
            <pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat>
            <pwg:DocumentFormat>application/pdf</pwg:DocumentFormat>
          </scan:DocumentFormats>
          <scan:SupportedResolutions>
            <scan:DiscreteResolutions>
[... 564 lines cut ...]
00:00:11.166: Kyocera ECOSYS M5526cdw: HTTP done response reception
00:00:11.166: Kyocera ECOSYS M5526cdw: HTTP DELETE http://192.168.1.153:9095/eSCL/ScanJobs/urn:uuid:4509a320-0060-00d0-006a-0055f5b0cd31: 200 OK
==============================
DELETE http://192.168.1.153:9095/eSCL/ScanJobs/urn:uuid:4509a320-0060-00d0-006a-0055f5b0cd31
Connection: close
Host: 192.168.1.153:9095

Status: 200 OK
Content-Length: 0
Upgrade: TLS/1.0
Accept-Encoding: identity
Date: Fri, 01 Oct 2021 18:26:37 GMT
Server: KM-MFP-http/V0.0.1

00:00:11.167: Kyocera ECOSYS M5526cdw: HTTP DELETE http://192.168.1.153:9095/eSCL/ScanJobs/urn:uuid:4509a320-0060-00d0-006a-0055f5b0cd31: OK
00:00:11.167: Kyocera ECOSYS M5526cdw: PROTO_OP_CLEANUP: decoding
00:00:11.167: Kyocera ECOSYS M5526cdw: PROTO_OP_CLEANUP: decoded: status="Success" next=PROTO_OP_FINISH delay=0
00:00:11.167: Kyocera ECOSYS M5526cdw: DEVICE_STM_CLEANUP->DEVICE_STM_DONE
00:00:11.167: Kyocera ECOSYS M5526cdw: DEVICE_STM_DONE->DEVICE_STM_CLOSED
00:00:11.167: Kyocera ECOSYS M5526cdw: removed from device table
00:00:11.167: Kyocera ECOSYS M5526cdw: closed protocol "eSCL"
00:00:11.167: Kyocera ECOSYS M5526cdw: closed decoder: jpeg
00:00:11.167: Kyocera ECOSYS M5526cdw: device destroyed
00:00:11.167: Kyocera ECOSYS M5526cdw: API: sane_close(): done
```

closed 2021-10-01 by heinrichat