alexpevzner/sane-airscan #180: IPv4 required? Scanning breaks when disabled

chrsmrtnx, 2021-09-30:
Hello,

Testing my HP Color LaserJet MFP M182nw last night, I noticed that sane-airscan breaks when I disable IPv4 on the printer.

To explain how I got here - since I don't have avahi-daemon installed, I edited airscan.conf to add:

HP_ColorLaserJet_MFP_M182_via_eSCL = https://laserjet2:443/eSCL, eSCL

I also threw in:

[options]
discovery = disable
ws-discovery = off

...for good measure. Please note that "laserjet2" is defined in /etc/hosts as an IPv6 ULA (fd00::etc.). Scanning worked fine, but I noticed that during the scan, a lot of traffic to and from the printer was going over IPv4 (and some over IPv6 as well). So I disabled IPv4 on the printer, and then noticed that sane-airscan started failing after throwing "Error during device I/O" at a certain point during the scan operation.

Looking at the debug output, I noticed the following at the end of the ~/airscan/trace/*.log (with actual IP addresses edited by me):

```
POST https://[fd00::1234:1234:1234:1234]:443/eSCL/ScanJobs
Connection: close
Content-Type: text/xml
Host: [fd00::1234:1234:1234:1234]
Content-Length: 742

<scan:ScanSettings xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03">
  <pwg:Version>2.0</pwg:Version>
  <pwg:ScanRegions>
    <pwg:ScanRegion>
      <pwg:ContentRegionUnits>escl:ThreeHundredthsOfInches</pwg:ContentRegionUnits>
      <pwg:XOffset>0</pwg:XOffset>
      <pwg:YOffset>0</pwg:YOffset>
      <pwg:Width>2550</pwg:Width>
      <pwg:Height>3508</pwg:Height>
    </pwg:ScanRegion>
  </pwg:ScanRegions>
  <pwg:InputSource>Platen</pwg:InputSource>
  <scan:ColorMode>RGB24</scan:ColorMode>
  <pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat>
  <scan:DocumentFormatExt>image/jpeg</scan:DocumentFormatExt>
  <scan:XResolution>300</scan:XResolution>
  <scan:YResolution>300</scan:YResolution>
</scan:ScanSettings>

Status: 201 Created
Cache-Control: no-cache, no-store, must-revalidate
Location: https://[FD00:443/eSCL/ScanJobs/85kj8b1v-cbcs-f7th-1017-oimkl6sk
Server: gSOAP/2.7
Content-Length: 0
Connection: close

00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP POST https://[fd00::1234:1234:1234:1234]:443/eSCL/ScanJobs: Created
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: PROTO_OP_SCAN: decoding
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: PROTO_OP_SCAN: decoded: status="Error during device I/O" next=PROTO_OP_FINISH delay=0
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: ScanJobs request: invalid location received
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: JOB status=Error during device I/O
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: DEVICE_STM_SCANNING->DEVICE_STM_DONE
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: device_start_wait: Error during device I/O
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: DEVICE_STM_DONE->DEVICE_STM_IDLE
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_start(): Error during device I/O
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_close(): called
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: DEVICE_STM_IDLE->DEVICE_STM_CLOSED
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: removed from device table
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: closed protocol "eSCL"
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: closed decoder: jpeg
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: device destroyed
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_close(): done
```

I'm struck by the line "Location: https://[FD00:443/eSCL/ScanJobs/85kj8b1v-cbcs-f7th-1017-oimkl6sk", with the incomplete IPv6 address, which makes me wonder whether somewhere an IPv6 address isn't being parsed correctly.

Interestingly, here's the same section I get when I turn IPv4 back on, and the scan succeeds:

```
Status: 201 Created
Cache-Control: no-cache, no-store, must-revalidate
Location: https://192.168.123.123:443/eSCL/ScanJobs/85kj8b1v-cuz6-d94s-1018-v6klwhhg
Server: gSOAP/2.7
Content-Length: 0
Connection: close

00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP POST https://[fd00::1234:1234:1234:1234]:443/eSCL/ScanJobs: Created
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: PROTO_OP_SCAN: decoding
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: PROTO_OP_SCAN: decoded: status="Success" next=PROTO_OP_LOAD delay=0
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: PROTO_OP_LOAD: submitting: attempt=0
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP GET https://192.168.123.123:443/eSCL/ScanJobs/85kj8b1v-cuz6-d94s-1018-v6klwhhg/NextDocument
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP resolving 192.168.123.123 443
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP trying 192.168.123.123:443
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {--} -> {-w}
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: device_start_wait: Success
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_start(): Success
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_get_params(): called
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: API: sane_get_params(): done
00:00:02.219: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {-w} -> {r-}
00:00:02.221: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {r-} -> {r-}
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP done TLS handshake
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {r-} -> {rw}
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP 116 bytes sent
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP done request sending
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {rw} -> {rw}
00:00:02.475: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP fdpoll: {rw} -> {r-}
00:00:02.716: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP 145 bytes received
00:00:02.716: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP GET https://192.168.123.123:443/eSCL/ScanJobs/85kj8b1v-cuz6-d94s-1018-v6klwhhg/NextDocument: got response headers (503)
00:00:02.716: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP using timeout: none
00:00:02.716: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP done response reception
00:00:02.716: HP_ColorLaserJet_MFP_M182_via_eSCL: HTTP GET https://192.168.123.123:443/eSCL/ScanJobs/85kj8b1v-cuz6-d94s-1018-v6klwhhg/NextDocument: 503 Service Unavailable
```

It works, but of course it's falling back to IPv4. Curious that sane-airscan has even learned of the IPv4 address of the printer. Now, to be sure, I can't rule out of the possibility that the printer itself is the problem - that, despite having disabled IPv4, there's some particular service or interface that isn't running properly on IPv6. Nonetheless, I thought I'd post this here in case the issue is on the client side.

Assuming it's not just an HP bug, the expected result would be that (1) scanning works even with IPv4 disabled, and (2) even if IPv4 is enabled, sane-airscan should prefer IPv6, particularly if it's passed an IPv6 address at the outset.

Thanks very much.

alexpevzner, 2021-10-04:
Hi @chrsmrtnx,

It looks like HP firmware bug. The invalid location (`https://[FD00:443/eSCL/ScanJobs/85kj8b1v-cbcs-f7th-1017-oimkl6sk`) was taken from the device response. Because this URL is syntactically invalid, `sane-airscan` refuses it with the error message:
```
00:00:02.051: HP_ColorLaserJet_MFP_M182_via_eSCL: ScanJobs request: invalid location received
``` 

I've tested IPv6 mode with my Kyocera ECOSYS M2040dn; it works. If there is a choice, sane-airscan prefers IPv6 addresses over IPv4.

chrsmrtnx, 2021-10-04:
Thank you - I appreciate the clarification/explanation of what I was seeing. Not surprising that it's an HP issue, sadly.

I'll close this now.

closed 2021-10-04 by chrsmrtnx