alexpevzner/sane-airscan #198: Preview or Scan via WSD works only once

dkadioglu, 2021-12-05:
Preview or Scan via WSD on my OKI MC 362dn works only once. After that one successful command, sometimes restarting the printers NIC or sometimes only rebooting the printer leads to another successful command but, again only once until restart.
After a single successful command, for all further commands I get an error message telling me, that the Device is busy.

OS: Arch Linux, fully updated
MFD: OKI MC 362dn, Firmware A13.80_0_5

As I have no clue, if this could be the fault of the MFD, the OS or Airscan, I'd like to kindly ask you, what information/logs would be helpful.

dkadioglu, 2021-12-27:
I just did two scans in a row, as expected, the first worked, the second failed. Please find attached the traces separated into 'good' from the first successful scan and 'bad' from the second failed scan. I would be happy about any sort of help in investigating this further. Thanks in advance.
[airscan.tar.gz](https://github.com/alexpevzner/sane-airscan/files/7781303/airscan.tar.gz)

alexpevzner, 2023-07-07:
Hi!

sorry for very long delay in response. Is this issue still relevant?

dkadioglu, 2023-08-06:
Hi, thanks for asking. Yes, unfortunately, this is still relevant. Just tried it again and the issue is still the same, i.e. an operation (e.g. Preview scan via Skanlite) runs only once, afterwards any other command is not successful and I have to restart the network stack of the device. If you need some new logs or any further info, please ask.

yellowpookie, 2024-06-03:
Have the same issue with Oki MB471
Oki FW: A14.90_0_4 (latest)
Debian 12.5
Sane-airscan git latest

Trace logs attached
[first_attempt.log](https://github.com/user-attachments/files/15536537/first_attempt.log)
[second_attempt.log](https://github.com/user-attachments/files/15536535/second_attempt.log)

Ситуация такая же как у автора. Первое сканирование проходит, вне зависимости от источника, любая попытка сканирования после, с любого источника, отвечает Device busy

alexpevzner, 2024-06-03:
Hi @yellowpookie,

could we please speak English here, please, so other people will be able to read our discussion?

Can you build `sane-airscan` from sources, for debugging purposes? There is no need to `make install` it, it would be enough to locate where `libsane-airscan.so.1` is installed on your distro (usually it is `/usr/lib64/sane/libsane-airscan.so.1` or `/usr/lib/sane/libsane-airscan.so.1`) and replace it with symbolic link to the `libsane-airscan.so.1` in your build directory).

Once you've succeed doing that, please re-run your tests and attach the new logs. Then I will be able to try to do something with this issue.

alexpevzner, 2024-06-06:
Hi!

Please update, rebuild and retest. And regardless of test result, please show me updated logs.

yellowpookie, 2024-06-06:
@alexpevzner issue isn't fixed this time. Same behavior, as I can see.
Logs:
[first_attempt.log](https://github.com/user-attachments/files/15661274/first_attempt.log)
[first_attempt.tar.gz](https://github.com/user-attachments/files/15661273/first_attempt.tar.gz)
[second_attempt.log](https://github.com/user-attachments/files/15661272/second_attempt.log)
[second_attempt.tar.gz](https://github.com/user-attachments/files/15661269/second_attempt.tar.gz)

alexpevzner, 2024-06-06:
Hi @yellowpookie,

once again, please.

And the question: what do you need to recover the scanner? Reboot it? Wait for some time?

yellowpookie, 2024-06-06:
@alexpevzner reboot MFP itself, or reboot network stack from web interface by action called "Reset Network Card". It restarts all network services on MFP, I guess, dhcp-client and web-server for sure.

Looks the same
[fourth_attempt.log](https://github.com/user-attachments/files/15670728/fourth_attempt.log)
[third_attempt.log](https://github.com/user-attachments/files/15670736/third_attempt.log)

alexpevzner, 2024-06-06:
Can you also run tests with ADF, please?

yellowpookie, 2024-06-14:
@alexpevzner sorry for delay, MFP is remotely located. ADF and ADF Duplex can work continuously, as I can see. ADF Duplex have issue with page size, like I guess. Flatbed still get stuck after first attempt.
ADF:
[adf-first.log](https://github.com/user-attachments/files/15838115/adf-first.log)
[adf-first.tar.gz](https://github.com/user-attachments/files/15838114/adf-first.tar.gz)
[adf-second.log](https://github.com/user-attachments/files/15838112/adf-second.log)
[adf-second.tar.gz](https://github.com/user-attachments/files/15838113/adf-second.tar.gz)
[adf-third.tar.gz](https://github.com/user-attachments/files/15838110/adf-third.tar.gz)
[adf-third.log](https://github.com/user-attachments/files/15838111/adf-third.log)
ADF Duplex:
[adf-duplex-first.log](https://github.com/user-attachments/files/15838131/adf-duplex-first.log)
[adf-duplex-first.tar.gz](https://github.com/user-attachments/files/15838130/adf-duplex-first.tar.gz)
[adf-duplex-second.log](https://github.com/user-attachments/files/15838129/adf-duplex-second.log)
[adf-duplex-second.tar.gz](https://github.com/user-attachments/files/15838128/adf-duplex-second.tar.gz)
[adf-duplex-third.log](https://github.com/user-attachments/files/15838125/adf-duplex-third.log)
[adf-duplex-third.tar.gz](https://github.com/user-attachments/files/15838127/adf-duplex-third.tar.gz)

hegdi, 2024-07-04:
Same issue here with Oki MC332.
ADF works. Flatbed once basically renders WSD useless, till reboot/reset of the MFP. Not even airscan-discover can find it anymore.
I have the device at hand so I can provide you with test results, if helpful.
I can also provide you with pcap traces from Windows showing Flatbed scanning repeatedly if you want to have a look.
If I can do something to assist with this issue feel free to ask.

yellowpookie, 2024-07-08:
That patch work for Oki MB471, but breaks HP LaserJet Pro MFP M125nw over WSD, at least.

hegdi, 2024-07-08:
Fair point, I don't have any other device to test. After checking the specification the intended behavior is pretty clear.
[Windows Driver specification](https://learn.microsoft.com/en-us/windows-hardware/drivers/image/retrieveimageresponse)

> If [Format](https://learn.microsoft.com/en-us/windows-hardware/drivers/image/format) specifies a single image format and the value of [ImagesToTransfer](https://learn.microsoft.com/en-us/windows-hardware/drivers/image/>imagestotransfer) is 0 or greater than 1, the client will send repeated [RetrieveImageRequest](https://learn.microsoft.com/en-us/windows-hardware/drivers/image/retrieveimagerequest) operation elements until the Scan Service replies with a ClientErrorNoImagesAvailable fault or until the ImagesToTransfer value is met.

Currently ``ImageToTransfer`` is either zero or some large number so for Flatbed it should be exactly one.

hegdi, 2024-07-08:
This should be fixed in #343 seems I'm to stupid to link it directly

alexpevzner, 2024-07-10:
@yellowpookie,

what happens to HP LaserJet Pro MFP M125nw with @hegdi's patch? Can you give more detailed explanation with logs?

@hegdi,

your patch changes `PROTO_OP_CLEANUP` to `PROTO_OP_FINISH` which effectively disables cleanup step and separately you change cleanup from cancel to `RetrieveImage`. What is correct of these two changes?

yellowpookie, 2024-07-10:
@alexpevzner I'll test today both patch above and #343 and post detailed information here.

hegdi, 2024-07-10:
@alexpevzner,

Let me try to clarify, the patch in my post wasn't fixing the root cause it only allowed me to proof my suspicion. According to the Windows Driver Specification if you send a ``RetrieveImageRequest`` with a ``ImageToTransfer`` of zero, indicating continuous scanning, you need to do so until the device answers with ``ClientErrorNoImagesAvailable``. 

This is how it is handled in the ADF case, which worked even before your patch.
That basically also means if the number of scans is known and fulfilled like ``ImageToTransfer`` equals 5 and this is the 5th image retrieved you don't need to send another ``RetrieveImageRequest`` cause you know the sequence is over.

Now let's come to the way it is implemented currently ( as in your master ). ADF case is just fine as it already follows this logic. But the Flatbed case was sending a ``ImageToTransfer`` of zero too, which would indicate more to come for the MFP and requiring another ``RetrieveImageRequest``. That is what I did clumsily in my patch, to see if that makes the MFP happy.

That is why I called it a Proof of Concept in the patch post. Now to my pull request #343.
As my explanation hopefully made it clear the right thing to do is send a ``ImageToTransfer`` of 1 if using the Flatbed. Which tells the MFP the sequence is over and no additional ``RetrieveImageRequest`` is needed.

So as your change introduced a forced ``CancelJobRequest`` after every scan but didn't fix the problem. I wanted to get rid of the forced cancel but keep, your also introduced, sending cancel in different error cases.

So my pull request should:
- disable the forced cancel after every scan
- set the ``ImageToTransfer`` number properly

As a side note, the forced cancel returns with invalid job id, if the job is over (``ImageToTransfer`` count reached).

Hopefully my explanation helps.

alexpevzner, 2024-07-10:
@hegdi,

my forced cancel came from the eSCL experience, where some devices really want it. My `Kyocera ECOSYS M2040dn` one of such devices, it feels itself busy after the scan for about a minute, which is very annoying. Cancel helps.

But WSD firmwares behave differently, and cancel seems to be useless here.

You #343 PR looks correct for me, I'm going to accept it. But I still want some clarification from @yellowpookie regarding HP LaserJet Pro MFP M125nw.

BTW, your commentary regarding affected devices contradicts with your change to README.md. You've mentioned OKI MC332dn, MC362dn and MB471 as affected, but added MB471 and MC332 **as supporting eSCL**. Please, update there.

hegdi, 2024-07-10:
@alexpevzner,

Oops... wrong column. Updated!

@yellowpookie,

#343 is enough for testing, no need to bother yourself with the earlier patch. Thank you for testing!

hegdi, 2024-07-10:
@yellowpookie,

> ADF Duplex have issue with page size, like I guess.

Yes the page size is wrong, scaling and/or cropping issue. Either cut off or squished.
But that is a topic for another issue.

yellowpookie, 2024-07-10:
@alexpevzner #343 works with Oki MB471 and HP LaserJet Pro MFP M125nw both.

alexpevzner, 2024-07-10:
OK, lets merge...

yellowpookie, 2024-07-11:
@hegdi @alexpevzner Both MFP tested today frequently, without any issues. Thank you both for great work!

dkadioglu, 2024-07-21:
@yellowpookie @hegdi @espindola @alexpevzner thank you very much for testing and/or contributing fixes for this issue! With the recent changes, this works for me with my MC362 now, too!

commit 1e8b0ee7 (2024-06-06): WSD: Attempt to fix #198

On some devices, first WSD scan attempt succeeds, but subsequent
attempts fail with the wscn:ServerErrorNotAcceptingJobs error code.

So far, known devices affected by this issue are:

  - OKI MC 362dn
  - OKI MB471

Note, both are from OKI over WSD.

The very similar issue was found with Kyocera ECOSYS M2040dn when
working via eSCL. And solution for Kyocera was to explicitly cancel
scan job after successful completion (though protocol doesn't require
that).

This change implements the similar approach on WSD. Hope it will
help (will see after testing).
closed 2024-11-18 by alexpevzner