alexpevzner/sane-airscan #201: OBS: produce binary packages for openSUSE Leap 15.3

vmiklos, 2022-01-08:
I see the OBS binaries are created for openSUSE Leap 15.2 already. I assume that building for 15.3 is a matter of enabling that in the OBS config: could you please do that?

Less technical users don't guess that 15.2 packages kind of work on 15.3 (it does for me), this would help them. :-)

Other than that, thanks a lot for this project, my HP Laser MFP 135w works fine with it!

alexpevzner, 2022-01-10:
Hi!

unfortunately, there is no 15.3 target on OBS. I'll probably should update the README to notice that 15.3 users should install 15.2 packages.

vmiklos, 2022-01-10:
Hmm, I have a toy `home:vmiklos` repo in OBS, I have this fragment to build 15.3 packages:

```
  <repository name="openSUSE_Leap_15.3">
    <path project="openSUSE:Leap:15.3" repository="standard"/>
    <arch>x86_64</arch>
  </repository>
```

Could you try adding that to your https://build.opensuse.org/projects/.../meta ?

alexpevzner, 2022-01-10:
Well. Looks that you are more experienced in OBS, that me. So, please help.

First of all, I don't have `meta` file in my project. 

I use this page to edit list of distros I build for: https://build.opensuse.org/repositories/home:pzz

I've added 15.3, using the `Add from a Distribution` link, but result now looks different from other 15.x targets. The directory name at https://download.opensuse.org/repositories/home:/pzz/ is `15.3`, not `openSUSE_Leap_15.3`, as I would expect, and it is not listed at this page: https://software.opensuse.org//download.html?project=home%3Apzz&package=sane-airscan

What am I missing?

alexpevzner, 2022-01-10:
I've renamed repo to "openSUSE:Leap:15.3", to be consistent with other repos. So now I'm closing the issue. Thank you for your help!

closed 2022-01-10 by alexpevzner