alexpevzner/sane-airscan #203: Ricoh SP C261SFNw problem

Termit19, 2022-01-12:
Tested on SP C261SFNw.
airscan-discover
[devices]
RICOH SP C261SFNw (3c6e19) = http://192.168.1.68:631/eSCL/, eSCL
  RICOH SP C261SFNw (3c6e19) = https://192.168.1.68:443/eSCL/, eSCL
Scan starts, MFU warms up, says: Scaning ESCL, but performs no scan. No image shows on PC and scan app freezes.

alexpevzner, 2022-05-04:
Hi @Termit19,

sorry for very long delay in response.

If this issue is still relevant, to deal with this problem I need a protocol trace. Please, uncomment the following lines in the `/etc/sane.d/airscan.conf` file:
```
[debug]
trace = ~/airscan/trace
```
The `~/airscan/trace` directory will be created automatically and logs will be saved there. Please, attach logs to this issue. `*,tar` files are not needed for now, I only need the `*.log` files

ktheticdev, 2025-07-14:
Same issue here. Should I attach logs?

ktheticdev, 2025-07-15:
[scanimage-RICOH.log](https://github.com/user-attachments/files/21228993/scanimage-RICOH.log)
Same with HTTPS, it seems:

alexpevzner, 2025-07-15:
Hi!

This machine seems to be problematic. But I can try again...

If I will ask you to try to build aitrscan by yourself, will it be too difficult for you?

ktheticdev, 2025-07-15:
Hi!

Totally fine by me.