alexpevzner/sane-airscan #212: Debian: Key expired at OBS

sysms, 2022-02-07:
Dear maintainer,

I just want to inform you, that your key (at least for the Debian-repositories) at OBS has expired:

Err:3 http://download.opensuse.org/repositories/home:/pzz/Debian_10  InRelease                                                                                                               
  The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>

W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://download.opensuse.org/repositories/home:/pzz/Debian_10  InRelease: The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>
W: Failed to fetch http://download.opensuse.org/repositories/home:/pzz/Debian_10/InRelease  The following signatures were invalid: EXPKEYSIG 23E07629B9E3C9E4 home:pzz OBS Project <home:pzz@build.opensuse.org>

Please update the key file.

Thanks!

callegar, 2022-02-08:
The same happens on ubuntu.

Kyshman, 2022-02-19:
Also affected on Ubuntu 18.04. I made a comment ([#1584432](https://build.opensuse.org/package/show/home%3Apzz/sane-airscan#comment-1584432)) at OBS:PZZ/sane-airscan/overview and thought to also share here

alexpevzner, 2022-02-19:
Hi,

I've tried to renew the key and rebuild the binaries. Please confirm, either it works or not.

sysms, 2022-02-19:
@alexpevzner 

At least for the Debian-repository, the key is currently still unchanged.

[image]

Kyshman, 2022-02-19:
Just checked on the Ubuntu 18.04 repository. The key is still expired

`gpg Release.key`

Output:
_gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048/0x23E07629B9E3C9E4 2019-11-30 [SC] [expired: 2022-02-07]
      Key fingerprint = CDE7 95FF 222A BFA5 503F  6E49 23E0 7629 B9E3 C9E4
uid                             home:pzz OBS Project <home:pzz@build.opensuse.org>_

alexpevzner, 2022-02-19:
Please, retry. Now, I hope, it should be better...

Kyshman, 2022-02-19:
I can confirm on Ubuntu 18.04 that key is now valid

`sudo apt-key list | grep -A 4 pzz`

Output:

_"/etc/apt/trusted.gpg.d/home_pzz.gpg
pub   rsa2048 2019-11-30 [SC] **[expires: 2024-04-23]**
      CDE7 95FF 222A BFA5 503F  6E49 23E0 7629 B9E3 C9E4
uid           [ unknown] home:pzz OBS Project <home:pzz@build.opensuse.org>"_

sysms, 2022-02-20:
Hello @alexpevzner ,

I can also confirm, that the key in the Debian-repositories is now updated, and installing from this repository is now working again!

Thanks!

closed 2022-02-20 by sysms

alexpevzner, 2022-02-20:
Big thanks to everybody!