alexpevzner/sane-airscan #222: How to manually configure airscan for a Brother scanner?

stevesr0, 2022-02-26:
I have airscan installed on a computer, but I haven't found out how to make it "work".  It doesn't show up in htop and there is no obvious executable to run from the command line.  

I need to configure it manually to work with my Brother MFC-7345N device, since airscan-discovery didn't detect it.

The example in the airscan.conf file, "Kyocera MFP Scanner" = http://192.168.1.102:9095/eSCL, - includes the ip address followed by 4 digit (port?) number and the type of scanning protocol.  

While I know the static ip address for my scanner, I am not sure how to find out what the proper port(?) number is for my Brother scanner.  My /etc/services file states that for the sane daemon, port 6566/tcp is used.

Appreciate any assistance.

Thanks.

Steve Rinsler

tangyanli, 2022-02-28:
I assume the escl protocl port is same as the http port (usually, is 80 or 8080).
You may check the http port on the your device side.

And it's a simple way to verify the port: open the address "http://ip-address:80/eSCL/ScannerCapabilities" on Web Browser.
If the port is correct, the Web Browser will display the scanner's capabilities.

debiantriage, 2022-02-28:
The Brother MFC-7345N  appears to date from about 2009 so is too old to support the escl protocol. WSD does also not appear to be supported because `airscan-discover` does not show it. The vendor drivers will have to be used.

alexpevzner, 2022-02-28:
Hi,

I've looked to the device manual, and seems that it doesn't support neither ESCL, nor WSD, and this is why `sane-airscan` can't detect it automatically. For this reason, manual configuration also will not help.

Just in case, you can run `airscan-discover -t` and post file `airscan-discover-zeroconf.log`, that it will generate, here.

stevesr0, 2022-02-28:
Hi,
I have attached the file.  In the interim, I fixed what was not allowing this computer to scan (ethernet to ethernet) with the Brother (brscan3) driver.  Since you indicated that this scanner is not (necessarily) able to work with the two driverless (?) protocols that sane-airscan uses, I assume that it is unlikely that sane-airscan can work?

Thanks for responses.

Steve Rinsler

[airscan-discover-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/8156576/airscan-discover-zeroconf.log)

alexpevzner, 2022-02-28:
Thanks!

Few more questions.
1. What is IP address of your device?
2. I see, `avahi` daemon is not running. Could you please fix it and rerun `airscan-discover -t`?

stevesr0, 2022-03-01:
I had to install the avahi-daemon on this computer.  See attached log file.

The IP address is 192.168.2.105

alexpevzner, 2022-03-01:
Sorry, but there is no attached log files.

stevesr0, 2022-03-01:
Hi Alexander,

I see the attachment on my yahoo account when I open the sent message.  I am attaching it to this message from the same account.  If you don't see anything, I will send another way.

Steve Rinsler

stevesr0, 2022-03-01:
I don't see the attachment, sent from my email, so I am attaching it from "here".

Steve Rinsler

[airscan-discover-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/8165028/airscan-discover-zeroconf.log)

mike-lloyd03, 2022-03-02:
Having the same problem.

alexpevzner, 2022-05-04:
Hi Steve,

This device doesn't seem to support WSD scanning, only WSD printing. Sorry for very long delay with my answer.

stevesr0, 2022-05-05:
Thanks.
I did manage to get scanning working (not via air-scan though).
Steve Rinsler

alexpevzner, 2022-05-05:
I'm closing the issue because I cannot help you in this case.

closed 2022-05-05 by alexpevzner