alexpevzner/sane-airscan #225: is there any known issue with duplex scanners?

markosjal, 2022-03-23:
Testing with duplex scanners and Twain2AirScan scanner sharing...

two types; Single Pass and Double Pass

So far only getting first page on double pass duplex scanner.

I can see that files are downloaded but never get into the Linux app

I tested same double pass scanner with Twain2AirScan configuration on Mopria Android and no issues there. 

I have no Windows 11 to test with and MacOS does not support duplex scanning. I therefore have only Mopria Android to test with.

I was curious if any issues were known about duplex scanning.

markosjal, 2022-03-24:
trying to test further getting mixed results

alexpevzner, 2022-05-04:
Hi @markosjal.

sorry for very long delay.

I have a duplex scanner and I use it regularly, without any issue. My scanner is single-pass, but in duplex mode it returns two images, and it looks, from the driver perspective, as two subsequent requests resulting two different images.

Did I answer your question?

espindola, 2022-05-11:
I just tried duplex scanning with a MFC-L2750DW and I only got the odd numbered pages. It works with brscan4. Let me know if I can help debug.

alexpevzner, 2024-11-18:
Yes, sometimes (very rarely) duplex doesn't work. One of these problems I've closed today (#346).

Also, there are few scanners that rotate every second image in the duplex mode by 180 degrees (#354).

closed 2024-11-18 by alexpevzner