alexpevzner/sane-airscan #226: problems with scan request xml

markosjal, 2022-03-24:
With a full duplex scanner , a request is made for a single sided document from Document scanner, and document scanner passes "scan:Duplex  True" . This is Linux Mint 20.2 Uba with SANE-Airscan from mint repo (?) shows version 0.99.26-1+86.1
`<?xml version="1.0" encoding="UTF-8"?><scan:ScanSettings xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm" xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03"><pwg:Version>2.0</pwg:Version><pwg:ScanRegions><pwg:ScanRegion><pwg:ContentRegionUnits>escl:ThreeHundredthsOfInches</pwg:ContentRegionUnits><pwg:XOffset>0</pwg:XOffset><pwg:YOffset>0</pwg:YOffset><pwg:Width>2550</pwg:Width><pwg:Height>3300</pwg:Height></pwg:ScanRegion></pwg:ScanRegions><pwg:InputSource>Feeder</pwg:InputSource><scan:ColorMode>RGB24</scan:ColorMode><pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat><scan:DocumentFormatExt>image/jpeg</scan:DocumentFormatExt><scan:XResolution>300</scan:XResolution><scan:YResolution>300</scan:YResolution><scan:Duplex>true</scan:Duplex></scan:ScanSettings>`

There is an additional issue which I will Show a scan from MacOS 10.15 for comparison...
`<?xml version="1.0" encoding="UTF-8"?><scan:ScanSettings xmlns:scan="http://schemas.hp.com/imaging/escl/2011/05/03" xmlns:pwg="http://www.pwg.org/schemas/2010/12/sm"><pwg:Version>2.0</pwg:Version><scan:Intent>TextAndGraphic</scan:Intent><pwg:ScanRegions pwg:MustHonor="true"><pwg:ScanRegion><pwg:ContentRegionUnits>escl:ThreeHundredthsOfInches</pwg:ContentRegionUnits><pwg:Width>2550</pwg:Width><pwg:Height>3300</pwg:Height><pwg:XOffset>0</pwg:XOffset><pwg:YOffset>0</pwg:YOffset></pwg:ScanRegion></pwg:ScanRegions><pwg:DocumentFormat>image/jpeg</pwg:DocumentFormat><pwg:InputSource>Feeder</pwg:InputSource><scan:ColorMode>RGB24</scan:ColorMode><scan:XResolution>150</scan:XResolution><scan:YResolution>150</scan:YResolution><scan:Duplex>false</scan:Duplex></scan:ScanSettings>`

Both are claiming "pwg:version 2.0" while SANE-AirScan sends "scan:DocumentFormatExt" while MacOS sends "pwg:DocumentFormat". I believe that for "pwg:version 2.0" , "pwg:DocumentFormat" is correct

alexpevzner, 2022-03-24:
Hi @markosjal,

> I believe that for "pwg:version 2.0" , "pwg:DocumentFormat" is correct

Formally yes, but I have experience with some buggy firmware, which requires `scan:DocumentFormatExt` and doesn't work otherwise.

markosjal, 2022-03-25:
FYI still working o the Duplex/simplex problems they may all be my problems, not sure yet

alexpevzner, 2022-03-25:
Please note, `sane-airscan` works perfectly well with my Kyocera ECOSYS M2040dn in all modes (platen, single-side and double-side duplex). And there are many other duplex devices, reported to work. So most likely, the problem is at your side...

markosjal, 2022-03-29:
Thanks for that , I think I have got through most of it now.

alexpevzner, 2022-05-04:
Seems to be resolved...

closed 2022-05-04 by alexpevzner