alexpevzner/sane-airscan #236: Builds for OpenSUSE Leap 5.14

samuelgrf, 2022-05-19:
Hi, can you please enable 5.14 builds on [OBS](https://build.opensuse.org/package/show/home:pzz/sane-airscan).

Also thank you for this excellent tool, I've already scanned hundreds of pages with it :).

alexpevzner, 2022-05-22:
Done, and for the [ipp-usb](https://github.com/OpenPrinting/ipp-usb) too.

I wonder how to get these packages officially included into the OpenSUSE repos. They are already in the Debian/Ubintu/Fedora/Arch/... and even some *BSD.

alexpevzner, 2023-07-07:
Sorry, this is very unlikely that I'll ever find my personal resources to drive inclusion these packages (`sane-airscan` and `ipp-usb`) into official openSUSE repositories. I'm even not the openSUSE user, though pay a lot of respect to its community for its build services, which I was using for a while for these two projects

closed 2023-07-07 by alexpevzner