alexpevzner/sane-airscan #243: samsung scx-3400 series doesn't work

luc-git, 2022-07-04:
Hello I have a problem my samsung scx-3400 doesn't work 

Here is scanimage -L result:

scanimage -L
device `xerox_mfp:libusb:001:004' is a Samsung SCX-3400 Series multi-function peripheral
device `airscan:w0:samsung scanner' is a WSD samsung scanner ip=192.168.1.18

and airscan-discover result:

sudo airscan-discover -d
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 2a01:cb15:8279:b900:224c:7c52:e2ae:1677
WSDD:  + 192.168.1.18
WSDD:  + fe80::d021:66a2:e2fc:16de%2
WSDD: 2a01:cb15:8279:b900:224c:7c52:e2ae:1677: started discovery, UDP port=45605
WSDD: 192.168.1.18: started discovery, UDP port=47046
WSDD: setsockopt(AF_INET6,IPV6_ADD_MEMBERSHIP): Address already in use
WSDD: fe80::d021:66a2:e2fc:16de: started discovery, UDP port=34843
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipp._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipp._tcp
MDNS: browse/ipv6 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipps._tcp
MDNS: browse/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipps._tcp
MDNS: browse/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_BROWSER_NEW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _ipps._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscan._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _uscans._tcp
MDNS: browse/ipv6: AVAHI_BROWSER_CACHE_EXHAUSTED _scanner._tcp
MDNS: resolve/ipv6 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipp._tcp
WSDD: directed probe: trying if=eth0, addr=2a01:cb15:8279:b900:224c:7c52:e2ae:1677
WSDD: HTTP POST http://[2a01:cb15:8279:b900:224c:7c52:e2ae:1677]/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 2a01:cb15:8279:b900:224c:7c52:e2ae:1677 80
WSDD: HTTP trying [2a01:cb15:8279:b900:224c:7c52:e2ae:1677]:80
MDNS: resolve/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipp._tcp
WSDD: directed probe: trying if=eth0, addr=192.168.1.18
WSDD: HTTP POST http://192.168.1.18/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.1.18 80
WSDD: HTTP trying 192.168.1.18:80
MDNS: resolve/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipp._tcp
WSDD: directed probe: trying if=lo, addr=127.0.0.1
WSDD: directed probe: skipping loopback address
MDNS: resolve/ipv6 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipps._tcp
WSDD: directed probe: trying if=eth0, addr=2a01:cb15:8279:b900:224c:7c52:e2ae:1677
WSDD: directed probe: already in progress, skipping
MDNS: resolve/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipps._tcp
WSDD: directed probe: trying if=eth0, addr=192.168.1.18
WSDD: directed probe: already in progress, skipping
MDNS: resolve/ipv4 "Samsung SCX-3400 Series @ raspberrypi": AVAHI_RESOLVER_FOUND _ipps._tcp
WSDD: directed probe: trying if=lo, addr=127.0.0.1
WSDD: directed probe: skipping loopback address
WSDD: HTTP 192.168.1.18:80: send(): Connection refused
WSDD: HTTP POST http://192.168.1.18/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: Connection refused
WSDD: directed probe: HTTP Connection refused
WSDD: HTTP [2a01:cb15:8279:b900:224c:7c52:e2ae:1677]:80: send(): Connection refused
WSDD: HTTP POST http://[2a01:cb15:8279:b900:224c:7c52:e2ae:1677]/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery: Connection refused
WSDD: directed probe: HTTP Connection refused
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscan._tcp
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _uscans._tcp
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse/ipv6: AVAHI_BROWSER_ALL_FOR_NOW _scanner._tcp
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: WSDD not finished...
WSDD: netif event
WSDD: netif addresses update:
WSDD: fe80::d021:66a2:e2fc:16de: done discovery
WSDD: 192.168.1.18: done discovery
WSDD: 2a01:cb15:8279:b900:224c:7c52:e2ae:1677: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: zeroconf_device_list_get: resulting list:
[devices]

luc-git, 2022-07-04:
Maybe not because my samsung scanner only support usb. It doesn't support wifi or ethernet. Does this project support usb only scanner?

alexpevzner, 2022-07-04:
It depends.

If `lsusb -v` shows interfaces with class 7, subclass 1, protocol 4, it means that your printer supports IPP over USB, and, most likely, eSCL is also supported this way. At this case, with installing additional package, the `ipp-usb`, your printer will be made compatible with `sane-airscan`.

Otherwise, you will have to figure out the appropriate USB driver, which will work with your printer.

luc-git, 2022-07-05:
[airscan-discover-zeroconf.log](https://github.com/alexpevzner/sane-airscan/files/9046453/airscan-discover-zeroconf.log)

luc-git, 2022-07-05:
here is lsusb output:

lsusb -v

Bus 001 Device 005: ID 04e8:344f Samsung Electronics Co., Ltd SCX-3400 Series
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x04e8 Samsung Electronics Co., Ltd
  idProduct          0x344f SCX-3400 Series
  bcdDevice            1.00
  iManufacturer           1 Samsung Electronics Co., Ltd.
  iProduct                2 SCX-3400 Series
  iSerial                 3 Z6RRB8KCAB006TD
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0037
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              10
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         7 Printer
      bInterfaceSubClass      1 Printer
      bInterfaceProtocol      2 Bidirectional
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              10
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval              10
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

alexpevzner, 2022-07-05:
I don't see any 7/1/4 interface here, so probably sane-airscan/ipp-usb will not work with your printer. Sorry for that.

I close the issue, as it is not relevant anymore.

luc-git, 2022-07-05:
Ah ok Thank you

luc-git, 2022-07-05:
Maybe sane-airscan doesn't support all scx-3400. But in the Readme it is written that this printer supports wsd

alexpevzner, 2022-07-05:
Unlike eSCL, WSD doesn't work via USB. It only works via network.

I remember that your device is USB-only, but there are network-capable devices within this series. For some of these devices WSD reported to work.