alexpevzner/sane-airscan #245: Brother DCP-L2550DW - scans are always JPEG-compressed

kogans, 2022-07-14:
Is it possible to disable the JPEG compression? The scans come out horribly blocky, while the proprietary driver  (brother4) doesn't have that problem.

kpfleming, 2023-04-03:
I just set this up (using version 0.99.27-1+b1 from the Debian Bookworm repository) with a DCP-L2550DW. Used the GNOME Document Scanner tool, set for Image scan at 300 dpi, and scanned a page of text. The resulting PDF was perfect, no compression issues at all, just like brscan4 produced.

alexpevzner, 2023-07-06:
Unfortunately, driverless scanning sometimes (depending on device) doesn't offer as much control and features like proprietary drivers offer, so it is not possible to disable JPEG compression...

closed 2023-07-06 by alexpevzner