alexpevzner/sane-airscan #246: unable to scan from ADF on HP Laserjet MFP M234sdw

vajonam, 2022-07-23:
I am not able to scan from the ADF with either hp-scan or simplescan. It errors out communication errors.  I have attached the logs 

[python3.10-HP-LaserJet-MFP.log](https://github.com/alexpevzner/sane-airscan/files/9172839/python3.10-HP-LaserJet-MFP.log)
[python3.10-HP-LaserJet-MFP-M234sdw.log](https://github.com/alexpevzner/sane-airscan/files/9172840/python3.10-HP-LaserJet-MFP-M234sdw.log)
[simple-scan-HP-LaserJet-MFP-M234sdw.log](https://github.com/alexpevzner/sane-airscan/files/9172841/simple-scan-HP-LaserJet-MFP-M234sdw.log)

Any help is appreciated. 

I am able to scan using the flatbed okay.

alexpevzner, 2022-07-23:
OK, will look to it.

I guess from your profile, you are able to build `sane-airscan` from source, correct?

vajonam, 2022-07-23:
Yes I can build it if needed, at the moment I am using the Ubuntu package. 

Just another data point, this printer was in a different VLAN, and eSCL provided that error. When I moved the printer into the same VLAN as the guest network the WSD protocol showed up. I was able to scan with this protocol. I would like keep the VLAN separation but I have a work around for now. 

I use avahi on the router to reflect mDNS packets arcoss VLANs and have the firewall rules to communicate between VLANs to printer host.

alexpevzner, 2022-07-23:
We can try debug this issue.

alexpevzner, 2022-07-23:
And you can temporary connect some computer to the network segment where the printer connected to, use `airscan-discover` utility to obtain WSD URL, and with this URL you will probably be able to use WSD across your router.

alexpevzner, 2023-07-06:
Looks like issue is not relevant anymore...

closed 2023-07-06 by alexpevzner