alexpevzner/sane-airscan #251: HP Laserjet M2727dnf MFP

burnbabyburn, 2022-08-10:
Hi, hope this is the right place for this prob.

My MFP needs the proprietary HP-binary-plugin (e.g. hplip-3.22.4-plugin.run)  to work correctly. I am using a network setup. (`Printer: 192.168.1.233 on localnet and 10.0.0.29 on different network,  sane-airscan: 192.168.1.234, IPP_Service_Host: 192.168.1.10`)
the printers http service pages shows an WS-Discovery Service so we should have a WSD up and running. 
But it seems to not register in MDNS/WSSD/ZEROCONF and so  sane-airscanis not able to detect it. I tested my normal setup with an avahi-reflector and also moved the printer to the same network segment as  sane-airscan. It made no difference.

while tcpdumping a `hpaio://` scan (`scanimage --test --verbose --progress`) i was able to see the WSD XML the printer reported:
```
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:wscn="http://tempuri.org/wscn.xsd" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
  <SOAP-ENV:Body>
    <wscn:ScanElements>
      <ScannerConfiguration>
        <DeviceSettings>
          <FormatSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ScanDocumentFormat" SOAP-ENC:arraySize="2">
            <item>jfif</item>
            <item>hpraw</item>
          </FormatSupported>
          <QualityFactorSupported>false</QualityFactorSupported>
          <ContentSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ContentType" SOAP-ENC:arraySize="1">
            <item>Auto</item>
          </ContentSupported>
          <DocumentSizeAutoDetectSupported>false</DocumentSizeAutoDetectSupported>
          <FeederCapacity>50</FeederCapacity>
          <DuplexRotation>180</DuplexRotation>
          <Rotation>0</Rotation>
        </DeviceSettings>
        <Platen>
          <ColorSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ColorEntryType" SOAP-ENC:arraySize="4">
            <item>BlackandWhite1</item>
            <item>GrayScale8</item>
            <item>RGB24</item>
            <item>RGB48</item>
          </ColorSupported>
          <PlatenMinimumSize>
            <Width>1920</Width>
            <Height>1920</Height>
          </PlatenMinimumSize>
          <PlatenMaximumSize>
            <Width>8500</Width>
            <Height>11690</Height>
          </PlatenMaximumSize>
          <PlatenOpticalResolution>
            <Width>1200</Width>
            <Height>1200</Height>
          </PlatenOpticalResolution>
          <FlatbedSupported>true</FlatbedSupported>
        </Platen>
        <ADF>
          <ADFSupported>true</ADFSupported>
          <ADFSupportsDuplex>true</ADFSupportsDuplex>
          <ADFMinimumSize>
            <Width>1920</Width>
            <Height>1920</Height>
          </ADFMinimumSize>
          <ADFMaximumSize>
            <Width>8500</Width>
            <Height>15000</Height>
          </ADFMaximumSize>
          <ADFOpticalResolution>
            <Width>300</Width>
            <Height>300</Height>
          </ADFOpticalResolution>
        </ADF>
      </ScannerConfiguration>
      <ScannerStatus>
        <ScannerCurrentTime>2018-07-23T12:50:41Z</ScannerCurrentTime>
        <ScannerState>Idle</ScannerState>
        <ScannerStateReason>None</ScannerStateReason>
        <ScanToStatus>
          <PaperInADF>false</PaperInADF>
          <ScanToAvailable>false</ScanToAvailable>
        </ScanToStatus>
      </ScannerStatus>
      <ScannerInfo>
        <ModelNumber>CB532A</ModelNumber>
      </ScannerInfo>
    </wscn:ScanElements>
  </SOAP-ENV:Body>
</SOAP-ENV:Envelope>
```
And i was able to forge a curl post for wsdiscovery - So  sane-airscan stuff should work
```
curl --verbose --header Accept: --header "Content-Type: application/soap+xml; charset=utf-8"  --header "Transfer-Encoding: chunked" --user-agent "gSOAP/2.7" --data @req  10.0.0.29:8289/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
*   Trying 10.0.0.29:8289...
* Connected to 10.0.0.29 (10.0.0.29) port 8289 (#0)
> POST /StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery HTTP/1.1
> Host: 10.0.0.29:8289
> User-Agent: gSOAP/2.7
> Content-Type: application/soap+xml; charset=utf-8
> Transfer-Encoding: chunked
>
* Mark bundle as not supporting multiuse
< HTTP/1.1 202 ACCEPTED
< Server: gSOAP/2.7
< Content-Type: application/soap+xml; charset=utf-8
< Transfer-Encoding: chunked
< Connection: close
<
<?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:wscn="http://tempuri.org/wscn.xsd"><SOAP-ENV:Body><wscn:ScanElements><ScannerConfiguration><DeviceSettings><FormatSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ScanDocumentFormat" SOAP-ENC:arraySize="2"><item>jfif</item><item>hpraw</item></FormatSupported><QualityFactorSupported>false</QualityFactorSupported><ContentSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ContentType" SOAP-ENC:arraySize="1"><item>Auto</item></ContentSupported><DocumentSizeAutoDetectSupported>false</DocumentSizeAutoDetectSupported><FeederCapacity>50</FeederCapacity><DuplexRotation>180</DuplexRotation><Rotation>0</Rotation></DeviceSettings><Platen><ColorSupported xsi:type="SOAP-ENC:Array" SOAP-ENC:itemType="wscn:ColorEntryType" SOAP-ENC:arraySize="4"><item>BlackandWhite1</item><item>GrayScale8</item><item>RGB24</item><item>RGB48</item></ColorSupported><PlatenMinimumSize><Width>1920</Width><Height>1920</Height></PlatenMinimumSize><PlatenMaximumSize><Width>8500</Width><Height>11690</Height></PlatenMaximumSize><PlatenOpticalResolution><Width>1200</Width><Height>1200</Height></PlatenOpticalResolution><FlatbedSupported>true</FlatbedSupported></Platen><ADF><ADFSupported>true</ADFSupported><ADFSupportsDuplex>true</ADFSupportsDuplex><ADFMinimumSize><Width>1920</Width><Height>1920</Height></ADFMinimumSize><ADFMaximumSize><Width>8500</Width><Height>15000</Height></ADFMaximumSize><ADFOpticalResolution><Width>300</Width><Height>300</Height></ADFOpticalResolution></ADF></ScannerConfiguration><ScannerStatus><ScannerCurrentTime>2018-07-23T15:20:04Z</ScannerCurrentTime><ScannerState>Idle</ScannerState><ScannerStateReason>None</ScannerStateReason><ScanToStatus><PaperInADF>false</PaperInADF><ScanToAvailable>false</ScanToAvailable></ScanToStatus></ScannerStatus>* Closing connection 0
<ScannerInfo><ModelNumber>CB532A</ModelNumber></ScannerInfo></wscn:ScanElements></SOAP-ENV:Body></SOAP-ENV:Envelope>

```
avahi  (thx to reflector) lists the printer on localnet and different net
```
=   eth0 IPv4 HP LaserJet M2727nf MFP (8A21A9)              Web Site             local
   hostname = [Drucker.local]
   address = [10.0.0.29]
   port = [80]
   txt = []
=   eth0 IPv4 HP LaserJet M2727nf MFP (8A21A9)              UNIX Printer         local
   hostname = [Drucker.local]
   address = [10.0.0.29]
   port = [515]
   txt = ["TBCP=T" "Binary=T" "Transparent=T" "adminurl=http://Drucker.local." "priority=50" "product=(HP LaserJet M2727nf MFP)" "ty=HP LaserJet M2727nf MFP" "rp=RAW" "qtotal=1" "txtvers=1"]
=   eth0 IPv4 HP LaserJet M2727nf MFP (8A21A9)              PDL Printer          local
   hostname = [Drucker.local]
   address = [10.0.0.29]
```
 airscan-discover -d
```
e.g.
MDNS: resolve-ipv4("_ipp._tcp", "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: HTTP POST http://192.168.1.10/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.1.10 80
WSDD: HTTP trying 192.168.1.10:80
```
But  sane-airscan only seems to scan Printers reporting _ipp._tcp and _ipps._tcp, which is not ideal in my case. The printer does not provide ipp and so the ipp service address is provided via [DrPsychick docker-cups-airprint](https://github.com/DrPsychick/docker-cups-airprint)
This results in the printer not beeing detect by sane-airscan. Now i am asking myself how could one solve this? 

1. Creating a custom _uscan _scanner avahi service wouldn't work as it would point to the Host serving the service. (in this case the cups-avahi container)
2.  traffic redirect?

logfile [airscan-discover -d](https://pastebin.com/CHbTMT1t) (valid 3weeks)
[pcap success scanning hpaio](https://temp.sh/CmRMS/scanimage.pcapng) (valid 2weeks) 
Hint: Pcap of a Windows scan with the HP software also uses same proto/service etc...

alexpevzner, 2022-08-11:
Hi!

alexpevzner, 2022-08-11:
Hi @burnbabyburn,

could you please show me trace, created by `airscan-discover -t`, will be saved as ` airscan-discover-zeroconf.log` and `sane-airscan` protocol trace? Console output is not detailed enough to deal with this kind of problems.

burnbabyburn, 2022-08-11:
> airscan-discover -t
to eliminate possible errors due to my setup i moved the MFP to the same net as the host.
Here is the [logfile ](https://a.tmp.ninja/uKlhbAg.log). As the printer itself does not publish any ipp it seems to be ignored by the scan.

burnbabyburn, 2022-08-11:
Did a clean ubuntu 20.04 LTS install. Captured the `hp-setup` traffic. Discovery tries via MDNS. My printer does not respond to that, even on Windows :)
So i set ip to manual. HPLIP seems to ask via snmp for oid [1.3.6.1.4.1.11.2.3.9.1.1.7](http://oidref.com/1.3.6.1.4.1.11.2.3.9.1.1.7) if ip is defined manually. The printer responds with its capabilities. They seem to exchange some magic packets and configure the fax device in the end. [hp-setup capture](https://a.tmp.ninja/NsWyOZMe.pcapng). Than the printer answers to MDNS.

> And show me sane-airscan trace in the manual configuration mode too. May be, detailed error response will put some insight to the root cause of the scanning problem (which looks separate from the discovery problem).

burnbabyburn, 2022-08-11:
```
root@ubuntu-VirtualBox:/# scanimage --verbose --progress --test
Output format is not set, using pnm as a default.
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.1.234
WSDD:  + fe80::1570:8ada:856a:f378%2
WSDD: 192.168.1.234: started discovery, UDP port=49685
WSDD: fe80::1570:8ada:856a:f378: started discovery, UDP port=56122
API: sane_get_devices(): called
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "AirPrint HP_LaserJet_M2727nf_MFP_fax @ odroidn2"
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "AirPrint LaserJet @ odroidn2"
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "AirPrint HP_LaserJet_M2727nf_MFP @ odroidn2"
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "HP laserJet M2727 MFP @ odroidn2"
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "HP_LaserJet_M2727nf_MFP @ odroidn2"
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv4("_ipps._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipps._tcp"): "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"
MDNS: browse-ipv4("_ipps._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipps._tcp"): "HP_LaserJet_M2727nf_MFP @ odroidn2"
MDNS: browse-ipv4("_ipps._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipps._tcp"): "HP laserJet M2727 MFP @ odroidn2"
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: HTTP POST http://192.168.1.10/StableWSDiscoveryEndpoint/schemas-xmlsoap-org_ws_2005_04_discovery
WSDD: HTTP resolving 192.168.1.10 80
WSDD: HTTP trying 192.168.1.10:80
WSDD: HTTP fdpoll: {--} -> {-w}
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint LaserJet @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint LaserJet @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint HP_LaserJet_M2727nf_MFP @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "AirPrint HP_LaserJet_M2727nf_MFP @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipp._tcp", "HP laserJet M2727 MFP @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "HP laserJet M2727 MFP @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipp._tcp", "HP_LaserJet_M2727nf_MFP @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "HP_LaserJet_M2727nf_MFP @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipp._tcp", "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
WSDD: directed probe: already in progress, skipping
MDNS: resolve-ipv4("_ipps._tcp", "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipps._tcp", "HP_LaserJet_M2727nf_MFP_fax @ odroidn2"): 192.168.1.10:631
WSDD: directed probe: trying if=enp0s3, addr=192.168.1.10
[... 40 lines cut ...]
HP: HTTP resolving 192.168.1.233 8289
HP: HTTP trying 192.168.1.233:8289
HP: HTTP fdpoll: {--} -> {-w}
HP: HTTP 970 bytes sent
HP: HTTP done request sending
HP: HTTP fdpoll: {-w} -> {rw}
HP: HTTP fdpoll: {rw} -> {r-}
HP: HTTP 145 bytes received
HP: HTTP POST http://192.168.1.233:8289: got response headers (400)
HP: HTTP 5 bytes received
HP: HTTP 656 bytes received
HP: HTTP done response reception
HP: HTTP POST http://192.168.1.233:8289: 400 Bad Request
HP: HTTP POST http://192.168.1.233:8289: Bad Request
HP: scanner capabilities query: Bad Request
HP: DEVICE_STM_PROBING->DEVICE_STM_PROBING_FAILED
HP: removed from device table
HP: closed protocol "WSD"
HP: device destroyed
API: sane_open("w0:HP"): Error during device I/O
scanimage: open of device airscan:w0:HP failed: Error during device I/O
API: sane_exit(): called
zeroconf: ZEROCONF_WSD: initial scan finished
API: sane_exit(): OK
```

alexpevzner, 2022-08-12:
This is some different protocol. Looks similar to WSD, but it is not WSD.

I currently don't have my personal resources to implement this protocol in the `sane-airscan` by myself, and I don't have a hardware to test implementation.

If you can code in C, you can consider implementing it by yourself (with my help, of course) and contributing to the `sane-airscan`. It is not as hard to do, as it sounds.

alexpevzner, 2022-08-12:
And if you will consider to try, this is better to contribute to the https://github.com/alexpevzner/sane-airscan-unstable

This is development branch of the `sane-airscan`, for my convenience managed as a separate repository. Currently there is a very small difference between these branches, and I synchronize them time to time. But all big changes I prefer to appear in the unstable branch first.

burnbabyburn, 2022-08-12:
TBH my last C code is 10yrs old now. But i can try though :) It would help a lot if you could pinpoint me to the relevant files i need to look at :)
Looking (very) roughly at your codebase, mainly airscan-wsd.c => Seems like the XMLNS is a lil bit different to standard WSD (wsa vs wscn or SOAP:Envelope vs SOAP-ENV:Envelope)
The protocol communication seems rather simple (maybe a [MS-BSSRR](https://docs.microsoft.com/en-us/openspecs/windows_protocols/ms-bdsrr/6a3d3e90-e56e-480b-9a1e-d004c45cccd6) flavor?)

alexpevzner, 2022-08-12:
Yes, it really looks like MS-BSSRR dialect.

First thing to do is to choose a protocol name (probably, BSSRR). Add appropriate constant to the ID_PROTO enum. Add it to the `id_name_table` table in the `airscan-id.c` file. Add protocol handler constructor prototype, similar to `proto_handler_escl_new` or `proto_handler_wsd_new` to the `airscan.h`. Add handling of the newly added protocol to the `proto_handler_new` function in the `airscan.h`

Then create a new protocol handler. eSCL and WSD handlers are `airscan-escl,c` and `airscan-wsd.c`, respecyively. I suggest to start from `airscan-wsd.c`, because protocol is quite similar to WSD. But still want a separate protocol handler rather that a special mode of the WSD handler, as protocols are different enough.

It should be enough for manually-configured device. 

Once we well have it all implemented, we may look how to add discovery.

tracend, 2022-09-15:
>  € Ubuntu 20.04 LTS also needs package `universal-ctags` for TAGS build target

I can confirm this. 

Firstly, thank you, this lib definitely helped me fix the printer setup on my Ubuntu but when compiling I got the message:
```
make: ctags: Command not found
```

Please add in the Readme file that Ubuntu, Debian and similar have the additional dependency: 
```
apt-get install universal-ctags
```

Thanks again

alexpevzner, 2023-07-06:
Looks like this issue is not relevant anymore...

closed 2023-07-06 by alexpevzner