alexpevzner/sane-airscan #252: HP LaserJet Pro MFP M127fn: confirmed to work with WSD

debiantriage, 2022-08-18:
https://answers.launchpad.net/hplip/+question/702802

> wagner@Linux-wet:~$ scanimage -L
device `airscan:w0:HP LaserJet Pro MFP M127fn[00A731]' is a WSD HP LaserJet Pro MFP M127fn[00A731] ip=192.168.0.128

> wagner@Linux-wet:~$ airscan-discover
[devices]
  HP LaserJet Pro MFP M127fn[00A731] = http://192.168.0.128:3911/scanner, WSD

The LaserJet Pro MFP M127fn was vended by Amazon towards the end of 2013. It does not understand the eSCL or IPP-over-USB protocols.

alexpevzner, 2023-07-06:
1/3 of devices in the list are WSD-only

When I was only considering to add WSD support, I was not sure that it will be so useful

closed 2023-07-06 by alexpevzner

debiantriage, 2023-07-06:
> 1/3 of devices in the list are WSD-only
> 
> When I was only considering to add WSD support, I was not sure that it will be so useful

I can remember your wondering about introducing WSD support. In my book it is a winner. It certainly provides me with another tool to help a user.

- Scanners in a Multi Function Device having AirPrint version 1.4 will
almost certainly provide  eSCL. No WSD needed.

-  With version AirScan 1.3 there there isn't any certainty of eSCL provision, but many do have it, as seems to be the case with HP devices.
This takes us back to the end of 2012.

-  I have found only one device that supports eSCL with AirPrint 1.2. We write this version off.

- WSD introduction appears to date from about 2009/2010. Another three years of supported scanners!

A possible drawback is the limited resolutions offered by some vendors (Brother?). I do not understand why this should be so. Is it a WSD restriction?  I have a frontend (based entirely on your work) that scans in any  requested resolution with WSD.

alexpevzner, 2023-07-06:
> A possible drawback is the limited resolutions offered by some vendors (Brother?). I do not understand why this should be so. Is it a WSD restriction?

No, I don't think so.

HW vendors usually don't write WSD/eSCL/etc servers at home. I guess they tend to license implementations from the 3rd party. The today's scanner firmware is like Lego constructor, build from different parts came from different origins (like most of today's software, unfortunately. They call it "code reuse").

My Kyocera has at least 4 different built-in HTTP servers used for different purposes (it becomes noticeable, because these HTTP servers put different strings into some "informative" HTTP headers, have different behaviour in nuances and so on). I believe, other HW vendors do the similar things

It may be limitation of these 3rd party components. Please note also, Brother devices are **very** limited in memory. They just don't have enough memory to keep the entire image at the resolution they offer. Probably, from the scanning unit image bytes go directly to the hardware JPEG encoder and then transferred to the host.

In the combination with the WSD server, it may be just not enough memory to deal with the high-resolution images