alexpevzner/sane-airscan #257: Brother DCP-L2550DW series I/O error

lake-effect, 2022-09-14:
## Environment
```
lsb_release -a
LSB Version:	core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.4 LTS
Release:	20.04
Codename:	focal
```

```
systemctl list-units "ipp-usb*" | grep service
  ipp-usb.service loaded active running Daemon for IPP over USB printer support
```

```
dpkg -l | grep -i airscan
ii  sane-airscan                                     0.99.27-1+89.1                              amd64        SANE backend for MFP and document scanners that
```

```
lsusb
Bus 003 Device 005: ID 04f9:0424 Brother Industries, Ltd
```

## Steps

1. Fire up `simple-scan`
2. Use first detected device (eSCL)
3. See error: scan failed

## Logs
[simple-scan-DCP-L2550DW-series.log](https://github.com/alexpevzner/sane-airscan/files/9563568/simple-scan-DCP-L2550DW-series.log)

It looks like Github won't let me attach a tarball.

```
simple-scan 
API: sane_init(): called
loading configuration file /etc/sane.d/airscan.conf
MDNS: AVAHI_CLIENT_S_RUNNING
WSDD: netif addresses update:
WSDD:  + 192.168.122.1
WSDD:  + 172.17.0.1
WSDD:  + 192.168.20.54
WSDD:  + fe80::e499:b4d:eed7:a9c6%7
WSDD: 192.168.122.1: started discovery, UDP port=52891
WSDD: 172.17.0.1: started discovery, UDP port=48868
WSDD: 192.168.20.54: started discovery, UDP port=43876
WSDD: fe80::e499:b4d:eed7:a9c6: started discovery, UDP port=49112
MDNS: browse-ipv4("_ipp._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_ipp._tcp"): "DCP-L2550DW series"
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipp._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_ipps._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
MDNS: browse-ipv4("_uscan._tcp"): AVAHI_BROWSER_NEW
MDNS: browse-ipv4("_uscan._tcp"): "DCP-L2550DW series"
API: sane_get_devices(): called
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_CACHE_EXHAUSTED
zeroconf: zeroconf_device_list_get: requested
zeroconf: device_list wait: requested
zeroconf: device_list wait: DNS-SD not finished...
MDNS: resolve-ipv4("_ipp._tcp", "DCP-L2550DW series"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_ipp._tcp", "DCP-L2550DW series"): 127.0.0.1:60000
WSDD: directed probe: trying if=lo, addr=127.0.0.1
WSDD: directed probe: skipping loopback address
MDNS: resolve-ipv4("_uscan._tcp", "DCP-L2550DW series"): AVAHI_RESOLVER_FOUND
MDNS: resolve-ipv4("_uscan._tcp", "DCP-L2550DW series"): 127.0.0.1:60000
zeroconf: found urn:uuid:e3248000-80ce-11db-8000-b422000e62af
zeroconf:   method:    ZEROCONF_USCAN_TCP
zeroconf:   interface: 1 (lo)
zeroconf:   name:      DCP-L2550DW series
zeroconf:   model:     Brother DCP-L2550DW series
zeroconf:   addresses:
zeroconf:     127.0.0.1
zeroconf:   protocol:  eSCL
zeroconf:   endpoints:
zeroconf:     http://127.0.0.1:60000/eSCL/
zeroconf:   device:    0000 (created)
WSDD: directed probe: trying if=lo, addr=127.0.0.1
WSDD: directed probe: skipping loopback address
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_uscan._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCAN_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_uscans._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_USCANS_TCP: initial scan finished
zeroconf: device_list wait: DNS-SD not finished...
MDNS: browse-ipv6("_scanner._tcp"): AVAHI_BROWSER_ALL_FOR_NOW
zeroconf: ZEROCONF_MDNS_HINT: initial scan finished
zeroconf: device_list wait: OK
zeroconf: zeroconf_device_list_get: building list of devices
zeroconf: DCP-L2550DW series (0): can: eSCL, use: eSCL
zeroconf: zeroconf_device_list_get: resulting list:
zeroconf:   eSCL  "e0:DCP-L2550DW series"
API: sane_get_devices(): done
WSDD: 172.17.0.1: done discovery
WSDD: 192.168.20.54: done discovery
WSDD: fe80::e499:b4d:eed7:a9c6: done discovery
WSDD: 192.168.122.1: done discovery
zeroconf: ZEROCONF_WSD: initial scan finished
zeroconf: initial scan timer expired
API: sane_open("e0:DCP-L2550DW series"): called
zeroconf: device_list wait: requested
zeroconf: device_list wait: OK
DCP-L2550DW series: device created
DCP-L2550DW series: DEVICE_STM_OPENED->DEVICE_STM_PROBING
DCP-L2550DW series: using protocol "eSCL"
DCP-L2550DW series: HTTP GET http://127.0.0.1:60000/eSCL/ScannerCapabilities
DCP-L2550DW series: HTTP using timeout: 5000 ms
DCP-L2550DW series: HTTP resolving 127.0.0.1 60000
DCP-L2550DW series: HTTP trying 127.0.0.1:60000
DCP-L2550DW series: HTTP fdpoll: {--} -> {-w}
DCP-L2550DW series: HTTP 84 bytes sent
DCP-L2550DW series: HTTP done request sending
DCP-L2550DW series: HTTP fdpoll: {-w} -> {rw}
DCP-L2550DW series: HTTP fdpoll: {rw} -> {r-}
DCP-L2550DW series: HTTP 183 bytes received
DCP-L2550DW series: HTTP GET http://127.0.0.1:60000/eSCL/ScannerCapabilities: got response headers (200)
DCP-L2550DW series: HTTP 4096 bytes received
DCP-L2550DW series: HTTP 3365 bytes received
DCP-L2550DW series: HTTP done response reception
DCP-L2550DW series: HTTP GET http://127.0.0.1:60000/eSCL/ScannerCapabilities: 200 OK
DCP-L2550DW series: HTTP GET http://127.0.0.1:60000/eSCL/ScannerCapabilities: OK
DCP-L2550DW series: scanner capabilities: XML: missed scan:ScannerCapabilities
DCP-L2550DW series: DEVICE_STM_PROBING->DEVICE_STM_PROBING_FAILED
DCP-L2550DW series: removed from device table
DCP-L2550DW series: closed protocol "eSCL"
DCP-L2550DW series: device destroyed
API: sane_open("e0:DCP-L2550DW series"): Error during device I/O
```

## Notes

This used to work over USB, but I recently unplugged everything and plugged it all back in, and suddenly it doesn't. I'm baffled.

alexpevzner, 2022-09-14:
Have you tried to reboot the printer and restart the `ipp-usb` daemon?

lake-effect, 2022-09-16:
Restarting the daemon worked, thanks.

closed 2022-09-16 by lake-effect