alexpevzner/sane-airscan #258: Kyocera FS-1028MFP - Device Busy

agapovmax, 2022-09-14:
Connecting Kyocera FS-1028 over LAN. Enabled all WSD, Apple network features.

+ OS - Astra Linux 1.7
+ dpkg -l
```sane-airscan   0.99.19-1~bpo10+1
sane-utils     1.0.32-4+ci202112071753+astra3
```

+ airscan-discover
```
[devices]
  Kyocera FS-1028MFP = http://10.0.0.49:5358/WSDScanner, WSD
```
+ airscan.conf
```
[devices]
"Kyocera MFP Scanner" = http://10.0.0.49:5358/WSDScanner, WSD
"Kyocera eSCL" = http://10.0.0.49:9095/eSCL, eSCL ; only for test
```

+ debug log
```
Kyocera FS-1028MFP: HTTP POST http://10.0.0.49:5358/WSDScanner
Kyocera FS-1028MFP: HTTP using timeout: 30000 ms
Kyocera FS-1028MFP: HTTP resolving 10.0.0.49 5358
Kyocera FS-1028MFP: HTTP trying 10.0.0.49:5358
Kyocera FS-1028MFP: HTTP fdpoll: {--} -> {-w}
Kyocera FS-1028MFP: HTTP 1841 bytes sent
Kyocera FS-1028MFP: HTTP done request sending
Kyocera FS-1028MFP: HTTP fdpoll: {-w} -> {rw}
Kyocera FS-1028MFP: HTTP fdpoll: {rw} -> {r-}
Kyocera FS-1028MFP: HTTP 1448 bytes received
Kyocera FS-1028MFP: HTTP POST http://10.0.0.49:5358/WSDScanner: got response headers (500)
Kyocera FS-1028MFP: HTTP 733 bytes received
Kyocera FS-1028MFP: HTTP done response reception
Kyocera FS-1028MFP: HTTP POST http://10.0.0.49:5358/WSDScanner: 500 Internal Server Error
Kyocera FS-1028MFP: HTTP POST http://10.0.0.49:5358/WSDScanner: Internal Server Error
Kyocera FS-1028MFP: PROTO_OP_SCAN: decoding
Kyocera FS-1028MFP: fault code: ServerErrorNotAcceptingJobs
Kyocera FS-1028MFP: PROTO_OP_SCAN: decoded: status="Success" next=PROTO_OP_CHECK delay=0
Kyocera FS-1028MFP: PROTO_OP_CHECK: submitting: attempt=0
```
+ fly-scan show message "Device busy"

+ simple-scan show message "Can't start scanning"
[sane-kyocera.zip](https://github.com/alexpevzner/sane-airscan/files/9566279/sane-kyocera.zip)

agapovmax, 2022-09-19:
up

alexpevzner, 2022-09-19:
Kyocera probably needs manual action on its front panel to initiate WSD scan: **Send->WSD Scan->From Computer**. At least, my Kyocera ECOSYS M2040dn works this way, and some other reported too.

https://github.com/alexpevzner/sane-airscan#note5

alexpevzner, 2022-09-19:
BTW, eSCL on my Kyocera can be enabled and disabled. If your printer supports eSCL, but it is disabled by default, it makes sense to enable it: eSCL works there much better and doesn't require any manual actions.

agapovmax, 2022-09-20:
@alexpevzner thanx. Send->WSD Scan->From Computer helps. 
Unfortunately, Kyocera MFP1028 doesn't support eSCL. Please add info about this model!

commit 2b0f71f1 (2023-07-06): README.md: added Kyocera FS-1028MFP (fixes #258)
closed 2023-07-06 by alexpevzner