alexpevzner/sane-airscan #263: Full width of scanner not accessible with airscan

brettpim, 2022-12-04:
I am running Ubuntu 22.04 on a Lenovo Thinkpad with Gnome has my desktop. I have xsane 0.999 installed. My Scanner is an HP Color LaserJet MFP M476dw.  My scanner is accessible by two backends; ```scanimage -l``` returns
> device `hpaio:/net/HP_Color_LaserJet_MFP_M476dw?ip=192.168.0.41' is a Hewlett-Packard HP_Color_LaserJet_MFP_M476dw all-in-one

>device `airscan:w1:HP Color LaserJet MFP M476dw (3DBC02)' is a WSD HP Color LaserJet MFP M476dw (3DBC02) ip=192.168.0.41, 2607:f2c0:ead8:3cf7:9ead:97ff:fe5c:7d9, 2607:f2c0:ead8:3cf7::57f, fdfd:6a08:40d:0:9ead:97ff:fe5c:7d9, fdfd:6a08:40d::57f

When I run xsane or xscanimage using the airscan backend I am only able to scan 21cm of width, not the entire 21.59cm(=8.5in) that the scanner is capable of.  When I use the hpaio backend I have access to the full width available.

alexpevzner, 2022-12-08:
Hi @brettpim,

may I ask you to gather protocol trace logs? To obtain it, please uncomment the following lines in the `/etc/sane.d/airscan.conf`:

```
[debug]
trace = ~/airscan/trace
enable = true
```

Trace files will be in the `~/airscan/trace` directory, which will be created automatically. Currently I want to look to the `.log` files only `.tar` files are not needed for now.

alexpevzner, 2023-07-06:
Closed as there is no response from bug reporter. I can't proceed without detailed logs...

closed 2023-07-06 by alexpevzner